srsran/srsRAN_Project #480: Only ping possible between srsUE and srsGNB

oliverxsch, 2024-02-19:
## Issue Description ##
Hello,
this was my old Issue, but because this problem occured and the old problem did not persist, i opened this new isssue.
https://github.com/srsran/srsRAN_Project/issues/423

## Setup Details ##
Only ping from UE to gnb is possible. 
If i ping the UE from the gnb, the messages like in the screenshots happen.
Im using two X310s connected via SMA cable, both with their own host pc.
Using X410 as the gnb does not change the situation.
[image]

Here are my configs and logs.:
[srsRan Github.zip](https://github.com/srsran/srsRAN_Project/files/14333220/srsRan.Github.zip)

## Additional Information ##
Please check my old issue: https://github.com/srsran/srsRAN_Project/issues/423
There are a lot of suggestions which we already tried.

Thanks in advance

pgawlowicz, 2024-02-19:
Hi,
could you list all interfaces on your srsUE host?  You can use `ifconfig`

oliverxsch, 2024-02-21:
> Hi, could you list all interfaces on your srsUE host? You can use `ifconfig`

Here:
[image]

pgawlowicz, 2024-02-21:
So there is no `tun_srsue` interface, the interface is created when srsUE attached and gets an IP address. Therefore, I believe it was not connected, or?

oliverxsch, 2024-02-21:
No, it did not connect. Sorry, i did not start the gnb at the same time.
Here is a screen with the gNB running and connected. the interface is there.
[image]
[image]

pgawlowicz, 2024-02-26:
could you print routing tables on both gnb and ue? 
could you try to ping the UE from Open5GS docker container?
```
docker exec -it open5gs_5gc ping 10.45.1.2
```

oliverxsch, 2024-02-28:
> could you print routing tables on both gnb and ue? could you try to ping the UE from Open5GS docker container?
> 
> ```
> docker exec -it open5gs_5gc ping 10.45.1.2
> ```

Okay. Here are the routing tables for UE and gNB:
[image]
[image]

And this is what happens if i use your command on the gnb to ping the UE:
[image]

pgawlowicz, 2024-02-28:
could you make screenshots of console traces from gnb and ue?

oliverxsch, 2024-02-28:
Now your command is working somehow.
But the iperfs commands provided in the tutorial do not work.
Thanks anyway!
I believe the tutorial got changed (to include the CN per docker) but the ping commands and iperf commands did not change?
How would i need to adjust the iperf command ?

pgawlowicz, 2024-02-28:
If the routes are configured correctly, then the commands provided in the tutorial should work.

oliverxsch, 2024-02-28:
Oh, i just found my problem.
I uses the command:
Next add default route for the UE (...) on the wrong machine xD

Thanks for your help

closed 2024-02-28 by oliverxsch

pgawlowicz, 2024-02-28:
iperf server on gnb:
```
iperf3 -s -i 1
```

iperf client on srsUE:
```
iperf3 -c 10.53.1.1 -i 1 -t 60 -u -b 10M
```

So in your case the UE has no idea how to reach iperf server, as the default route is going somewhere else. 
You can start the client inside open5gs docker and then run `iperf3 -c 10.45.1.1 -i 1 -t 60 -u -b 10M`