srsran/srsRAN_Project #489: over the air setup - GNB is not pinging the UE

Mohammed24699, 2024-02-22:
## Issue Description ##
I am trying to connect the UE and GNB (over the air setup). The UE can ping the GNB (10.45.1.1), but, the GNB is not pinging the UE. 

## Setup Details ##
computer 1 (UE) - intel xeon(R) w-2255 cpu @ 3.70 GHZ * 20
computer 2 (gnb) - 11 th gen intel core -19 11900k @ 3.50 GHZ * 16
 
We installed the latest UE( 23_11 )in one system and GNB in another system and connected the USRP (b210) to each PC. USRP's are connected with SMA cables attaching 20 dB attenuators, and the experiment is conducted in a low-latency kernel.

## Expected Behavior ##
According to the documentation (https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html), after the UE gives the IP address (10.45.1.2), we need to ping the UE and GNB each other.

## Actual Behaviour ##
There is no problem pinging GNB with UE. But GNB is not pinging the UE.

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue, e.g. RF setup, application config files]

## Additional Information ##
I'm performing the kernel in a low-latency kernel with the official config files (I'm attaching them below); take a look at the UE and GNB log files too.

[gnb.log.log](https://github.com/srsran/srsRAN_Project/files/14380292/gnb.log.log)

pgawlowicz, 2024-02-23:
Hi, Did you configure the routing correctly?

pgawlowicz, 2024-02-23:
what about the default route at ue?

Mohammed24699, 2024-02-23:
[image]

pgawlowicz, 2024-02-23:
you need to add the default gw:

```
sudo ip route add default via 10.45.1.1 dev tun_srsue
```

Mohammed24699, 2024-02-23:
i ran this on UE, still GNB does not
[image]
[image]
 ping UE

pgawlowicz, 2024-02-23:
the route on gnb is wrong, it should be 
```
10.45.0.0       10.53.1.2       255.255.0.0     UG    0      0        0 br-....
```

and you have:
```
10.45.0.0       10.53.1.2       255.255.255.0     UG    0      0        0 br-....
```

so you do not have a route for address 10.45.1.2

Mohammed24699, 2024-02-23:
udo route change -net 10.45.0.0 netmask 255.255.0.0 dev br-6341da468396
Usage: route [-nNvee] [-FC] [<AF>]           List kernel routing tables
       route [-v] [-FC] {add|del|flush} ...  Modify routing table for AF.

       route {-h|--help} [<AF>]              Detailed usage syntax for specified AF.
       route {-V|--version}                  Display version/author and exit.

        -v, --verbose            be verbose
        -n, --numeric            don't resolve names
        -e, --extend             display other/more information
        -F, --fib                display Forwarding Information Base (default)
        -C, --cache              display routing cache instead of FIB

  <AF>=Use -4, -6, '-A <af>' or '--<af>'; default: inet
  List of possible address families (which support routing):
    inet (DARPA Internet) inet6 (IPv6) ax25 (AMPR AX.25) 
    netrom (AMPR NET/ROM) ipx (Novell IPX) ddp (Appletalk DDP) 
    x25 (CCITT X.25)

Mohammed24699, 2024-02-25:
[image]
[image]

Still, the downlink doesn't ping.

pgawlowicz, 2024-02-25:
The default route at srsue goes over eno1 but should go over tun_srsue

Mohammed24699, 2024-02-25:
[image]
changed it to default, still same problem

pgawlowicz, 2024-02-25:
Does the ping from UE to core work? How do you ping from the core?

Mohammed24699, 2024-02-25:
[image]

[image]

1. Ping from UE to core works
2. I'm pinging UE from the docker container.

pgawlowicz, 2024-02-26:
could you try to ping the UE from Open5GS docker container:
```
docker exec -it open5gs_5gc ping 10.45.1.2
```

Mohammed24699, 2024-02-26:
[image]

Mohammed24699, 2024-03-01:
Hello, do you have any idea how I can make my UE more stable? Sometimes it gives the IP address, sometimes it doesn't.
I'm using the official config files. Do I need to make any changes to TX and RX gains?
[image]
[image]

pgawlowicz, 2024-03-06:
Do you use external clock (or gpsdo clock) for your b210s?

Mohammed24699, 2024-03-06:
yes we use CDA 2990 octo clock

pgawlowicz, 2024-03-06:
could you provide the console trace from both gnb and srsUE when the UE is connected and sending a ping? you need to press `t` in both consoles.

Mohammed24699, 2024-03-06:
[image]
[image]
[image]
[image]

pgawlowicz, 2024-03-11:
did you try tuning rx/tx gains on both gnb and srsUE?  I would start with reducing TX gain of gnb a (by 5-10db). Could you make a screenshort of the CPU load at srsUE host PC? eg. from `htop`

closed 2024-03-13 by Mohammed24699

pgawlowicz, 2024-03-13:
@Mohammed24699 did you solve the issue?

Mohammed24699, 2024-03-13:
As you suggested, we tried tuning the tx and rx gains in the config files of both gNB and UE. Now that the UE provides the IP, it's quite stable. We're currently conducting an iperf3 test at 10 m/s, and we've had the UE stable for about 23 min. Thanks for the help.

pgawlowicz, 2024-03-13:
Great! thanks for your feedback!

Mohammed24699, 2024-03-26:
I am just wondering that can we connect 2 srs UEs to 1gNB(open 5gs) ??

Mohammed24699, 2024-03-26:
I am just wondering that can we connect 2 srs UEs to 1gNB(open 5gs) ?? **over the air setup**

Mohammed24699, 2024-03-26:
https://github.com/srsran/srsRAN_Project/issues/327

currently i am looking into this, i guess it will help. if you have any information related to this please let me know.

pgawlowicz, 2024-03-26:
sure, you can connect 2 srsUEs, just need to assign different USIM parameters to each and also add them to Open5GS database. You can take the configs from here https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#multi-ue-emulation and change the RF section to use USRP.

Mohammed24699, 2024-03-26:
i created a new issue here https://github.com/srsran/srsRAN_Project/issues/532.

Mohammed24699, 2024-09-23:
Hello,

We are using two different GPS clocks by replacing the OctoClock. I have a question regarding the configuration: Do we need to change anything in the clock and sync section of the gNB and UE settings? Additionally, while connecting to the USRP B210, we are currently connecting the GPS clocks to the "REF IN" port. Is this correct, or should we be using the "GPS ANT" port instead?

Thanks,

pgawlowicz, 2024-09-24:
You need to use "REF IN" port and use "external" clock source. The "GPS ANT" port is only or the internal GPSDO module of b210 (if installed).

Mohammed24699, 2024-09-24:
That makes sense. thanks a lot.

Mohammed24699, 2024-10-14:
Hello,

We have collected the IQ samples of srsRAN gNB and UE during their communication in various scenarios and are using the WaveJudge software to analyze the 5G metrics. However, we are unable to see all the messages in the software. To access the complete set of messages, we need the values of K_SSB, KOffset for PSS, PBCH, and SSS in the gNB configuration.

Any guidance or information you could provide on these values would be greatly appreciated.

Thank you for your assistance.

[image]

pgawlowicz, 2024-10-15:
The question is not related to the original issue in this thread, which is closed. Please start a new thread. Please use the Discussion section.

Mohammed24699, 2024-10-15:
Thanks for the recommendation.