srsran/srsRAN_Project #490: Connecting gNB to open5gs on srsRAN Project

744739005, 2024-02-25:
## Issue Description ##
The gNB is unable to establish a connection with the dockerized version of open5gs provided in the srsRAN_Project when using ZMQ instead of UHD.

## Setup Details ##
Ubuntu 22.04.1 LTS
ZMQ
PC: 8GB Ram

## Expected Behavior ##
[What you expect to happen]
See attachment.
## Actual Behaviour ##
gNB can't attach to open5gs
## Steps to reproduce the problem ##
Hello srsRAN community,
When I followed the instructions on https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html and executed "docker compose up --build 5gc", it first opened a dockerized version of Open5GS. Then, under the srsRAN_Project/configs/ directory, I entered sudo gnb -c ./gnb_zmq.yaml. The gnb_zmq.yaml is the gNB configuration file for the ZMQ service provided in the link. The following error occurred:

--== srsRAN gNB (commit bcf941b34) ==--

Connecting to AMF on 10.53.1.2:38412 Failed to load RF plugin libsrsran_radio_zmq.so: libsrsran_radio_zmq.so: cannot open shared object file: No such file or directory Factory for radio type zmq not found. Make sure to select a valid type. srsGNB ERROR: Unable to create radio session.

The Open5GS console recorded the following:

open5gs_5gc | 02/25 07:22:56.422: [amf] INFO: gNB-N2 accepted[10.53.1.1]:36457 in ng-path module (../src/amf/ngap-sctp.c:113) open5gs_5gc | 02/25 07:22:56.422: [amf] INFO: gNB-N2 accepted[10.53.1.1] in master_sm module (../src/amf/amf-sm.c:706) open5gs_5gc | 02/25 07:22:56.422: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1034) open5gs_5gc | 02/25 07:22:56.422: [amf] INFO: gNB-N2[10.53.1.1] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:745) open5gs_5gc | 02/25 07:22:56.585: [amf] INFO: gNB-N2[10.53.1.1] connection refused!!! (../src/amf/amf-sm.c:758) open5gs_5gc | 02/25 07:22:56.585: [amf] INFO: [Removed] Number of gNBs is now 0 (../src/amf/context.c:1061)

## Additional Information ##
[gNB_FILES.zip](https://github.com/srsran/srsRAN_Project/files/14395587/gNB_FILES.zip)

pgawlowicz, 2024-02-25:
Did you installed zmq and did you compiled the srsRAN project with zmq enabled? See here https://github.com/srsran/srsRAN_Project/blob/main/CMakeLists.txt#L63

744739005, 2024-02-26:
Thank you so much for addressing my issue. I truly appreciate the time and effort you've put into resolving it. Your assistance has made a significant difference, and I'm grateful for the support from the community. Keep up the fantastic work!

closed 2024-02-26 by 744739005