srsran/srsRAN_Project #492: File transfer between 2 UE's for COTS UE 5G SA setup

ff187, 2024-02-28:
Dear All, 

Greetings. I am wondering if any of you have tried to transfer a file (text/image/audio/video) from one UE to another UE using the 5G SA setup.  Any comments/feedback/hints regrading the process will be highly appreciated. Thank you.

pgawlowicz, 2024-02-29:
Just set up an iperf server on one UE, and start an iperf client on the other UE. If everything is correct, it should work out of the box.

pgawlowicz, 2024-02-29:
BTW. it is not an issue, next time please use the Discussion section.

closed 2024-02-29 by pgawlowicz

ff187, 2024-02-29:
Dear @pgawlowicz ,

In this the UE's are two COTS UE **(OnePlus 9 Pro)**. How can I send the files in this case? As iperf works well if the UE's are PC instead of android phones. Thank you.

pgawlowicz, 2024-03-01:
simply install iperf from app store

ff187, 2024-03-09:
> simply install iperf from app an store

Dear @pgawlowicz,

Thanks a lot for your feedback. I have tried that earlier. Somehow the iperf can only ping upto the gateway which is I think 10.45.0.1 by default. But I can not either ping or iperf from one UE to another UE directly. Please let me know if you have any further feedback/guidelines from where I can learn. I just need a small piece information to bridge the UE's with the the help of gateway. Thanks once again for your time and help.

ff187, 2024-03-12:
> could you share your configs? Do you run dockerized Open5gs? Do the phones get IP addresses in the same subnet?

Dear @pgawlowicz 

I am hereby sharing the configuration files, log files and pcap files. No, I did not run dockerized Open5gs. The phones did not get IP addresses on the same subnet. Please have a look and let me know your valuable feedback. Thank you.

[image]
[image]

[Configuration Files.zip](https://github.com/srsran/srsRAN_Project/files/14577577/Configuration.Files.zip)
[Log Files.zip](https://github.com/srsran/srsRAN_Project/files/14577578/Log.Files.zip)
[PCAP Files.zip](https://github.com/srsran/srsRAN_Project/files/14577579/PCAP.Files.zip)

ff187, 2024-03-12:
> could you share your configs? Do you run dockerized Open5gs? Do the phones get IP addresses in the same subnet?

Dear @pgawlowicz 

I am hereby sharing the configuration files, log files and pcap files. No, I did not run dockerized Open5gs. The phones did not get IP addresses on the same subnet. Please have a look and let me know your valuable feedback. Thank you.

[image]
[image]

[Configuration Files.zip](https://github.com/srsran/srsRAN_Project/files/14577631/Configuration.Files.zip)
[Log Files.zip](https://github.com/srsran/srsRAN_Project/files/14577632/Log.Files.zip)
[PCAP Files.zip](https://github.com/srsran/srsRAN_Project/files/14577633/PCAP.Files.zip)

pgawlowicz, 2024-03-13:
hi, not sure what we see on the print screens. could you explain?

ff187, 2024-03-13:
> hi, not sure what we see on the print screens. could you explain?

Dear @pgawlowicz 

The COTS UE's are connected to the gNB which should be 10.45.0.1 but it is showing 10.45.0.9 and 10.45.0.10 respectively for the two COTS UE's which is wrong for sure.  I think I need to work with the routing table in this case to deal with multiples UE's. Is there any way to check the routing table of gNB? Thank you.

pgawlowicz, 2024-03-13:
indeed, this is wrong, did you configure static IP addresses in the subscriber database?

ff187, 2024-03-13:
> indeed, this is wrong, did you configure static IP addresses in the subscriber database?

Dear @pgawlowicz,

No, I did not configure static IP addresses in the subscriber database.

ff187, 2024-03-13:
> could you connect with adb to the phones and print routing tables?
> ```
> adb shell
> route -n
> ```

Dear @pgawlowicz ,

My lab is closed now. I will get back to you at the earliest. Thank you very much.

pgawlowicz, 2024-03-22:
what about?
```
ip route show
```

pgawlowicz, 2024-03-22:
Seems that the phones are not in the same subnetwork, hence they have different routes. It probably has to do with Open5gs config. Could you check their IPs on the web GUI?

You can find it like this:

[image]

pgawlowicz, 2024-03-23:
Could you try to assign the IPs for the UEs? There should be an edit button.

pgawlowicz, 2024-04-02:
hi @ff187 any update on this issue?

jorgebaranda, 2024-06-10:
Dear @pgawlowicz and @ff187, 
Let me share with you my experience in case we can find together an explanation for this. I am in a similar situation and browsing in the internet I found this thread.
In my case, I wanted, as @ff187, to communicate 2UEs using a 5G SA network. In my setup, I am using an Open5GS, a COTS UE (Realme9) and a B210. But since I do not have 2 COTS UEs, I tried to do it with ZMQ, so I have to expand a little bit the setup by adding an additional gNB using ZMQ.
Summarising, on one hand, I have an "Over-the-air" (OTA) setup (B210+"srsRAN gnb uhd"+REalme9) setup and on the other hand, I have a ZMQ setup ("srsRAN gnb zmq" + srsUE) using both the same OpenGs core.

Unfortunately, this didn't work because I observed a similar think that @ff187 in the Realme9 mobile phone. With what it seems the same free-app to do pings, iperfs, ...., I see the connection/attachment details in the Realme9 phone. The UE was given an address in a /30 network, so no other UE can be contacted. See following image:
[image]

However, checking connection details on the  "srsUE" UE by making "ifconfig", I saw that the srsUE was given a /24 address that was containing the address of the Realme9, so in theory is reacheable. Then, I did the following experience: rather than running the "srsRAN gnb uhd", I run a second "srsRAN gnb zmq" process in another computer and another "srsUE" connecting to same Open5GS core. Effectively, when attached the second srsUE, this worked since both UEs were given /24 address and I could ping between each other because they were in the same subnet.

In another experiment, by casuality, I left an "srsUE zmq" attached to the network and I attached the "Realme9", then I enter and leave flight mode and after the second attachment, the Realme9 was given a /29 network as shown by the same app in the following image:
[image]

With this network configuration the (Realme9 UE attached to B210+ "srsRAN gnb uhd") was able to reach the srsUE attached to "srsRAN gnb zmq". So, now, here you have my question? Where/what/how is determining the configuration of the UE? Is it a matter of Open5Gs or is it a matter of how COTS UE is understanding messages from gNB? What do you think? 
It is difficult for me to explain/understand that when connecting with zmq, the srsUE gets a /24 address, but the smartphone a /30 (and *by chance* a /29).

Best regards to all community

pgawlowicz, 2024-06-11:
Hi @jorgebaranda,
Thank you for your report. Do you run the dockerized version of the Open5gs, or a manually installed one?
Could you try changing the following Open5gs config, and see if there is any effect?

```
smf:
...
   session:
-    - subnet: ${UE_GATEWAY_IP}
+    - subnet: 10.45.1.0/24
+      gateway: 10.45.1.1

upf:
...
   session:
-    - subnet: ${UE_GATEWAY_IP}
+    - subnet: 10.45.1.0/24
+      gateway: 10.45.1.1
```

jorgebaranda, 2024-06-18:
Hi Piotr,
Thanks a lot for your answer and considering to follow this thread. In my setup, I run Open5Gs (v2.6.4) distributed in different Kubernetes cluster by means of Helm charts. One k8s cluster is running all the core NFs unless the upf and the other K8s is running the UPF and the different gNBs (uhd and zmq) also as Helm charts.
Here, I attach you the config files of the SMF and UPF. Indeed they are specifying a /16, so I do not understand even for srsUE a /24 addr is given. Would it be a matter of the configuration of the smartphone(UE) itself? But the signalling between gNB and the UE (either it is smartphone or zmq should be the same, isn't it?)
Thanks for your help,
Jordi Baranda
[smf-config.txt](https://github.com/user-attachments/files/15882404/smf-config.txt)
[upf-config.txt](https://github.com/user-attachments/files/15882406/upf-config.txt)

jorgebaranda, 2024-06-20:
Hi @pgawlowicz,
Let me add additional information to this thread, which may be of help. I've been capturing some pcap in the gnb, amf, smf and upf entities running as k8s pods during the attachment process of a srsUE ZMQ client (please see attached files). I've been able to see the assigned IP address in these traces in the packet containing the "PDU session establishment accept" but I have not been able to see the "/" network assigned. Is this somehow specified in such messages? After that, I've also done an ifconfig in the srsUE client and I've seen it had a /24 assignment.  

Further information to understand the pcaps:
- GNB is running in a computer with IP 10.1.14.54 and it is accessible through a load balancer service with IP 10.0.242.101 --> The "PDU session establishment accept" is in packet 11 
- AMF is running in a computer with IP 10.1.14.55 and it is accessible through a load balancer service with IP 10.0.241.102 --> The "PDU session establishment accept" is in packet 168
- SMF is running in a computer with IP 10.1.14.55 and it is accessible through a load balancer service with IP 10.0.241.103 --> The "PDU session establishment accept" is in packet 116
- UPF is running in a computer with IP 10.1.14.54 and it is accessible through a load balancer service with IP 10.0.242.103 

[Pcaps_zmq_240619.zip](https://github.com/user-attachments/files/15908732/Pcaps_zmq_240619.zip)

Thanks for your help,

pgawlowicz, 2024-06-24:
Hi @herlesupreeth, could you comment here?

herlesupreeth, 2024-06-24:
@jorgebaranda 

>  I've been able to see the assigned IP address in these traces in the packet containing the "PDU session establishment accept" but I have not been able to see the "/" network assigned

The IP address range prefix is not conveyed in any of the NGAP messages. I am not sure about the mechanism in a smartphone but in case of srsUE a TUN interface is created so we need to manually adjust the prefix (/16 or /24) based on the subnet settings in SMF/UPF of 5GC

i.e. in UE configuration file you need to set as follows since your prefix is /16 in SMF configuration for internet DNN

```
ip_devname = tun_srsue
ip_netmask = 255.255.0.0
```

jorgebaranda, 2024-06-26:
Hi @pgawlowicz, @herlesupreeth,
You're right @herlesupreeth, in the UE configuration file supplied to srsUE you can control this. But this is not the case in a smartphone. 
@pgawlowicz, I realised that in the open5gs (v2.6.4), I couldn't apply the changes in the configuration that you mention. I used the setup suggested by srs, which is using open5gs (v2.7.0). Even, I performed the modifications you mention, this didn't cause any change in the behaviour of the smartphone and reflected a /30 network.  
Do you have any clue or idea how this could be changed in an Androd smartphone?

Thank you very much for your help

herlesupreeth, 2024-06-26:
> Even, I performed the modifications you mention, this didn't cause any change in the behaviour of the smartphone and reflected a /30 network.
> Do you have any clue or idea how this could be changed in an Androd smartphone?

@jorgebaranda I have used that app before and it shows incorrect prefix as far as I remember. The important thing to check (rather than prefix) is whether you are able to ping internet or not through that app.

jorgebaranda, 2024-06-26:
@herlesupreeth, there is internet connectivity because I can browse. However, I have two UEs in the network and I can not do a ping between them. This is the final goal, to have connectivity between UEs, because with the internet is already achieved.

pgawlowicz, 2024-06-26:
hi @jorgebaranda, do you see pings in core network, when you ping one UE from the other one?

jorgebaranda, 2024-06-27:
Hi @pgawlowicz, In my long post when I reopened this issue, I explained that by casuality, once, the smartphone got an /29 address so the other UE (using srs ue zmq) was within the network range and they could ping each other. 
_"In another experiment, by casuality, I left an "srsUE zmq" attached to the network and I attached the "Realme9", then I enter and leave flight mode and after the second attachment, the Realme9 was given a /29 network as shown by the same app in the following image"_

Repeating in a new experiment, the smartphone got the IP 10.46.0.6/30 and the srs ue zmq 10.46.0.7. When I make ping:
a) From srs ue zmq (10.46.0.7) to smarphone (10.46.0.6). I saw in the "PingTools app": Address error Broadcast ping not supported
b) From smartphone (10.46.0.6) to srs ue zmq (10.46.0.7), I saw the ping arriving to the UPF but they didn't reach the the UPF.
07:34:05.735368 IP 10.46.0.7 > 10.46.0.6: ICMP echo request, id 41201, seq 5, length 64
07:34:05.735389 IP 10.46.0.1 > 10.46.0.7: ICMP redirect 10.46.0.6 to host 10.46.0.6, length 92
07:34:05.735392 IP 10.46.0.7 > 10.46.0.6: ICMP echo request, id 41201, seq 5, length 64
07:34:06.725620 IP 10.46.0.7 > 10.46.0.6: ICMP echo request, id 41201, seq 6, length 64
07:34:06.725637 IP 10.46.0.1 > 10.46.0.7: ICMP redirect 10.46.0.6 to host 10.46.0.6, length 92
07:34:06.725640 IP 10.46.0.7 > 10.46.0.6: ICMP echo request, id 41201, seq 6, length 64
07:34:07.736808 IP 10.46.0.7 > 10.46.0.6: ICMP echo request, id 41201, seq 7, length 64
07:34:07.736821 IP 10.46.0.7 > 10.46.0.6: ICMP echo request, id 41201, seq 7, length 64

So this is why I suspect it is a matter of the networking stack at the smartphone.

pgawlowicz, 2024-06-27:
Hi @jorgebaranda, thanks for the explanation. 

If the smartphone gets IP 10.46.0.6/30, its broadcast address is 10.46.0.7 (and network address is 10.46.0.4). So I think, it receives a Ping from srsUE (10.46.0.7) and tries to reply, but it thinks the address is a broadcast address.

Could you try to allocate a different IP address for the srsUE (outside smartphone's network, e.g., 10.46.0.2) and try again?

jorgebaranda, 2024-06-27:
Hi all, 
A new update. Playing around with attaching and deattaching the "smartphone" and the "srsUE-zmq", I realised the following.
It seems the "PingTools" app presents a random netmask. I've seen /27, /28, /29, /30 (/30 is the most frequent). Then, if the srsUE-zmq is assigned the broadcast IP address showed in the "PingTools" app, it seems that ping is not possible. In an experiment, the smartphone has 10.46.0.18/30 and I could ping the "srsUE-zmq", which had the 10.46.0.21/24. 
I hope this is the reason.
Thanks for your help

pgawlowicz, 2024-08-06:
Hi @jorgebaranda, 
is it possible to set the same IP address with a different subnet using `ip` or `ifconfig` command?