srsran/srsRAN_Project #495: O-RAN cannot connect with the xApp

panitsasi, 2024-03-04:
## Issue Description ##
O-RAN NearRT-RIC and xApp

I followed all the steps but the expected output is not same as yours. 

## Setup Details ##
I run the core network by following these commands: 
cd ./srsRAN_Project/docker
docker compose up --build 5gc
[image]

I run the NEAR RT-RIC with ./flexric/build/examples/ric/nearRT-RIC
and the output is looking the same
[image]

Then i run the gNB and the output look like this: 
[image]
The output of the NEAR-RT-RIC after starting the gNB is: 
[image]

Then i start the UE and the output is: 

When i try to ping from UE to the core i am taking this error: 
[image]

When i run the default xApp i am taking these errors (the return values are 0): 

[image]

Can you help me debugging with these issues ?

yagoda, 2024-03-04:
Hi, you are missing a picture of the UE, did it attach well? have you set the network namespace in the ue.conf?

panitsasi, 2024-03-05:
The picture with the UE: 

[image]

Yes, i have set it.

pgawlowicz, 2024-03-06:
could you print routing table on your srsUE host? `route -n`

panitsasi, 2024-03-06:
When i execute this command : 
sudo ip netns exec ue1 ping -i 0.1 10.45.1.1
the output is correct. After executing the following commands : 
sudo ip ro add 10.45.0.0/16 via 10.53.1.2
route -n
The output that i get is: 

[image]

pgawlowicz, 2024-03-06:
it seems that you have two versions of Open5gs on the machine, i.e., dockerized version and one installed from source, right?

Check this section: https://docs.srsran.com/projects/project/en/latest/tutorials/source/flexric/source/index.html#nd-open5gs-instance-installed-manually

panitsasi, 2024-03-06:
Yes that's true. Instead of removing the dockerized version is there any way to run the 5G core network manually (if yes the commands are sudo systemctl start open5gs-hssd etc ...?) If i initiate the core network manually will i face the same errors ?

pgawlowicz, 2024-03-11:
Of course, you can run the Open5gs manually. For this purpose, please follow this tutorial: https://open5gs.org/open5gs/docs/guide/01-quickstart/

You will also need to update the default Open5GS configs, to match the gNB config (or vice versa). You can find some hints here: https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html#id1

pgawlowicz, 2024-04-11:
@panitsasi any update on this issue?

closed 2024-04-15 by pgawlowicz