srsran/srsRAN_Project #496: Cannot connect Over-the-air Setup

panitsasi, 2024-03-04:
Do you know why this error is popping up when i try to create the RF using a SDR (USRP B210)? 
[image]

pgawlowicz, 2024-03-04:
the external clock cannot be locked. If you are using GPSDO you need to put the antenna outside

pgawlowicz, 2024-03-04:
Then you should change the gnb config to use internal clock and time source. Note that the clock inside b200 is not very good and there might be issues when trying to connect a UE to the gnb, when running with b210 without external clock.

panitsasi, 2024-03-05:
I changed the gnb and ue configuration files and here is the output. 

The core network is running

[image]

The gnb integrated with the USRP is running

[image]

The configuration file of the gnb is this: 
[image]

While i try to connect a UE: 

[image]

Where the configuration file is: 

[image]

The UE cannot connect to the gNB. Is this happening because i do not use an external clock?

pgawlowicz, 2024-03-06:
could you run `uhd_find_devices` on the srsUE host PC?

panitsasi, 2024-03-06:
Yes, I can. 

[image]

robertfalkenberg, 2024-03-07:
If you run both, gNB and UE, on the same computer, you will also need two USRPs: one for gNB and one for UE.
(However, we recommend to run on separate computers)

pgawlowicz, 2024-04-15:
@panitsasi any update on this issue?

closed 2024-04-17 by pgawlowicz