srsran/srsRAN_Project #500: srsGNB ERROR: Failed to create SCTP gateway for ZMQ-based setup

hududed, 2024-03-07:
## Issue Description ##
srsGNB ERROR: Failed to create SCTP gateway for ZMQ-based setup

## Setup Details ##
Ubuntu 22.04.4 LTS
No UHD, single machine

## Expected Behavior ##
Something similar to the one for over-the-air
```
--== srsRAN gNB (commit 374200dee) ==--

Connecting to AMF on 10.53.1.2:38412
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200'
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz...
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 23.040000 MHz...
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Cell pci=1, bw=20 MHz, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz
```
## Actual Behaviour ##
❯ sudo gnb -c ./gnb_zmq.yaml             

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in executor blocking mode.

--== srsRAN gNB (commit bcf941b34) ==--

Connecting to AMF on 10.53.1.2:38412
Failed to connect SCTP socket to 10.53.1.2:38412. error="Connection timed out" timeout=1381ms
srsGNB ERROR: Failed to create SCTP gateway.

## Steps to reproduce the problem ##
Ran all the steps to install ZMQ, `sudo make install` for `srsRAN_4G` and the `srsRAN_Project`. Compiled with g++, c++14.
Compiled and ran open5GS with `docker compose up --build 5gc`
Downloaded `gnb_zmq.yaml` from docs and ran `sudo gnb -c ./gnb_zmq.yaml`

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

pgawlowicz, 2024-03-11:
Hi, Did you start a core network that the gNB can reach at `10.53.1.2:38412` ? For example open5gs?

hududed, 2024-03-11:
Yeah I assumed this was done with the `docker compose up --build 5gc`, I've only uncommented the 38412 port and changed the config yml:
```

version: "3.9"

services:
  5gc:
    container_name: open5gs_5gc
    build:
      context: open5gs
      target: open5gs
      args:
        OS_VERSION: "22.04"
        OPEN5GS_VERSION: "v2.6.1"
    env_file:
      - ${OPEN_5GS_ENV_FILE:-open5gs/open5gs.env}
    privileged: true
    ports:
      - "3000:3000/tcp"
      # Uncomment port to use the 5gc from outside the docker network
      - "38412:38412/sctp"
    command: 5gc -c open5gs-5gc.yml
    healthcheck:
      test: ["CMD-SHELL", "nc -z 127.0.0.20 7777"]
      interval: 3s
      timeout: 1s
      retries: 60
    networks:
      ran:
        ipv4_address: ${OPEN5GS_IP:-10.53.1.2}

  ...

configs:
  gnb_config.yml:
    file: ${GNB_CONFIG_PATH:-../configs/gnb_zmq.yml} # Path to your desired config file
```

pgawlowicz, 2024-03-11:
could you comment it again and rebuild the docker container?

hududed, 2024-03-11:
@pgawlowicz Hm, same issue

```
The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in executor blocking mode.

--== srsRAN gNB (commit bcf941b34) ==--

Connecting to AMF on 10.53.1.2:38412
Failed to connect SCTP socket to 10.53.1.2:38412. error="Connection timed out" timeout=1374ms
srsGNB ERROR: Failed to create SCTP gateway.
```

pgawlowicz, 2024-03-11:
No, it should be independent of build. Could you share the open5gs log output? is there anything?

hududed, 2024-03-11:
The docker build log? `docker logs open5gs_5gc` ?

pgawlowicz, 2024-03-11:
when you start open5gs docker with `docker compose up 5gc` and then you start `gnb`. Is there any info about the incoming gnb connection in the open5gs console log?

hududed, 2024-03-11:
```
nc: connect to 127.0.0.1 port 27017 (tcp) failed: Connection refused
waiting for mongodb

> open5gs@2.6.1 dev
> node server/index.js

Connection to 127.0.0.1 27017 port [tcp/*] succeeded!
> Using external babel configuration
> Location: "/open5gs/webui/.babelrc"
 DONE  Compiled successfully in 1792ms5:00:27 PM

Mongoose: accounts.count({}, {})
> Ready on http://31cbeee603c5:3000
Mongoose: accounts.findOne({ '$or': [ { username: 'admin' } ] }, { projection: { hash: 0, salt: 0 } })
Mongoose: subscribers.createIndex({ imsi: 1 }, { unique: true, background: true })
Mongoose: accounts.createIndex({ username: 1 }, { unique: true, background: true })
Mongoose: accounts.insertOne({ roles: [ 'admin' ], _id: new ObjectId("65ef38acebec5ee8e16076f6"), username: 'admin', salt: '377f36be2a117410fc00517117e28883258dda57ced12998a0c94dbe0fd232a1', hash: '0de40d493cbd20de4b71147618b1831e9ec514aad1ce1d1d4b41c2571d6480e0f933966c7808c41bf9cf5d958d4fd85d56da51cbac3bf6f08c50ed3b24e990b309bc9dd4d3b5df33e880d0f84810176c795d2ce61f55548ef8ecc932af6ea63bf7991292d51dbdbe0c46f41800ad33d8d11b91d428170df4a178c2c365de8644c5e823e482def19f3e5a39c09188fd299884d5044a30365519c5892b5acbdd04659505f70daad3310e43ffd64d173dea3ae7cb7da1cc51ee67fd09c02223de4c9c4290d1a39005623aa352de749d8de308710c1a3b4d485393df95379cc8ccba7f446ffe1f9d0b20d08d0f2ac6cb58b8a1bb94fa2f64c5fce402a3ac63b8def49f8f44ea2c374065ce63d030565a8e8d64fb90c9569545d7d893accb32bceedc4224b5fed536f3b621f213437cef6a5cea131011111ee1fc1ca431ee7918170d19fa97ccee688ee49d66ef5553062f700fc9e86a4f72fdef90d5412076945c81151e9dd8930e1a46118dbcd8bab24a1f51c9b6e5ddf4ccbb28d652f7dd198ff1f5b308da5b210b8e308b5be4b06da9d0d1ae4833f9928b336e22912f573011651e38bc482139a57169657b6e7e5c3478b93ccba37ba9f1a3ec821c877e1d0de6afd5f3ffb188e3582fc2d308e73a1b2ec5ce18cf3c57726ea7fa06be925ac70c1e479bd3ddc5bc0c64e531b90a56d505bfd1b24346b1d904d0274426b5962915', __v: 0}, { session: null })
SUBSCRIBER_DB=001010123456780,00112233445566778899aabbccddeeff,opc,63bfa50ee6523365ff14c1f45f88737d,8000,9,10.45.1.2
Reading subscriber data from cmd.
Added subscriber with Inserted ID : 65ef38ad7068246dbf37c147
65ef38ad7068246dbf37c147
Open5GS daemon v2.6.1

03/11 17:00:29.828: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:126)
Open5GS daemon v2.6.1

03/11 17:00:29.839: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:126)
03/11 17:00:29.842: [sbi] INFO: nghttp2_server() [http://127.0.0.10]:7777 (../lib/sbi/nghttp2-server.c:395)
03/11 17:00:29.842: [app] INFO: NRF initialize...done (../src/nrf/app.c:31)
Open5GS daemon v2.6.1

03/11 17:00:29.887: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:126)
03/11 17:00:29.890: [sbi] INFO: nghttp2_server() [http://127.0.1.10]:7777 (../lib/sbi/nghttp2-server.c:395)
03/11 17:00:29.890: [app] INFO: SCP initialize...done (../src/scp/app.c:31)
03/11 17:00:29.891: [nrf] INFO: [dd92cd74-dfc8-41ee-8c9b-8913dc450259] NF registered [Heartbeat:10s] (../src/nrf/nf-sm.c:190)
03/11 17:00:29.891: [sbi] INFO: [dd92cd74-dfc8-41ee-8c9b-8913dc450259] NF registered [Heartbeat:10s] (../lib/sbi/nf-sm.c:214)
Open5GS daemon v2.6.1

03/11 17:00:29.940: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:126)
03/11 17:00:29.952: [pfcp] INFO: pfcp_server() [127.0.0.7]:8805 (../lib/pfcp/path.c:30)
03/11 17:00:29.952: [pfcp] INFO: ogs_pfcp_connect() [127.0.0.4]:8805 (../lib/pfcp/path.c:61)
03/11 17:00:29.952: [gtp] INFO: gtp_server() [10.53.1.2]:2152 (../lib/gtp/path.c:30)
03/11 17:00:29.952: [app] INFO: UPF initialize...done (../src/upf/app.c:31)
Open5GS daemon v2.6.1

03/11 17:00:29.999: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:126)
Open5GS daemon v2.6.1

03/11 17:00:30.045: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:126)
03/11 17:00:30.047: [gtp] INFO: gtp_server() [127.0.0.4]:2123 (../lib/gtp/path.c:30)
03/11 17:00:30.047: [gtp] INFO: gtp_server() [127.0.0.4]:2152 (../lib/gtp/path.c:30)
03/11 17:00:30.047: [pfcp] INFO: pfcp_server() [127.0.0.4]:8805 (../lib/pfcp/path.c:30)
03/11 17:00:30.047: [pfcp] INFO: ogs_pfcp_connect() [127.0.0.7]:8805 (../lib/pfcp/path.c:61)
03/11 17:00:30.047: [sbi] INFO: NF Service [nsmf-pdusession] (../lib/sbi/context.c:1436)
03/11 17:00:30.048: [upf] INFO: PFCP associated [127.0.0.4]:8805 (../src/upf/pfcp-sm.c:173)
03/11 17:00:30.048: [sbi] INFO: nghttp2_server() [http://127.0.0.4]:7777 (../lib/sbi/nghttp2-server.c:395)
03/11 17:00:30.048: [app] INFO: SMF initialize...done (../src/smf/app.c:31)
03/11 17:00:30.048: [smf] INFO: PFCP associated [127.0.0.7]:8805 (../src/smf/pfcp-sm.c:174)
03/11 17:00:30.049: [nrf] INFO: [dda4dd0c-dfc8-41ee-9ae2-c320ffc3fc18] NF registered [Heartbeat:10s] (../src/nrf/nf-sm.c:190)
03/11 17:00:30.049: [sbi] INFO: [dda4dd0c-dfc8-41ee-9ae2-c320ffc3fc18] (NRF-notify) NF registered (../lib/sbi/nnrf-handler.c:632)
03/11 17:00:30.049: [sbi] INFO: [dda4dd0c-dfc8-41ee-9ae2-c320ffc3fc18] (NRF-notify) NF Profile updated (../lib/sbi/nnrf-handler.c:642)
03/11 17:00:30.050: [sbi] INFO: [dda4dd0c-dfc8-41ee-9ae2-c320ffc3fc18] NF registered [Heartbeat:10s] (../lib/sbi/nf-sm.c:214)
03/11 17:00:30.057: [sbi] INFO: NF Service [namf-comm] (../lib/sbi/context.c:1436)
03/11 17:00:30.057: [sbi] INFO: nghttp2_server() [http://127.0.0.5]:7777 (../lib/sbi/nghttp2-server.c:395)
03/11 17:00:30.058: [amf] INFO: ngap_server() [10.53.1.2]:38412 (../src/amf/ngap-sctp.c:61)
03/11 17:00:30.058: [sctp] INFO: AMF initialize...done (../src/amf/app.c:33)
[... cut ...]
03/11 17:00:30.365: [sbi] INFO: [ddda8f74-dfc8-41ee-bb54-932a5d5b1bb9] NF registered [Heartbeat:10s] (../lib/sbi/nf-sm.c:214)
03/11 17:00:32.456: [smf] WARNING: PFCP[REQ] has already been associated [127.0.0.7]:8805 (../src/smf/pfcp-sm.c:217)
03/11 17:00:32.456: [upf] WARNING: PFCP[RSP] has already been associated [127.0.0.4]:8805 (../src/upf/pfcp-sm.c:213)
 DONE  Compiled successfully in 6145ms5:00:36 PM

 WAIT  Compiling...5:00:37 PM

 DONE  Compiled successfully in 174ms5:00:37 PM
```

Doesn't seem like it

pgawlowicz, 2024-03-11:
Did you install open5gs from source as well?
Could you print your routing table `route -n`?

hududed, 2024-03-11:
No I didnt, just used the docker for that 

```
❯ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.0.0.1        0.0.0.0         UG    20600  0        0 wlp69s0
10.0.0.0        0.0.0.0         255.255.255.0   U     600    0        0 wlp69s0
10.7.0.0        0.0.0.0         255.255.0.0     U     0      0        0 nordtun
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-6fe3e2af8a94
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 wlp69s0
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-10a6c38bbec6
```

pgawlowicz, 2024-03-11:
No the VPN doest no matter. The gNB tries to reach 10.53.1.2, and there is a route for that. 
BTW. do you run the gnb on the same machine as open5gs docker?

hududed, 2024-03-11:
Yes, single machine, different terminals

hududed, 2024-03-12:
```
❯ ping -c 4 10.53.1.2
PING 10.53.1.2 (10.53.1.2) 56(84) bytes of data.
From 10.53.1.1 icmp_seq=1 Destination Host Unreachable
From 10.53.1.1 icmp_seq=2 Destination Host Unreachable
From 10.53.1.1 icmp_seq=3 Destination Host Unreachable
From 10.53.1.1 icmp_seq=4 Destination Host Unreachable

--- 10.53.1.2 ping statistics ---
4 packets transmitted, 0 received, +4 errors, 100% packet loss, time 3049ms
pipe 4
```
No that didnt work

pgawlowicz, 2024-03-12:
Then there is something wrong with the docker network. Could you remove it and start docker compose again to add it?

hududed, 2024-03-12:
Hm so I ran:
```
❯ docker compose down
❯ docker network prune
❯ docker compose up --build 5gc
❯ docker network ls
NETWORK ID     NAME             DRIVER    SCOPE
b060779592ec   bridge           bridge    local
7c7dbc02bd28   docker_metrics   bridge    local
1db9647f07ce   docker_ran       bridge    local
f1f160724b7f   host             host      local
c4a0dbcac33f   none             null      local
❯ docker network inspect docker_ran
[
    {
        "Name": "docker_ran",
        "Id": "1db9647f07ce6349dafa02898b39f215d2859619f79186b2e0aebeeaea87f063",
        "Created": "2024-03-12T15:59:39.472409944Z",
        "Scope": "local",
        "Driver": "bridge",
        "EnableIPv6": false,
        "IPAM": {
            "Driver": "default",
            "Options": null,
            "Config": [
                {
                    "Subnet": "10.53.1.0/24"
                }
            ]
        },
        "Internal": false,
        "Attachable": false,
        "Ingress": false,
        "ConfigFrom": {
            "Network": ""
        },
        "ConfigOnly": false,
        "Containers": {
            "f6378553dfc94c125181d1ea5548bdfe0c3e81a9ef02d0b9af619f546d73db4f": {
                "Name": "open5gs_5gc",
                "EndpointID": "5513fa4462a785586495b24f8e1670940b616db7d9325ed40e8f447a5b12a9ea",
                "MacAddress": "02:42:0a:35:01:02",
                "IPv4Address": "10.53.1.2/24",
                "IPv6Address": ""
            }
        },
        "Options": {},
        "Labels": {
            "com.docker.compose.network": "ran",
            "com.docker.compose.project": "docker",
            "com.docker.compose.version": "2.24.6"
        }
    }
]
```
But ping still doesn't work

pgawlowicz, 2024-03-13:
hmm, strange. Coul you go to the docker container and ping your host?
```
docker exec -it open5gs_5gc bash
then inside docker:
ping 10.53.1.1
```

hududed, 2024-03-13:
Yes works inside the docker
```
❯ docker exec -it open5gs_5gc bash
root@15e61fd8c4b3:/open5gs# ping 10.53.1.1
PING 10.53.1.1 (10.53.1.1) 56(84) bytes of data.
64 bytes from 10.53.1.1: icmp_seq=1 ttl=64 time=0.112 ms
64 bytes from 10.53.1.1: icmp_seq=2 ttl=64 time=0.040 ms
64 bytes from 10.53.1.1: icmp_seq=3 ttl=64 time=0.041 ms
^C
--- 10.53.1.1 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2051ms
rtt min/avg/max/mdev = 0.040/0.064/0.112/0.033 ms
```

pgawlowicz, 2024-03-14:
yes, you can install manually, you need to follow this tutorial  https://open5gs.org/open5gs/docs/guide/01-quickstart/

And then update the config to match the PLMN, TAC, etc to those used in srsGNB config.

hududed, 2024-03-14:
Hm that seems like a lot more pain points for reproducing. srsGNB (`gnb` in container)? That is not run using docker container with 5gc is it? Is there no other way to debug the docker?

hududed, 2024-03-14:
I also noticed that when docker open5gs is running the port is udp not sctp
```
❯ netstat -tuln | grep 38412
udp6       0      0 :::38412                :::*
```
What could cause it to be exposed as udp?

pgawlowicz, 2024-03-15:
Could you try to start srsgnb inside docker?

Please set a path to a proper config file here:
https://github.com/srsran/srsRAN_Project/blob/main/docker/docker-compose.yml#L144
Then run:
```
docker compose up gnb
```
Please check details here:
https://github.com/srsran/srsRAN_Project/blob/main/docker/docker-compose.yml#L50

hududed, 2024-03-15:
```
configs:
  gnb_config.yml:
    file: ${GNB_CONFIG_PATH:-../configs/gnb_zmq.yaml} # Path to your desired config file
```

```
❯ docker compose up gnb
[+] Running 2/0
 ✔ Container open5gs_5gc  Running                                                                                                                                                 0.0s 
 ✔ Container srsran_gnb   Created                                                                                                                                                 0.0s 
Attaching to srsran_gnb
srsran_gnb  | 
srsran_gnb  | The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
srsran_gnb  | Lower PHY in executor blocking mode.
srsran_gnb  | 
srsran_gnb  | --== srsRAN gNB (commit ) ==--
srsran_gnb  | 
srsran_gnb  | Failed to bind SCTP socket to 10.53.1.3:0. Cannot assign requested address
srsran_gnb  | srsGNB ERROR: Failed to create SCTP gateway.
srsran_gnb  | 
srsran_gnb exited with code 1
```
Will that work with gnb_zmq ?
```
amf:
  addr: 10.53.1.2 # The address or hostname of the AMF.
  bind_addr: 10.53.1.1 # A local IP that the gNB binds to for traffic from the AMF.
```

pgawlowicz, 2024-03-18:
Yes, zmq is only for RF channel emulation, it does not affect this part. 
Why does the open5gs docker try to bind socket to 10.53.1.3? It should use 10.53.1.2.

hududed, 2024-03-18:
@pgawlowicz I didnt really change much at all from the docker compose file.
```
# docker-compose 
gnb:
  ...
  # Command to run into the final container
  command: gnb -c /gnb_config.yml amf --addr ${OPEN5GS_IP:-10.53.1.2} --bind_addr ${GNB_IP:-10.53.1.3}
```
If anything I changed the bind_addr in the zmq yml from 10.53.1.1 to 10.53.1.3:
```
# gnb_zmq.yml
amf:
  addr: 10.53.1.2 # The address or hostname of the AMF.
  bind_addr: 10.53.1.3 # A local IP that the gNB binds to for traffic from the AMF.
```
what is the appropriate OPEN5GS_IP and GNB_IP here?

pgawlowicz, 2024-03-20:
Hi, I have just checked, and it is enough to simply run:
```
docker compose up gnb
```
It starts both open5gs and gnb containers, and the gnb can connect.
I use this config file: 
[gnb_zmq.yaml.txt](https://github.com/srsran/srsRAN_Project/files/14664045/gnb_zmq.yaml.txt)

The AMF IP and bind address are overwritten with the command line arguments in docker-compose file.
The setup runs out of the box.

hududed, 2024-03-21:
Ok the config file did not help either, getting the same srsGNB error. That is really strange.

By the way, following the docker setup [here](https://github.com/herlesupreeth/docker_open5gs) works for me. So I'll probably spend some time comparing the difference why one works and the other doesnt.

So if gnb is run via docker, do you know of if the [testing](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zmq-testing) works the same way?

pgawlowicz, 2024-03-21:
If the IP allocated to UEs by the core network are in the same range (i.e.,10.45.0.0/16) and the core network (open5gs) runs at 10.53.1.2, the testing commands should be the same. 
Your host PC has to know how to reach the UE IP through core network. You can always try to ping UE from the open5gs container, then you do not need a route to UEs for your host pc.

achkonsa, 2024-03-24:
I just delete the srsRAN_Project folder and remove the libzmq installed from apt, then install the libzmq from source and clone again...so cruel but it did work in my single machine... AND I cmake with this command: cmake ../ -DENABLE_EXPORT=ON -DENABLE_ZEROMQ=ON.....I wish it can be helpful...  :)

closed 2024-04-02 by pgawlowicz