srsran/srsRAN_Project #502: srsGNB Docker Image will not compiled with DPDK enabled

IbrahimBabiker87, 2024-03-08:
## Issue Description ##
Docker Image will not compiled with DPDK enabled

```
329.8 [ 80%] Linking CXX executable pusch_processor_validator_test
330.0 [ 80%] Built target pusch_processor_validator_test
330.0 [ 80%] Linking CXX executable pdsch_processor_unittest
330.1 [ 80%] Built target pdsch_processor_unittest
330.3 [ 80%] Linking CXX executable rrc_ue_test
330.5 [ 80%] Built target rrc_ue_test
331.0 [ 80%] Linking CXX executable pusch_processor_unittest
331.0 [ 80%] Built target pusch_processor_unittest
331.0 make: *** [Makefile:146: all] Error 2
------
failed to solve: process "/bin/sh -c /src/.gitlab/ci/builders/builder.sh  /src" did not complete successfully: exit code: 2
```

## Setup Details ##
last SRSRAN build 342df75

## Expected Behavior ##
to work fine

## Actual Behaviour ##
It gives bellow error in make:

```
329.8 [ 80%] Linking CXX executable pusch_processor_validator_test
330.0 [ 80%] Built target pusch_processor_validator_test
330.0 [ 80%] Linking CXX executable pdsch_processor_unittest
330.1 [ 80%] Built target pdsch_processor_unittest
330.3 [ 80%] Linking CXX executable rrc_ue_test
330.5 [ 80%] Built target rrc_ue_test
331.0 [ 80%] Linking CXX executable pusch_processor_unittest
331.0 [ 80%] Built target pusch_processor_unittest
331.0 make: *** [Makefile:146: all] Error 2
------
failed to solve: process "/bin/sh -c /src/.gitlab/ci/builders/builder.sh  /src" did not complete successfully: exit code: 2
```

## Steps to reproduce the problem ##
trying to compose up docker image for srsgnb with DPDK enabled.

## Additional Information ##

tried to pass -DENABLE_DPDK=ON to builder script  "/bin/sh -c /src/.gitlab/ci/builders/builder.sh  but it doesnt work.

```
329.8 [ 80%] Linking CXX executable pusch_processor_validator_test
330.0 [ 80%] Built target pusch_processor_validator_test
330.0 [ 80%] Linking CXX executable pdsch_processor_unittest
330.1 [ 80%] Built target pdsch_processor_unittest
330.3 [ 80%] Linking CXX executable rrc_ue_test
330.5 [ 80%] Built target rrc_ue_test
331.0 [ 80%] Linking CXX executable pusch_processor_unittest
331.0 [ 80%] Built target pusch_processor_unittest
331.0 make: *** [Makefile:146: all] Error 2
------
failed to solve: process "/bin/sh -c /src/.gitlab/ci/builders/builder.sh -DENABLE_DPDK=ON  /src" did not complete successfully: exit code: 2
```

tried to enable DPDK from CMakeLists.txt but it doesnt work

   
```
    BUILD_TYPE: "" # Empty for cmake default
    ASSERT_LEVEL: "" # Empty for cmake default
    ENABLE_EXPORT: "" # Empty for cmake default
    ENABLE_FFTW: "" # Empty for cmake default
    ENABLE_UHD: "" # Empty for cmake default
    ENABLE_DPDK: "" # Empty for cmake default
    ENABLE_ZEROMQ: "True" # Empty for cmake default
    ENABLE_ASAN: "" # Empty for cmake default
    ENABLE_TSAN: "" # Empty for cmake default
    ENABLE_GCOV: "" # Empty for cmake default
    ENABLE_WERROR: "" # Empty for cmake default

```

pgawlowicz, 2024-06-20:
@IbrahimBabiker87 did you try with the newest release?

IbrahimBabiker87, 2024-06-21:
@pgawlowicz 
I have edited the builder.sh  script and worked fine for me by then, didn't test with new releases since.

thank you very much