srsran/srsRAN_Project #504: Missed incoming User-Plane PRACH messages

ehitimum, 2024-03-12:
## Issue Description ##
After the `2024-03-12T05:36:04.692031 [GNB     ] [I] Radio Unit started successfully`, I get these two 

```
2024-03-12T05:36:04.693232 [FAPI    ] [W] [  477.11] Real-time failure in FAPI: Received late DL_TTI.request from slot 477.10
2024-03-12T05:36:04.693520 [SCHED   ] [I] [  477.12] Processed slot events: ErrorIndication slot=477.10
```
and then I started getting these error untill I stop my gnb.
```
2024-03-12T05:36:04.706010 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '477.17' and sector#0
2024-03-12T05:36:04.707012 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '477.19' and sector#0
2024-03-12T05:36:04.711014 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '478.7' and sector#0
2024-03-12T05:36:04.712014 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '478.9' and sector#0
2024-03-12T05:36:04.716017 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '478.17' and sector#0
2024-03-12T05:36:04.717008 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '478.19' and sector#0
2024-03-12T05:36:04.721015 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '479.7' and sector#0
2024-03-12T05:36:04.722011 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '479.9' and sector#0
2024-03-12T05:36:04.726011 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '479.17' and sector#0
2024-03-12T05:36:04.727013 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '479.19' and sector#0
2024-03-12T05:36:04.731020 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '480.7' and sector#0
2024-03-12T05:36:04.732009 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '480.9' and sector#0
2024-03-12T05:36:04.736008 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '480.17' and sector#0
```

## Setup Details ##
This is my complete log file, 
[gnb.log](https://github.com/srsran/srsRAN_Project/files/14567804/gnb.log)

I have a grand master that is connected with cu/du machine and ru machine with a front haul switch. I have tried many different prach configuration. Currently prach configuration index is 165.

This is the gnb configuration file,  
```
# This example configuration outlines how to configure the srsRAN Project CU/DU to use an O-RU and split 7.2. This is specifically for use
# with the Benetel R550 RU. This config will create a single TDD MIMO cell transmitting in band 78, with 100 MHz bandwidth and 30 kHz sub-carrier-spacing.
# The parameters used to configure the RU are found in the `ru_ofh` sub-section. This configuration makes used of the OFH Lib from SRS to enable split 7.2.

amf:
  addr: 127.0.0.5                                               # The address or hostname of the AMF.
  bind_addr: 127.0.0.2                                            # A local IP that the gNB binds to for traffic from the AMF.

ru_ofh:
  ru_bandwidth_MHz: 50                                           # RU instantaneous bandwidth.
  t1a_max_cp_dl: 500                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 250                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 465                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 250                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 250                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 80                                                  # Minimum T1a on User-Plane in microseconds.
  ta4_max: 500                                                    # Maximum Ta4 on User-Plane in microseconds.
  ta4_min: 25                                                     # Minimum Ta4 on User-Plane in microseconds.
  is_prach_cp_enabled: false                                      # Configures if Control-Plane messages should be used to receive PRACH messages.
  #is_prach_cp_enabled: true                                      # Configures if Control-Plane messages>
  compr_method_ul: bfp                                            # Uplink compression method.
  compr_bitwidth_ul: 8                                            # Uplink IQ samples bitwidth after compression.
  compr_method_dl: bfp                                            # Downlink compression method.
  compr_bitwidth_dl: 8                                            # Downlink IQ samples bitwidth after compression.
  compr_method_prach: bfp                                         # PRACH compression method.
  compr_bitwidth_prach: 8                                         # PRACH IQ samples bitwidth after compression.
  enable_ul_static_compr_hdr: true                                # Configures if the compression header is present for uplink User-Plane messages (false) or not present (true).
  enable_dl_static_compr_hdr: true                                # Configures if the compression header is present for downlink User-Plane messages (false) or not present (true).
  iq_scaling: 0.10                                                # IQ samples scaling factor applied before compression, should be a positive value smaller than 10.
  cells:
    - network_interface: enp6s0f1                                 # Ethernet interface name used to communicate with the RU.
      ru_mac_addr: 0C:83:CC:0F:E9:03                              #0C:83:CC:0F:E9:04    # RU MAC address.
      du_mac_addr: 98:b7:85:1e:dc:d6                              # DU MAC address.
      vlan_tag: 564                                                 # VLAN tag value.
      prach_port_id: [4]                                          # PRACH eAxC port value.
      dl_port_id: [0, 1]                                          # Downlink eAxC port values.
      ul_port_id: [0, 1]                                          # Uplink eAxC port values.

[... 79 lines cut ...]
2024-03-06T07:15:12.067812 [zmq:tx:0:0] [I] Waiting for request.
2024-03-06T07:15:12.067849 [zmq:rx:0:0] [I] Waiting for data.
2024-03-06T07:15:12.067897 [zmq:rx:0:0] [I] Waiting for reading samples. Completed 0 of 11520 samples.
2024-03-06T07:15:13.067814 [zmq:tx:0:0] [I] Waiting for request.
2024-03-06T07:15:13.067850 [zmq:rx:0:0] [I] Waiting for data.
2024-03-06T07:15:13.067903 [zmq:rx:0:0] [I] Waiting for reading samples. Completed 0 of 11520 samples.
2024-03-06T07:15:13.769620 [GNB     ] [I] Stopping Radio Unit...

```
## Actual Behaviour ##
Instead what I am getting is this:
```
2024-03-12T05:36:04.706010 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '477.17' and sector#0
2024-03-12T05:36:04.707012 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '477.19' and sector#0
2024-03-12T05:36:04.711014 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '478.7' and sector#0
2024-03-12T05:36:04.712014 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '478.9' and sector#0
```

## Additional Information ##
I have a slight understanding that this might have something to do with time synchronization since I still didn't successfully synchronize the time yet because my GM has issues. So I want expert opinion as why I am getting these including these two,
```
2024-03-12T05:36:04.693232 [FAPI    ] [W] [  477.11] Real-time failure in FAPI: Received late DL_TTI.request from slot 477.10
2024-03-12T05:36:04.693520 [SCHED   ] [I] [  477.12] Processed slot events: ErrorIndication slot=477.10
```
Thank you.

aibtw, 2024-03-18:
I am no expert but I have an addition that might help someone else to understand the problem better.
I noticed that even when I was able to get time sync and **connected UE to RU successfully**, I still got `Missed incoming User-Plane PRACH` very frequently. 

So this makes me think it might be something else other than timing?

ehitimum, 2024-04-21:
> I am no expert but I have an addition that might help someone else to understand the problem better. I noticed that even when I was able to get time sync and **connected UE to RU successfully**, I [...]

Is it possible for you to provide your .log result and also srsran config file so that I can compare it with mine. Also I myself wanted to connect **UE** with **RU** so it will be helpful.

Thank You for your feedback.

andrepuschmann, 2024-04-22:
What RU are you using?

closed 2024-10-16 by pgawlowicz