srsran/srsRAN_Project #509: Error: Buffer pool is empty

uptools, 2024-03-14:
## Issue Description ##
  A few seconds after srsenb we get the following errors (repeatedly):
  **Error - buffer pool is empty**
  
  ## Setup Details ##
  srsepc is running on another computer
  srsenb is running on a 32 bit arm/neon with 2 cores and 1gb ram, and it is correctly initializing the Soapy driver and connecting to sdr with only 1 port. srsenb is configured for only 6 prb's.
  srsenb is compiled in "Release" version
  
  We found that the error comes when srsenb is not deallocating and returning buffers fast enough to the buffer pool.
  We tried recompiling with SRSRAN_MAX_PORTS set to 1 to decrease mem usage, but the same error occurs.
  
  Is this caused by slow processing? or by not having enough memory? Or srsenb cannot run on arm32/neon ?

pgawlowicz, 2024-03-15:
do you work with the srsENB from the srsRAN-4G project?
If yes, you might try to reduce `nof_prealloc_ues` in the enb.conf file. 
https://github.com/srsran/srsRAN_4G/blob/master/srsenb/enb.conf.example#L451

How many UEs do you want to serve? what is the bandwidth?

uptools, 2024-03-15:
We worked it from srsENB_4G project.
We tried reducing nof_prealloc_ues from 8 to 2. and the same error arises.
We also tried reducing nof_phy_threads from 3 to 1 and the error stays the same.

The bandwidth is only 180KHz x 6 = 1080KHz (as we are using 6 prb's).

pgawlowicz, 2024-03-15:
could you check the memory load? with htop?

uptools, 2024-03-19:
The "buffer pool is empty" error starts when the srsenb allocation reaches approximately 410MB, even though there is still available mem.

[image]
[image]

pgawlowicz, 2024-03-19:
@frankist could you comment here?

pgawlowicz, 2024-04-11:
@uptools any update on this issue?

closed 2024-04-17 by pgawlowicz