srsran/srsRAN_Project #510: UE and srsGNB cannot connect together

xdnewlun1, 2024-03-15:
## Issue Description ##
I have setup Open5GS and the srsGNB on Ubuntu 22.04 Desktop. The COTS UE I am trying to use is the CrossCall Core-Z5 which I have successfully used on other 5G projects. When connecting the 5G UE to the GNB, the UE will attempt to connect to the PLMN 901 70 5G for about 3 minutes then will say connection unavailable. In packet captures there are no packets from the UE trying to register with the Core. (See Packet Capture below)

## Setup Details ##
Ubuntu 22.04 Desktop
32 Core - 128GB RAM

Ettus Research B210 w/ Leo Bodnar external gps clock
UHD v4.6.0.0

Open5GS v2.7.0
srsRAN 5G gNB v23.10.1 (bcf941b34)

## Expected Behavior ##
When the PLMN is discovered by the UE, it should be able to connect to the RAN and show packets in wireshark attempting to register with the Core.

## Actual Behaviour ##
The UE attempts to connect to the RAN but eventually fails out with no packets shown in WireShark.

## Steps to reproduce the problem ##
See Configuration Files and Packet Captures in zip file, GitHub was being weird about uploading multiple files.
[troubleshooting-documents.zip](https://github.com/srsran/srsRAN_Project/files/14618935/troubleshooting-documents.zip)

pgawlowicz, 2024-03-20:
Hi, Could you collect gnb logs with increased logging level?

xdnewlun1, 2024-03-20:
Here is the gnb_log with debug logging level.
File was too big to upload, so i had to compress it first.
[gnb_log.zip](https://github.com/srsran/srsRAN_Project/files/14669643/gnb_log.zip)

pgawlowicz, 2024-03-21:
could you try to reduce rx_gain at gnb?

xdnewlun1, 2024-03-22:
I can test this on Monday, and will let you know how it goes.

xdnewlun1, 2024-03-27:
I tested it today and at a level of 30, 20, and 10 there was still no show of PLMN on the UE, or any packets in Wireshark showing an attempt to connect from the UE.

pgawlowicz, 2024-04-02:
could you share logs for each rx_gain config?

desimetallica, 2024-04-08:
I'm quite interested, I've the same smartphone and I can't even find the network... Do you have rooted the phone and installed the Network Signal Guru as indicated in the [guide](https://docs.srsran.com/projects/4g/en/latest/app_notes/source/5g_sa_COTS/source/index.html?highlight=cots#network-signal-guru) ?

flenz00, 2024-05-03:
Same issue here

xdnewlun1, 2024-05-03:
I did not root the device as asked by @desimetallica. I did however recently get a Google Pixel 6 connected using the following gnb config. I had to change to a lower bang and band 71 was working with most of the devices i checked.
[gnb_config_as_a_pdf.pdf](https://github.com/srsran/srsRAN_Project/files/15202041/gnb_config_as_a_pdf.pdf)

flenz00, 2024-05-03:
Thanks, but with those values, the crosscall still didn’t work right?

xdnewlun1, 2024-05-03:
Yeah, which is why I am still testing to try and figure out how to get the crosscall working.

desimetallica, 2024-05-03:
we are trying to do the same. We have also an Ericsson gnb over our heads (dot product from ericsson for indoor coverage), and we can find it in "5gforce" application, the network actually appears with its NRARFC MCC and MNC. On the contrary for our deployment with srsRAN_project we can't see the network. No matter what frequencies or bands we are using.

xdnewlun1, 2024-05-03:
Yeah, I am running into the same problem. I have a deployment of a proprietary 5G that works on band n77, but when i use the exact smae radio sepcs on the srsRAN I get nothing on the phone.

desimetallica, 2024-05-21:
Hello @xdnewlun1, I would like to report you that my configuration with crosscall z5 is working now. We can report that also our samsung galaxy S22 plus is working with this deploy.

- Crosscall Z5 build: SKQ1.220303.001 
- APN: apn: internet, name: internet, protocol: ipv4, apn type: default,mms,supl,dun,hipri,fota,ims,cbs,xcap, NoAuth, mnc: 01 mcc: 001
- sim: sysmocom sysmoISIM-SJA2 with configured plmn 00101 (check the configuration of routing indicator and other stuff at [link](https://downloads.osmocom.org/docs/pysim/master/html/suci-tutorial.html).)
- The chart can be found in the following GitHub repository: [openverso-charts](https://github.com/Gradiant/openverso-charts). 
This charts uses Helm to deploy 5GS into kubernetes cluster, release of Helm Chart 2.2.2 and release of Open5GS 2.7.0
- SrsRAN: release 24.04
- We use SDR: x310 with UBX-160 daughterboard and we use externl clocksource from OctoClock-G CDA-2990.
- UHD version: 4.3.0
- Force5gonly or other applications are not required/mandatory.
- Our SDR is connected with PC with 10Gbps ethernet link and we have more than a decent CPU and RAM setup (Intel(R) Xeon(R) w9-3475X + 128GB ram).
- Please try to find good values of gain with srsUE and next try with smartphone. 

SrsRAN configfile:

```
# This example configuration outlines how to configure the srsRAN Project gNB to create a single FDD cell
# transmitting in band 3, with 20 MHz bandwidth and 15 kHz sub-carrier-spacing. A USRP B200 is configured 
# as the RF frontend using split 8, with a Leo Bodnar GPSDO as an external reference clock. 

# This particular configuration is intended to be used with srsUE. Note the `pdcch` parameters set in the 
# `cell_cfg` section. These are set to match the capabilities of srsUE. 

amf:
  addr: 192.168.12.130 #172.22.0.10 #127.0.0.5 #127.0.1.100                                               # The address or hostname of the AMF.
  bind_addr: 192.168.7.227 #127.0.0.5                                            # A local IP that the gNB binds to for>
#  tai: 

ru_sdr:
  device_driver: uhd                                              # The RF driver name.
#  device_args: type=b200,num_recv_frames=64,num_send_frames=64    # Optionally pass arguments to the selected RF driver.
  device_args: addr=192.168.40.2,type=x300,send_frame_size=8000,recv_frame_size=8000,master_clock_rate=184.32e6    # Optionally pass arguments to the selected RF driver.
    #device_args: type=x300,master_clock_rate=184.32e6,send_frame_size=8000,recv_frame_size=8000    # Optionally pass arguments to the selected RF driver.
  sync: external #internal #gpsdo                                         
  clock: external
  srate: 30.72 #61.44 #23.04 #30.72 # 5.76 11.52 23.04                                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 3 #1 #2 #10 #55 #45? # 60                                        # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 8 #12 # 3 # 50 #20? # 50                                               # Receive gain of the RF might need to adjusted to the given situation. 
    #  otw_format: sc12

cell_cfg:
  dl_arfcn: 637212 #640000 #368500 #388000 #361500 # 388000                                                # ARFCN of the downlink carrier (center frequency).
  band: 78 #3 #2 # 66                                                     # The NR band.
  channel_bandwidth_MHz: 20 #100 #20 # 5 10 15 20                                       # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                                                  # Subcarrier spacing in kHz used for data.
  plmn: 00101 # "20895"                                                   # PLMN broadcasted by the gNB.
  tac: 40960 #7
  pci: 1                                                       # Tracking area code (needs to match the core configuration). 
  pdsch:
    min_ue_mcs: 1 #9
    max_ue_mcs: 9 #10
  pusch:
    min_ue_mcs: 1 #9
    max_ue_mcs: 9 #10                           

  pdcch:
    common:
      ss0_index: 0 # 1 #0
      coreset0_index: 2 #12
    dedicated:
      ss2_type: common                                            # Set search space to common to match capabilities
      dci_format_0_1_and_1_1: false
    
  prach:
    prach_config_index: 1 #7 #1
      #prach_root_sequence_index: 1                                  # PRACH root sequence index.
      #zero_correlation_zone: 0                                      # Zero correlation zone.
      #prach_frequency_start: 0

[... 118 lines cut ...]

webui:
  ingress:
    enabled: false
  services:
    http:
    ## @param service.type Kubernetes Service type
    ##
      type: LoadBalancer
      loadBalancerIP: "192.168.12.132"

populate:
  enabled: true
# initCommands to insert Subscribers into core network.
  initCommands:  
  - open5gs-dbctl add_ue_with_apn 001010000000001 465B5CE8B199B49FAA5F0A2EE238A6BC E8ED289DEBA952E4283B54E88E6183CA internet #srsran
    #  - open5gs-dbctl add_ue_with_apn 222010000000002 465B5CE8B199B49FAA5F0A2EE238A6BC E8ED289DEBA012E4283B54E88E6183CA internet #srsran
  - open5gs-dbctl add_ue_with_apn 001010000090877 D1AD471B94630D6698298C001FCAEA3E 8FDC110EF6D98F23DD92E801378BD08B internet #sysmocom
  - open5gs-dbctl add_ue_with_apn 001010000090878 5A2F1F69F08D584552EABE7C23710B9E 7472641E4AC24FBC04AD8C80631B4C64 internet #sysmocom
    #  - open5gs-dbctl add_ue_with_slice 999700000000002 465B5CE8B199B49FAA5F0A2EE238A6BC E8ED289DEBA012E4283B54E88E6183CA internet 1 111111
    #  - open5gs-dbctl add_ue_with_slice 999700000000002 465B5CE8B199B49FAA5F0A2EE238A6BC E8ED289DEBA012E4283B54E88E6183CA internet 1 111111

```
If you have any further question please ask us.
Good luck....

pgawlowicz, 2024-05-21:
@desimetallica could you share your gnb logs?

desimetallica, 2024-05-21:
Hello, @pgawlowicz what type of logs do you need? In warning I got just 1 entry but if I set up debug level the log became quite big (110M) in just 30 seconds.

pgawlowicz, 2024-05-21:
``debug`` mode please

desimetallica, 2024-05-21:
Here the zipped log in debug mode.

[gnb.zip](https://github.com/srsran/srsRAN_Project/files/15388496/gnb.zip)

pgawlowicz, 2024-05-21:
very nice! Thanks for your feedback! Looks very good PUSCH SNR around 30db!

closed 2024-06-20 by pgawlowicz