srsran/srsRAN_Project #516: UE does not connect - RAN550 split 7.2

abcd11abcd, 2024-03-19:
### Discussed in https://github.com/srsran/srsRAN_Project/discussions/398

<div type='discussions-op-text'>

<sup>Originally posted by **AzeezEbrahim** January 10, 2024</sup>
Hi, i trying to use split 7.2 setup, i am getting a locked from the RU and everything seems ok, **However, UEs does not connects.**

anyone can help?  am i missing something?

**Issue**
UE is not connecting to the network. But it is able search for the cell.

**Setup and connections:**

- Ubuntu 22.04
- 13th Gen Intel® Core™ i9-13900K × 32
- 64GB RAM
- Intel Ethernet Network Adapter E810-XXVDA4T (**without** internal GNSS module) --> Grandmaster
- Benetel RAN650
- External GNSS & ublox GPS antenna

**DU Config**

```
amf:
  addr: <ip of core pc>                                               # The address or hostname of the AMF.
  bind_addr: 127.0.0.1                                            # A local IP that the gNB binds to for traffic from the AMF.

ru_ofh:
  ru_bandwidth_MHz: 100                # RU instantaneous bandwidth.
  t1a_max_cp_dl: 392                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 258                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 300                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 285                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 300                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 155
  ta4_max: 300
  ta4_min: 85
  is_prach_cp_enabled: 0            # Configures if Control-Plane messages should be used to receive PRACH messages.
  is_dl_broadcast_enabled: 1            # Optional BOOLEAN (0). Sets downlink broadcast enabled flag. Supported: [0, 1]. 
  #ignore_ecpri_payload_size: 1     # Optional BOOLEAN (0). Sets whether or not to ignore eCPRI payload size field value. Supported [0. 1].
  compr_method_ul: bfp                 # Uplink compression method.
  compr_bitwidth_ul: 9                 # Uplink IQ samples bitwidth after compression.
  compr_method_dl: bfp                 # Downlink compression method.
  compr_bitwidth_dl: 9                 # Downlink IQ samples bitwidth after compression.
  compr_method_prach: bfp              # PRACH compression method.
  compr_bitwidth_prach: 9              # PRACH IQ samples bitwidth after compression.
  enable_ul_static_compr_hdr: 1     # Configures if the compression header is present for uplink User-Plane messages (false) or not present (true).
  enable_dl_static_compr_hdr: 1     # Configures if the compression header is present for downlink User-Plane messages (false) or not present (true).
  iq_scaling: 5.5                      # IQ samples scaling factor applied before compression, should be a positive value smaller than 10.
  cells:
    - network_interface: enp1s0f0np0      # Ethernet interface name used to communicate with the RU.
      ru_mac_addr: <xxxxxxx>   # RU MAC address.
      du_mac_addr: <xxxxxxx>   # DU MAC address.
      vlan_tag: 3                      # VLAN tag value.
      prach_port_id: [4]               # PRACH eAxC port value.
      dl_port_id: [0, 1]                 # Downlink eAxC port values.
      ul_port_id: [0]                 # Uplink eAxC port values.
      
cell_cfg:
  dl_arfcn: 637212                                                 # ARFCN of the downlink carrier (center frequency).
  band: 78                                                        # The NR band.
  channel_bandwidth_MHz: 100                                       # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                                                  # Subcarrier spacing in kHz used for data.
  plmn: "00101"                                                   # PLMN broadcasted by the gNB.
  tac: 1                                                          # Tracking area code (needs to match the core configuration).
  pci: 1                                                          # Physical cell ID.
  nof_antennas_dl: 1                                             # Downlink number of antennas.
  nof_antennas_ul: 1                                              # Downlink number of antennas.
  prach:
[... 139 lines cut ...]
[INFO] Enable PA Protection on Transmitters
[INFO] End Radio Configuration - Start U-Plane Data
[INFO]  
[INFO] Set the attenuation for TX1/2/3/4
[INFO] Radio bringup complete
 13:03:48 up 5 min,  load average: 0.10, 0.40, 0.23

```

</div>

kpi.sh
SAMPLE_TIME          | RX_TOTAL             | RX_ON_TIME           | RX_EARLY             | RX_LATE              | RX_ON_TIME_C         | RX_EARLY_C           | RX_LATE_C            | TX_TOTAL             
 14:05:24.696231      | 39606                | 36791                | -0                   | 3                    | 2810                 | -0                   | -0                   | -0                   
 14:05:25.697682      | 39603                | 36797                | -0                   | -0                   | 2800                 | -0                   | -0                   | -0                   
 14:05:26.689592      | 39605                | 36792                | -0                   | 2                    | 2800                 | -0                   | -0                   | -0                   
 14:05:27.691531      | 39603                | 36799                | -0                   | 1                    | 2802                 | -0                   | -0                   | -0     

i'm also trying to use bentel 650 RU with srsRAN project. The TDD pattern i'm using is 7-2 on both DU and RU side. The TX_TOTAL field is showing as 0 in my case also. Means, no uplink messages are flowing. The configuration file is as same as the one you are using (du and ru mac addresses will change). Im using 100MHz bandwidth. Im using an external GNSS clock for RU synchronization.

The problem I'm facing here, the UE is unable to regsiter to the core network. Im using samsung s22 mobile. UE is able recognize the network but not registering.

How did you solve this problem? how to make the uplink messages to flow?

All the configurations and logs are same on my side. That's why I am not sharing them here

aibtw, 2024-03-19:
At this point it is good to revise the trivial stuff from the srsRAN guide. 
Like, try running `sudo ip link set <YOUR_INTERFACE> mtu 9600`. I usually forget it and it prevents the UE from connecting.

abcd11abcd, 2024-03-20:
I tried keeping the mtu to 9600 also. still the same problem. I forgot to mention this earlier. Any other solution?

aibtw, 2024-03-20:
Hummm, can't think of other issues to be honest. But maybe the timing parameters or whatever they are called. Try these ones, it worked for us:
```
  t1a_max_cp_dl: 500                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 250                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 465                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 250                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 250                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 80                                                  # Minimum T1a on User-Plane in microseconds.
  ta4_max: 500                                                    # Maximum Ta4 on User-Plane in microseconds.
  ta4_min: 25                                                     # Minimum Ta4 on User-Plane in microseconds.
```

Otherwise, I would suggest sharing the logs and pcaps so that other guys can help.

abcd11abcd, 2024-03-21:
attaching the gnb.log file. Kindly refer

[gnb.log.txt](https://github.com/srsran/srsRAN_Project/files/14686512/gnb.log.txt)

andrepuschmann, 2024-03-21:
Hey @abcd11abcd - we are testing with the RAN650 ourselves but are using the newer 1.0.4 firmware. Do you plan to update the FW of your unit?

abcd11abcd, 2024-03-22:
as of now, we are using 0.8 version. Is it mandatory to upgrade to 1.0.4?

andrepuschmann, 2024-03-22:
> as of now, we are using 0.8 version. Is it mandatory to upgrade to 1.0.4?

It's not mandatory but recommended as the way forward, especially with newer releases of srsRAN.

hudouseko, 2024-04-10:
Hi, not sure if you are still troubleshooting, but looking into the log file you attached, I see the following:
1) You should adjust settings on DU server using https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance
2) You have mismatch in uplink eAxCId. According to log RU is sending eAxC=5 ("Dropped received Open Fronthaul User-Plane packet as decoded eAxC value '5' is not configured in reception"), while you have configured in uplink eAxCId=0 ("  ul_port_id: [0]                 # Uplink eAxC port values.").

ismagom, 2024-05-15:
Hi @abcd11abcd , please follow the discussion on https://github.com/srsran/srsRAN_Project/discussions/398

closed 2024-05-15 by ismagom