srsran/srsRAN_Project #521: Inconsistent xApp while testing uplink traffic with multiple UEs

mas-oliveira, 2024-03-21:
## Issue Description ##
I'm trying to connect 3 UEs at the moment and to generate some uplink traffic in order to collect some metrics using an xApp very similar to the x_app_oran_moni that is provided in the FlexRIC project. When I perform the iPerf in the uplink direction using more than 2 links the RIC have an inconsistent behavior.

## Setup Details ##
I'll attach the UEs config and the python script that I'm currently using to launch as many UEs as I wish.
Note: I have to change from .conf to .txt so github can accept the files (I'm actually using the same content but with .conf extension)
[ue3_zmq.txt](https://github.com/srsran/srsRAN_Project/files/14695530/ue3_zmq.txt)
[ue2_zmq.txt](https://github.com/srsran/srsRAN_Project/files/14695531/ue2_zmq.txt)
[ue1_zmq.txt](https://github.com/srsran/srsRAN_Project/files/14695532/ue1_zmq.txt)
This is the Python script that I'm using to launch the UEs based on GNU Radio solution provided in Multi-UE Scenario guide.
[dynamic_multi_ue_scenario.txt](https://github.com/srsran/srsRAN_Project/files/14695539/dynamic_multi_ue_scenario.txt)

## Expected Behavior ##
I was expecting that the RIC continues to provide O-RAN metrics while I keep adding traffic in the uplink.

## Actual Behaviour ##
nearRT-RIC: /home/marco/flexric/src/sm/kpm_sm/kpm_sm_v02.03/dec/kpm_dec_asn.c:184: kpm_dec_ind_msg_asn: Assertion `rval.code == RC_OK && "Are you sending data in ATS_ALIGNED_BASIC_PER syntax?"' failed.
Aborted (core dumped)

## Steps to reproduce the problem ##
sudo ./srsue ue_zmq_confs/ue1_zmq.conf 
sudo ./srsue ue_zmq_confs/ue2_zmq.conf 
sudo ./srsue ue_zmq_confs/ue3_zmq.conf 
I'm launching the 3 UE's in different terminals. 
Then I execute my script => python3 dynamic_multi_ue_scenario.py  
UE1:
CH0 rx_port=tcp://127.0.0.1:2100
CH0 tx_port=tcp://127.0.0.1:2101
PDU Session Establishment successful. IP: 10.45.1.2
RRC NR reconfiguration successful.
UE2:
CH0 rx_port=tcp://127.0.0.1:2200
CH0 tx_port=tcp://127.0.0.1:2201
PDU Session Establishment successful. IP: 10.45.1.4
RRC NR reconfiguration successful.
UE3:
CH0 rx_port=tcp://127.0.0.1:2300
CH0 tx_port=tcp://127.0.0.1:2301
PDU Session Establishment successful. IP: 10.45.1.5
RRC NR reconfiguration successful.

Then I launch the 3 iPerf servers on each UE:
sudo ip netns exec ue1 iperf3 -s 
sudo ip netns exec ue2 iperf3 -s 
sudo ip netns exec ue3 iperf3 -s 

And in order to generate the uplink traffic on the open5gs core I run the following commands:
root@cb69bb0281a9:/open5gs# iperf3 -c 10.45.1.2 -i 1 -t 9000 -u -b 2M
root@cb69bb0281a9:/open5gs# iperf3 -c 10.45.1.4 -i 1 -t 9000 -u -b 2M
root@cb69bb0281a9:/open5gs# iperf3 -c 10.45.1.5 -i 1 -t 9000 -u -b 2M

The traffic still being generated by the iPerf, and the links look ok, but the RIC when I add traffic to the uplink direction towards the second or third UE fires the error: 
nearRT-RIC: /home/marco/flexric/src/sm/kpm_sm/kpm_sm_v02.03/dec/kpm_dec_asn.c:184: kpm_dec_ind_msg_asn: Assertion `rval.code == RC_OK && "Are you sending data in ATS_ALIGNED_BASIC_PER syntax?"' failed.

yagoda, 2024-03-21:
Hi, 

This error is coming from FlexRic so we cant say for sure what could be the issue as this is not our software. 
Perhaps if you share the e2.pcap this could shed some light?

Regards,
Justin

mas-oliveira, 2024-03-21:
> Hi,
> 
> This error is coming from FlexRic so we cant say for sure what could be the issue as this is not our software. Perhaps if you share the e2.pcap this could shed some light?
> 
> Regards, Justin

Hello Justin,
[e2ap.zip](https://github.com/srsran/srsRAN_Project/files/14696160/e2ap.zip)

mas-oliveira, 2024-03-21:
Sorry, I have accidentally closed the issue.

pgawlowicz, 2024-04-04:
Hi @mas-oliveira, 
could you check the CPU load during the emulation? The E2 Agent is executed in a low-priority thread, therefore if the gnb has a lot to do, the E2 agent will struggle. This issue should be only present with zmq based setup. To reduce the CPU load you can increase the slow-down factor.

mas-oliveira, 2024-04-04:
> Hi @mas-oliveira, could you check the CPU load during the emulation? The E2 Agent is executed in a low-priority thread, therefore if the gnb has a lot to do, the E2 agent will struggle. This issue [...]

Thanks. I set the value up to 14 and it was possible to connect the 3 UEs, the problem is related to the CPU Load

mas-oliveira, 2024-04-04:
> Also, please note that in order to generate uplink traffic you should rather use the following commands
> 
> ```
> sudo ip netns exec ue1 iperf -c 10.45.1.1 -i 1 -t 9000 -u -b 2M
> sudo ip netns exec ue2 iperf -c 10.45.1.1 -i 1 -t 9000 -u -b 2M
> sudo ip netns exec ue3 iperf -c 10.45.1.1 -i 1 -t 9000 -u -b 2M
> ```
> 
> All UEs will send UDP traffic to the core network IP. Note that when running in UDP mode, `iperf` client does not require server to be running. In the case of `iperf3` server has to always run.

Yes thanks, I had noticed that I had the inverse ways, I prefer to use the iPerf3 because it is possible to get more info about the connection.

pgawlowicz, 2024-04-08:
Hi @mas-oliveira, did you try the provided patch? does it solve you issue?

mas-oliveira, 2024-04-09:
> Hi @mas-oliveira, did you try the provided patch? does it solve you issue?

Hello @pgawlowicz, sorry but I can't apply the patch:
marco@marco-NUC7i7BNH:~/srsRAN_Project$ git apply e2sm_kpm.patch
error: patch failed: lib/e2/e2sm/e2sm_kpm/e2sm_kpm_du_meas_provider_impl.cpp:544
error: lib/e2/e2sm/e2sm_kpm/e2sm_kpm_du_meas_provider_impl.cpp: patch does not apply

I'll send my current version of e2sm_kpm_du_meas_provider_impl.cpp
[e2sm_kpm_du_meas_provider_impl.zip](https://github.com/srsran/srsRAN_Project/files/14917200/e2sm_kpm_du_meas_provider_impl.zip)

pgawlowicz, 2024-04-09:
could you try to apply the patch manually? Those are only two small fixes (each touching one line) in two files

mas-oliveira, 2024-04-09:
> could you try to apply the patch manually? Those are only two small fixes (each touching one line) in two files

Yes sure, I applied on scheduler_metric_handler.cpp but on e2sm_kpm_du_meas_provider_impl.cpp I wasn't able to find for (unsigned idx = 0; ue_aggr_rlc_metrics.size(); idx++) { 
Am I working with the wrong version?

pgawlowicz, 2024-04-09:
ok, seems you are using an older version, but this one does not contain the bug the patch is fixing. So it should be good. Now, you should be able to request RRU.Prb* and DRB.PacketSuccessRateUlgNBUu metrics. Could you confirm?

mas-oliveira, 2024-04-09:
> ok, seems you are using an older version, but this one does not contain the bug the patch is fixing. So it should be good. Now, you should be able to request RRU.Prb* and DRB.PacketSuccessRateUlgNBUu metrics. Could you confirm?

Hello again Piotr,

First of all thanks again for all of the support.

I tried to run with the following metrics:
                    { name = "DRB.UEThpDl" },
                    { name = "DRB.UEThpUl" },
                    { name = "DRB.RlcPacketDropRateDl" },
                    { name = "DRB.RlcSduTransmittedVolumeDL" },
                    { name = "DRB.RlcSduTransmittedVolumeUL" },
                    { name = "DRB.PacketSuccessRateUlgNBUu" },
                    { name = "RRU.PrbAvailDl" },
                    { name = "RRU.PrbAvailUl" }
And everything worked properly. But when I try to add the { name = "RRU.PrbTotUl" } the xApp just stay stucked at some point.

Generated of req_id = 1 
E42_RIC_SUBSCRIPTION_REQUEST 31 
adding event fd = 6 ev-> 6 
[xApp]: RIC SUBSCRIPTION REQUEST sent
[xApp]: SUBSCRIPTION RESPONSE received
Pending event size before remove = 1 
[xApp]: Successfully SUBSCRIBED to ran function = 2 

I get this message but then I don't receive on my xApp any KPM message. This behavior is verified when I add the request to get the total number of PRB from any direction, uplink (RRU.PrbTotUl or RRU.PrbTotDl)

pgawlowicz, 2024-04-09:
could you try to use the newest srsgnb version and apply the patch there?

mas-oliveira, 2024-04-10:
> could you try to use the newest srsgnb version and apply the patch there?

The behavior stills the same with the newer version and the patch already applied.

When I try to subscribe to the total amount of PRBs:

Near RT RIC logs
> [iApp]: E42 SETUP-REQUEST received
[iApp]: E42 SETUP-RESPONSE sent
[iApp]: SUBSCRIPTION-REQUEST xapp_ric_id->ric_id.ran_func_id 2  
[E2AP] SUBSCRIPTION REQUEST generated
[NEAR-RIC]: nb_id 411 port = 25031  

xApp Logs

> Generated of req_id = 1 
E42_RIC_SUBSCRIPTION_REQUEST 31 
adding event fd = 6 ev-> 6 
[xApp]: RIC SUBSCRIPTION REQUEST sent
[xApp]: SUBSCRIPTION RESPONSE received
Pending event size before remove = 1 
[xApp]: Successfully SUBSCRIBED to ran function = 2 

And it gets everything stuck after that.
Without the total PRBs and only with the available metrics the xApp still running properly.

mas-oliveira, 2024-04-11:
Hello @pgawlowicz,

Sorry, it was a problem of my xApp, I'm now running with the default one and it's everything ok.
Thanks for the help!

closed 2024-04-11 by mas-oliveira

chonghua1, 2024-05-22:
> factor

Hi @mas-oliveira,
I also have the same issue, could you please share how to increase the slow-down factor? Thank you!

mas-oliveira, 2024-05-22:
> > factor
> 
> Hi @mas-oliveira, I also have the same issue, could you please share how to increase the slow-down factor? Thank you!

Sure @chonghua1
You just need to adjust the slow down ratio value on GNU Radio interface, or if you move the GNU Radio blocks to python/C++ code you just need to increase the value of the variable that probably is called by slow_down_ratio.
By doing this the throttle blocks will be updated because they depend on the slow down ratio and then your computer will have more time to process the ZMQ messages.

chonghua1, 2024-05-22:
> Sure @chonghua1 You just need to adjust the slow down ratio value on GNU Radio interface, or if you move the GNU Radio blocks to python/C++ code you just need to increase the value of the variable [...]

Got it, thank you!