srsran/srsRAN_Project #523: ASN.1 issue - gNB crash when activating a PDU

Luna5G, 2024-03-21:
Issue Description

After upgrading to latest FW on a SIMCOM X62 based 5G Module, I get the following message when activating a 5G SA PDU session:

"/root/srsRAN/lib/asn1/asn1_utils.cpp:216: asn1::SRSASN_CODE asn1::cbit_ref::unpack(T&, uint32_t) [with T = unsigned int; uint32_t = unsigned int]: Assertion `n_bits <= sizeof(T) * 8' failed - unpack_bits() only supports up to 32 bits."

Attach is working fine, and I can downgrade it to the old FW and it works again. As the gNB crashes it does not save the logs. What is the best way to debug this ? or any idea where the problem can be ?

andrepuschmann, 2024-03-22:
Could you perhaps share a NGAP PCAP? You can capture that with `tcpdump` or Wireshark directly on the interface?

Luna5G, 2024-03-22:
> Could you perhaps share a NGAP PCAP? You can capture that with `tcpdump` or Wireshark directly on the interface?

What Interface to use?, the one between the AMF and the CU/DU ? or maybe the UPF and CU/DU ?

andrepuschmann, 2024-03-22:
> What Interface to use?, the one between the AMF and the CU/DU ? or maybe the UPF and CU/DU ?

AMF should be the one causing the bad behavior. Please capture a PCAP on that.

Luna5G, 2024-03-22:
Here is both the CU-AMF & the CU-UDP log (made 2 separate times)
[log.zip](https://github.com/srsran/srsRAN_Project/files/14721173/log.zip)

andrepuschmann, 2024-03-22:
Hey @Luna5G - the unpacking of that NGAP message worked in our local test here. Could you perhaps do the following?
* compile srsRAN with `-DCMAKE_BUILD_TYPE=Debug` 
* run the gNB with `gdb --args ./gnb -t ...` and when gdb hits the assert type `bt+Enter` and provide the output.

Thanks

Luna5G, 2024-03-23:
Here the bt dump I got, as you see from the dump, there is a bit of startup and then comes the crash. (I did the bt+Enter after the crash, should I have done it before, then let me know)
[bt dump.txt](https://github.com/srsran/srsRAN_Project/files/14731239/bt.dump.txt)

BTW, on the Open5GS issue site, there was a link to this: https://github.com/mouse07410/asn1c/issues/179
I have no idea if it's related, but could be.
I hope you get something out of the dump (I don't :)

andrepuschmann, 2024-03-23:
Ok, thanks for the backtrace. Could you please do the following:
* Apply the below patch
* recompile and run the gNB with MAC PCAPs enabled
* attach the UE like normal
* exit the gNB
* upload the PCAP

Thanks

```
diff --git a/lib/rrc/ue/procedures/rrc_ue_capability_transfer_procedure.cpp b/lib/rrc/ue/procedures/rrc_ue_capability_transfer_procedure.cpp
index 371317b19a..71ee7bcd0a 100644
--- a/lib/rrc/ue/procedures/rrc_ue_capability_transfer_procedure.cpp
+++ b/lib/rrc/ue/procedures/rrc_ue_capability_transfer_procedure.cpp
@@ -50,6 +50,7 @@ void rrc_ue_capability_transfer_procedure::operator()(coro_context<async_task<bo
           transaction.response().msg.c1().ue_cap_info().crit_exts.ue_cap_info().ue_cap_rat_container_list);
       for (const auto& ue_cap_rat_container :
            transaction.response().msg.c1().ue_cap_info().crit_exts.ue_cap_info().ue_cap_rat_container_list) {
+#if 0
         if (ue_cap_rat_container.rat_type.value == asn1::rrc_nr::rat_type_e::nr) {
           asn1::cbit_ref            bref{ue_cap_rat_container.ue_cap_rat_container.copy()};
           asn1::rrc_nr::ue_nr_cap_s ue_nr_cap;
@@ -69,6 +70,7 @@ void rrc_ue_capability_transfer_procedure::operator()(coro_context<async_task<bo
         } else {
           logger.log_warning("Unsupported RAT type {}", ue_cap_rat_container.rat_type);
         }
+#endif
       }
     }
     procedure_result = context.capabilities.has_value();
```

Luna5G, 2024-03-24:
Here the MAC files after removing the RAT check.
[gnb_mac.zip](https://github.com/srsran/srsRAN_Project/files/14734848/gnb_mac.zip)

andrepuschmann, 2024-03-28:
Hey @Luna5G - is the 5G module you're using a SIMCOM SIM8262E? If so, what was the reason you upgraded the firmware and where did you it from?

Luna5G, 2024-03-29:
It is, SIM8260G, but I think the SIM8262E showed the same issue. I got the FW from SIMCOM, I always try to use the latest FW. If you can describe your findings, I can send it to SIMCOM for comments

Luna5G, 2024-04-23:
I just tested SIM8260G with your new 24.04 release, it looks like it's working, so you can close this issue :-)

closed 2024-04-23 by Luna5G