srsran/srsRAN_Project #529: MSG4 retransmission

lpg-delete, 2024-03-26:
## Issue Description ##
[Describe the issue in detail]

## Setup Details ##
MSG4(rrc setup + ue contention resolution identity) retransmission is executed in ue_fallback_scheduler.cpp. gnb considers UE out of fallback mode after receiving SR or CSI report, which means UE uses rrc setup configuration. A situation: UE received MSG4, decoded correctly, pucch resource using sib1 feedback MSG4 ack, but gnb failed to detect ack, gnb thinks MSG4 needs to be retransmitted, gnb retransmitted MSG4 using pucch resource is still in sib1 pucch resource, UE still uses rrc setup resources, which can cause unnecessary retransmissions. If gnb receives an SR or CSI report from UE, is it considered that UE has received MSG4, even if gnb does not receive the correct ACK for MSG4?

## Expected Behavior ##
Cancel fallback scheduling when UE is out of fallback mode (MSG4 retransmission)

carlo-gal, 2024-03-28:
Thank you for reporting this.
We are aware of the problem. We have already considered the solution you proposed, but we eventually discard it, as it has some drawbacks. 
When we create a UE in the scheduler, we start allocating CSI and SR opportunities, potentially before the UE has received the MSG4. Now, if we consider the SR and/or CSI reception at the GNB as a trigger to cancel the MSG4 scheduling (whether it's a retx or not), a false positive detection of CSI and SR before the MSG4 has been acked (i.e., the GNB detects a SR or CSI even though the UE didn't send any) would interrupt the MSG4 scheduling, even though the UE didn't detect it. This would leave the UE without MSG4, and would force a new PRACH.

Solving this problem requires a change (among several others) in the way the PUCCH allocate resources. We eventually plan to solve this issue, but the problem is not trivial and requires quite some work.

lpg-delete, 2024-03-28:
Thank you for your attention to this issue, I will look forward to the final resolution of the code, this issue will be closed.

closed 2024-03-28 by lpg-delete