srsran/srsRAN_Project #533: UE gets disconnected in a few seconds

smunaut, 2024-03-27:
## Issue Description ##

When connecting a UE to gNB, the phone connects and has network access but within a few seconds, it gets kicked out.

I've attached the gnb log files from such a session and what seems relevant is :
```
2024-03-27T10:37:08.414684 [MAC     ] [W] [   517.5] ue=0: RLF detected. Cause: 100 consecutive HARQ-ACK KOs.
2024-03-27T10:37:08.414697 [SCHED   ] [I] [   517.6] DL HARQ rnti=0x4601 cell=0 h_id=2: Discarding HARQ process tb=0 with tbs=101. Cause: Maximum number of reTxs 4 exceeded
2024-03-27T10:37:08.414702 [DU-MNG  ] [I] ue=0: RLF detected with cause="MAC max KOs reached". Timer of 4000 msec to release UE started...
```

[gnb_log.txt](https://github.com/srsran/srsRAN_Project/files/14774876/gnb_log.txt)

Now the interesting part is that this is running `2f90c8b60e9396a7aed59645c98dbcbccda2bf7c`
But using the exact same setup ( same open5gs / same phone / same sim / same rf setup and almost same config files, just replaced `rf_driver` with `cu_sdr`), an older version of srsran_project ( `d60e56c12b91920953b967cf8c9024fd6b8c30d1` ) works perfectly and I can stay connected with no issues.

## Setup Details ##

Everything runs on a single linux machine, using a B205 mini fed with a 10 MHz GPSDO and the config setup to use `clock: external`.
Two vert900 antenna put 90 deg off each other. UHD 3.15 LTS driver. Kernel 6.5.

Phone is a EB2103 ( OnePlus Nord CE 5G ) although I also tried IN2013 ( OnePlus 8) and Samsung A54 with same behavior.

smunaut, 2024-03-27:
Release `23_10_1` works fine too FWIW.
I'll try to bisect but not sure how that will work out.

smunaut, 2024-03-27:
`66e7ef966adb5b08634b897483db882514857d06` is my best guess from the bisecting ... not super easy to tell because although in recent commits the bug happens very quickly, in some older commits, it happens after a longer time, but it seems to start at that commit at least.

frankist, 2024-03-28:
Thanks for reporting this @smunaut. The git commit hashs you mentioned as stable are quite old, so it is weird that this would remain undetected for so long.

Can you share new logs with the gnb mac log set to debug level? I want to observe the SINR of the PUCCH that is reporting `ack=222` (which means inability to detect the ACKs)

smunaut, 2024-03-28:
Hi @frankist 

Sure, here's the requested log attached.
[gnb_log.zip](https://github.com/srsran/srsRAN_Project/files/14789508/gnb_log.zip)
Once with just MAC as debug and once with both PHY and MAC.  (I didn't see the "edit" at first :sweat_smile: )

And yeah ,those commits are quite old so I'm not sure why I would be the only one to hit that, I find it weird too.
Although at those commit, although the same issue seems to happen, it's not as "immediate", it can take a couple of minutes.

But with 23_10 for sure it works fine, I've been connected all morning without issue and when I go out of RF coverage, I do get lots of errors and then loose connection, but when I come back, it reconnects immediately.  

With latest master when the issue occur, getting the phone to reconnect is a hassle, I need to go in/out of airplane mode several time, change APN, use the 5G forcer on/off and eventually it will retry but it takes a while ...

smunaut, 2024-04-02:
@ismagom  Sure.  Here's the same log, using the same config on 23_10 and running a speedtest while connected.

[gnb-working.zip](https://github.com/srsran/srsRAN_Project/files/14833983/gnb-working.zip)

herlesupreeth, 2024-04-03:
@achkonsa Your issue maybe different than what @smunaut is facing. In your case the behavior maybe due to UE inactivity timer. You could use the below configuration in your gNB config file so that UE is not released after 5 seconds of inactivity (inactivity time is by default set to 5 seconds)

```
cu_cp:
  inactivity_timer: 7200
```

achkonsa, 2024-04-03:
thank you so much!!! It helps me a loooooot. And I want to ask that does 7200 mean the gnb will end after 7200 seconds? Or it will keep connection all the time? Thanks a loooooot!!!!! By the way, my partner does not meet this issue. I wonder why...His version is `23_10_1-821-g0b2702cca`

herlesupreeth, 2024-04-03:
> And I want to ask that does 7200 mean the gnb will end after 7200 seconds?

`inactivity_timer` represents the time in seconds after which UE gets release if there is no activity in UE (e.g. no data flow from/to UE etc). Its not the time after which gNB will end.

achkonsa, 2024-04-04:
OK!!!!Thanks for explaining. But I still wonder why there is an inactivate UE keeping connected for a looong time... There is no `inactivity_timer` inside the gnb configuration... :) And it seems that the UE can be connected forever...... I have not seen it end even though the UE is inactivate...

smunaut, 2024-04-04:
@achkonsa Please open a discussion if you want to discuss this idle behavior but this is distinct from the actual problem discussed in this issue.

ismagom, 2024-04-04:
Hi @smunaut , we were able to reproduce the issue and provided a fix. Could you test if: 

https://github.com/srsran/srsRAN_Project/tree/fix_disconnect

fixes it for you? 

FYI, the problem seemed to be we were using the extended size bit in the MAC header to indicate the size of a PDU which didn't need it.

smunaut, 2024-04-04:
@ismagom :tada:  It works perfectly. Been connected for 10 min without issues.  It also goes IDLE after 5s inactivity and comes back directly no issues there either.

Thanks for the fix !

smunaut, 2024-04-04:
And as a side question : was I anywhere close with the bisect as to where that was introduced ? :sweat_smile:

ismagom, 2024-04-04:
The issue was introduced in commit 065ac91f4eb14bda47a8c26a0bba5d8b199167c7

closed 2024-04-11 by pgawlowicz