srsran/srsRAN_Project #535: srsRAN_Project || gNB || Error during Installation and Running

LingeshWaran, 2024-03-29:
## Issue Description ##
srsran_performance throws an exception & Unable to start gNB

## Setup Details ##
Dear Sir/Madam,
Warm greetings. I am serving as an assistant professor in the dept. of ECE, St. Joseph's College of Engineering,OMR, Chennai, India. I am planning to set up a basic 5G E2E testbed using COTS UE, NI USRP B210, srsRAN5G and Open5GS Core. I started with srsUE and ZeroMQ libraries to emulate the RF environment. I encountered the following error while installing srsRAN_Project in my laptop with Oracle virtual box running ubuntu with 4 cores and 4GB RAM:

**root@5GTIC:/srsRAN_Project# sudo scripts/srsran_performance**
**Would you like to set the CPU governor to performance: [Y/n]: Y**
**tee: '/sys/devices/system/cpu/cpu*/cpufreq/scaling_governor': No such file or directory**

I continued to complete the installation of Zero MQ libraries, srsRAN4G and Open5GS. I was able to start my Core but gNB terminal showed the following exception:

**root@5GTIC:/srsRAN_Project/build/apps/gnb# sudo gnb -c /home/lingesh/Downloads/gnb_zmq.yaml
The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
srsGNB ERROR: The CPU does not support the required CPU features that were configured during compile time: sse4.1(ok) avx(ok) avx2(ok) fma(na) pclmul(ok)**

I am able to start UE too. Please help me in this regard. Kindly find the attached snapshots.

## Expected Behavior ##
CPU performance should be tuned and gNB  should start

## Actual Behaviour ##
**root@5GTIC:/srsRAN_Project# sudo scripts/srsran_performance**
**Would you like to set the CPU governor to performance: [Y/n]: Y**
**tee: '/sys/devices/system/cpu/cpu*/cpufreq/scaling_governor': No such file or directory**

**root@5GTIC:/srsRAN_Project/build/apps/gnb# sudo gnb -c /home/lingesh/Downloads/gnb_zmq.yaml
The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
srsGNB ERROR: The CPU does not support the required CPU features that were configured during compile time: sse4.1(ok) avx(ok) avx2(ok) fma(na) pclmul(ok)**

## Steps to reproduce the problem ##
I am planning to set up a basic 5G E2E testbed using COTS UE, NI USRP B210, srsRAN5G and Open5GS Core. I started with srsUE and ZeroMQ libraries to emulate the RF environment. I encountered these errors
[image]
[image]
 while installing srsRAN_Project in my laptop with Oracle virtual box running ubuntu with 4 cores and 4GB RAM:

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

[image]
[image]

ninjab3s, 2024-04-01:
Hi @LingeshWaran,
we dont recommend to install srsRAN Project in a VM. Due to the real-time requirements of the DU using a VM is usually not straight forward. You need to use a VM that supports real-time applications like ESXi from VMWare [1] and you need to know exactly what you are doing. Is it possible to use a docker in your setup?

Regarding the srsran_performance script: Inside a VM you cannot influence the performance governor of the host system, because the VM doesnt have the rights to modify these parameters.

[1] https://docs.vmware.com/en/VMware-Edge-Compute-Stack/3.0/ecs-enterprise-edge-ref-arch/GUID-99972F62-3FAE-42B2-A850-66402274B616.html

LingeshWaran, 2024-04-01:
Dear @ninjab3s,
Thanks for your time and consideration. I disabled ISA in the CMakeLists.txt file:  L.No.: 69 and commented Have_FMA section to avoid automatic detection. Now, srsRAN is installed in my laptop. 
[image]

ninjab3s, 2024-04-01:
Hi @LingeshWaran,
I am glad that its working for you. Please close this Discussion if all your issues are resolved.

LingeshWaran, 2024-04-02:
Issue resolved.

closed 2024-04-02 by LingeshWaran

MRxTIGERRICE, 2024-06-18:
> ## Issue Description
> srsran_performance throws an exception & Unable to start gNB
> 
> ## Setup Details
> Dear Sir/Madam, Warm greetings. I am serving as an assistant professor in the dept. of ECE, St. Joseph's College of Engineering,OMR, Chennai, India. I am planning to set up a basic 5G E2E testbed using COTS UE, NI USRP B210, srsRAN_Project and Open5GS Core. I started with srsUE and ZeroMQ libraries to emulate the RF environment. I encountered the following error while installing srsRAN_Project in my laptop with Oracle virtual box running ubuntu with 4 cores and 4GB RAM:
> 
> **root@5GTIC:/srsRAN_Project# sudo scripts/srsran_performance** **Would you like to set the CPU governor to performance: [Y/n]: Y** __tee: '/sys/devices/system/cpu/cpu_/cpufreq/scaling_governor': No such file or directory_*
> 
> I continued to complete the installation of Zero MQ libraries, srsRAN4G and Open5GS. I was able to start my Core but gNB terminal showed the following exception:
> 
> **root@5GTIC:/srsRAN_Project/build/apps/gnb# sudo gnb -c /home/lingesh/Downloads/gnb_zmq.yaml The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}. srsGNB ERROR: The CPU does not support the required CPU features that were configured during compile time: sse4.1(ok) avx(ok) avx2(ok) fma(na) pclmul(ok)**
> 
> I am able to start UE too. Please help me in this regard. Kindly find the attached snapshots.
> 
> ## Expected Behavior
> CPU performance should be tuned and gNB should start
> 
> ## Actual Behaviour
> **root@5GTIC:/srsRAN_Project# sudo scripts/srsran_performance** **Would you like to set the CPU governor to performance: [Y/n]: Y** __tee: '/sys/devices/system/cpu/cpu_/cpufreq/scaling_governor': No such file or directory_*
> 
> **root@5GTIC:/srsRAN_Project/build/apps/gnb# sudo gnb -c /home/lingesh/Downloads/gnb_zmq.yaml The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}. srsGNB ERROR: The CPU does not support the required CPU features that were configured during compile time: sse4.1(ok) avx(ok) avx2(ok) fma(na) pclmul(ok)**
> 
> ## Steps to reproduce the problem
> I am planning to set up a basic 5G E2E testbed using COTS UE, NI USRP B210, srsRAN_Project and Open5GS Core. I started with srsUE and ZeroMQ libraries to emulate the RF environment. I encountered these errors ![gNB](https://private-user-images.githubusercontent.com/9608742/318012995-bbe6f056-8b3c-4450-acee-ab14ec8160e6.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2ODg5ODMsIm5iZiI6MTcxODY4ODY4MywicGF0aCI6Ii85NjA4NzQyLzMxODAxMjk5NS1iYmU2ZjA1Ni04YjNjLTQ0NTAtYWNlZS1hYjE0ZWM4MTYwZTYuUE5HP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MThUMDUzMTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzUzZDQzMWZlMmYyYWU1NmU1YzRkMzkyNmRjM2M2ZjBiMDBiNjdiN2Y5ZGEwZmI3YTIyNTg1ZDQxNjg4MDYwZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.B1F-swyfpqNSgMLcPj6FwbFFwDOfDaGWvwolyBK_3l0) ![GrafanaUI](https://private-user-images.githubusercontent.com/9608742/318013029-d2ff6423-cd94-4fb8-b1dc-cba340ca3e41.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2ODg5ODMsIm5iZiI6MTcxODY4ODY4MywicGF0aCI6Ii85NjA4NzQyLzMxODAxMzAyOS1kMmZmNjQyMy1jZDk0LTRmYjgtYjFkYy1jYmEzNDBjYTNlNDEuUE5HP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MThUMDUzMTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTVjMWViMjEwNWRkYTYwOTk4ZGMxZDkzZTQ0YWYzZGJmOGIxNGI3NjQwNjVkNTZhZGViY2U2ZWViNWY3ZGJhZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.3R6ImOOdJLZE6lziziWR9_OodP9whyyUpoPSKVH1Kvo) while installing srsRAN_Project in my laptop with Oracle virtual box running ubuntu with 4 cores and 4GB RAM:
> 
> ## Additional Information
> [Any additional information, configuration or data that might be necessary to reproduce the issue]
> 
> ![OPEN5GS](https://private-user-images.githubusercontent.com/9608742/318013037-ce8ad7ba-42bc-4b4d-b993-22ea56e9a3f1.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2ODg5ODMsIm5iZiI6MTcxODY4ODY4MywicGF0aCI6Ii85NjA4NzQyLzMxODAxMzAzNy1jZThhZDdiYS00MmJjLTRiNGQtYjk5My0yMmVhNTZlOWEzZjEuUE5HP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MThUMDUzMTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2Y2NmViOWIwMzQ2NjM3ZmMxYTNiOWJiODg5ZWM4YTZmNDY1ZWFlZjU2ZjA1YTMxYjU1ZjY2MWIxOGI5OTVjYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.t7IyBi3j3PPB4-_T1v-gW7tVOHodBNLpicqQbHVB-gs) ![ZMQUE](https://private-user-images.githubusercontent.com/9608742/318013059-ce070f7d-3282-4fb1-8c2d-994a91258e4a.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2ODg5ODMsIm5iZiI6MTcxODY4ODY4MywicGF0aCI6Ii85NjA4NzQyLzMxODAxMzA1OS1jZTA3MGY [... cut ...] U0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MThUMDUzMTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODUxYTFlNmIyNzdhMzRhYmY1M2I1ZTk5OTEyZDRhZDA1ODNkMzFjMmMyZGQ3OWM1N2E4MTYwYjUxNWI5YzQxYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Hoz-WyNZJwJWr5pkTOYv1tpe6bcsOdNqKACY3xUnORE)

How were you utilising grafana in this scenario sir and what plugin did u use specifically?