srsran/srsRAN_Project #557: Intra-eNB & S1 Handover

Mohammed24699, 2024-04-17:
## Getting error while initiating srsenb ##

## Setup Details ##
with the help of this https://docs.srsran.com/projects/4g/en/hoverxref_test/app_notes/source/handover/source/index.html documentation we followed everything.

## Expected Behavior ##
[image]

## Actual Behaviour ##
[image]

## Steps to reproduce the problem ##
By seeing the error can someone please tell me what the problem is ??

## Additional Information ##
Attaching enb.conf and rr.conf files.

[enb conf. txt](https://github.com/srsran/srsRAN_Project/files/15004814/enb.conf.txt)
[rr conf.txt](https://github.com/srsran/srsRAN_Project/files/15004816/rr.conf.txt)

pgawlowicz, 2024-04-17:
you need to put your configs to `/root/.config/srsran/` or start with `sudo srsenb ./enb.conf`

Mohammed24699, 2024-04-17:
[image]

pgawlowicz, 2024-04-17:
`sudo srsue ./ue.conf --rat......`

Mohammed24699, 2024-04-17:
[image]

Do I need to change anything in the ue config file ??
[image]

Mohammed24699, 2024-04-17:
I make the line 27 as comment. I think it worked. Let me know if I am doing it in the right way.

pgawlowicz, 2024-04-17:
There was a change in the config structure. 

For the ZMQ-based setup, it does not matter, you can comment out this parameter. 

But if you want to run over usrp, you need to set it with `rat.eutra.dl_earfcn` (check here : https://github.com/srsran/srsRAN_4G/blob/master/srsue/src/main.cc#L130), so that the TX/RX freq are set correctly.

Mohammed24699, 2024-04-19:
I want to use usrp in future.
I checked that config files of rnb, rr, enb and they are same as the documentation. I am still getting the error. How can I solve it.
[image]

pgawlowicz, 2024-04-19:
Did you comment out the dl_earfcn line in the config file?

Mohammed24699, 2024-04-19:
It worked when I comment out the dl_earfcn line.

Mohammed24699, 2024-04-19:
I can do the intra handover without any issue.

pgawlowicz, 2024-04-19:
We will update the configs in the tutorial

closed 2024-04-23 by pgawlowicz

Mohammed24699, 2024-04-24:
Hi, I'm trying s1 handover now and I'm facing issues. Let me tell you the step by step process what i did so that you can correct me where i went wrong.

1. Changed the tac value to 7 in mme.yaml file of open 5gc. 
[image]

2. Added subscriber to the network 
[image]

3. added usim credentials to the ue config file 
[image]
[image]
4. changed mme address in enb config file. 
[image]
5. changed rr1 and rr2 config files according to the documentation.
6. Attached the bash scripts to the config files of enb1, enb2 and UE
7. Initiated eNB. But got errors 
[image]

Attaching the config files here
[enb.txt](https://github.com/srsran/srsRAN_Project/files/15101421/enb.txt)
[enb1.txt](https://github.com/srsran/srsRAN_Project/files/15101422/enb1.txt)
[enb2.txt](https://github.com/srsran/srsRAN_Project/files/15101424/enb2.txt)
[rr1.txt](https://github.com/srsran/srsRAN_Project/files/15101426/rr1.txt)
[rr2.txt](https://github.com/srsran/srsRAN_Project/files/15101427/rr2.txt)

pgawlowicz, 2024-04-25:
you should remove (or comment out) this part in your configs:
```
LOG_ARGS="--log.all_level=debug"

PORT_ARGS="tx_port=tcp://*:2101,rx_port=tcp://localhost:2100"
ZMQ_ARGS="--rf.device_name=zmq --rf.device_args=\"${PORT_ARGS},id=enb,base_srate=23.04e6\""

OTHER_ARGS="--enb_files.rr_config=rr1.conf"

sudo srsenb enb.conf ${LOG_ARGS} ${ZMQ_ARGS} ${OTHER_ARGS} $@
```

Mohammed24699, 2024-04-25:
I couldn't able to open two eNB's in the same time. I tried changng the port numbers in the eNB config files but it showing this error.
[image]

but the eNB1 is started.
[image]

Mohammed24699, 2024-04-25:
here I'm attaching the config files
[enb1.txt](https://github.com/srsran/srsRAN_Project/files/15121966/enb1.txt)
[enb2.txt](https://github.com/srsran/srsRAN_Project/files/15121967/enb2.txt)

pgawlowicz, 2024-04-25:
you cannot use the same zmq ports in the two enb, and there is no sib anr rr configs in your working directory.

Mohammed24699, 2024-04-25:
what should i do to use another zmq port to the second enb ??

Mohammed24699, 2024-04-26:
I figured it out that we need to run the bash file first. Now, UE is not connecting to rrc.
[image]
[image]
[image]

pgawlowicz, 2024-04-26:
could you share open5gs config and logs?

Mohammed24699, 2024-04-28:
[config.tar.gz](https://github.com/srsran/srsRAN_Project/files/15143528/config.tar.gz)
[log.tar.gz](https://github.com/srsran/srsRAN_Project/files/15143529/log.tar.gz)

please have a look

pgawlowicz, 2024-04-29:
in MME logs you have:
```
04/23 14:51:15.334: [mme] INFO: [Added] Number of eNBs is now 1 (../src/mme/mme-context.c:2829)
04/23 14:51:15.334: [mme] INFO: eNB-S1[127.0.1.1] max_num_of_ostreams : 30 (../src/mme/mme-sm.c:150)
04/23 14:51:15.335: [mme] WARNING: S1-Setup failure: (../src/mme/s1ap-handler.c:234)
04/23 14:51:15.335: [mme] WARNING:     Cannot find Served TAI. Check 'mme.tai' configuration (../src/mme/s1ap-handler.c:235)
04/23 14:51:15.336: [mme] INFO: eNB-S1[127.0.1.1] connection refused!!! (../src/mme/mme-sm.c:166)
04/23 14:51:15.336: [mme] INFO: [Removed] Number of eNBs is now 0 (../src/mme/mme-context.c:2863)
```

So there is TAI mismatch between enb and open5gs configs.

Mohammed24699, 2024-06-10:
hello,
we are doing s1 handover in another computer but we are facing some issues.

[image]

"epc.conf" file is also in the same directory.

your help will be appriciated. Thanks.

Mohammed24699, 2024-06-10:
attaching the file epc config file here : 
[epc.txt](https://github.com/user-attachments/files/15755393/epc.txt)

pgawlowicz, 2024-06-10:
hmm, can you try with an absolute path to the config file?

Mohammed24699, 2024-06-20:
while I'm initiating the enb's , I got the error saying "S1 Setup Failure. Cause: misc - unknown-PLMN"
Here are the screenshots
[image]
[image]

And I also crosschecked the PLMN in the ue config file and the localhost:9999, it's the same 90170

pgawlowicz, 2024-06-20:
It seems that the PLMN used by enb and EPC is different.

Mohammed24699, 2024-06-20:
[mme.txt](https://github.com/user-attachments/files/15917703/mme.txt)
[enb.txt](https://github.com/user-attachments/files/15917704/enb.txt)
I changed them in mme and gnb config file manually. Do i need to change ny where else?

pgawlowicz, 2024-06-21:
`tac` parameter is also in `rr.conf` file in `cell_list` section.

Mohammed24699, 2024-07-22:
Hello,

Howcan we do s1 handover in 5G. where can i find the confile file of gnb for s1 handover.

Mohammed24699, 2024-07-22:
Hello,
How can we do s1 handover in 5G. where can i find the configuration file of gnb for s1 handover.