srsran/srsRAN_Project #560: Redmi note12 turbo cant search the gnb

Sticcolet, 2024-04-18:
## Issue Description ##
I try to connect my Redmi note12 turbo to my srsgnb but find it cant find the carrier. However I have successfully connected my Redmi k60 with the same setup. Does that means  Redmi note12 does not support srsRAN? or something error with the SIM card？
## Setup Details ##
Ubuntu 22.04 Desktop
USRPX310 with external GPS 
PLMN：00101
band 1 with 10Khz bandwidth
SCS：15kHz

## Expected Behavior ##
COTS UE connected

## Actual Behaviour ##
COTS UE even cant find the Carrier

pgawlowicz, 2024-04-18:
Hi, 
not sure about Redmi note 12 turbo, but Redmi note 12 was reported to work correctly [here](https://github.com/srsran/srsRAN_Project/discussions/36#discussioncomment-8558838)

Could you share the gnb logs?

Sticcolet, 2024-04-19:
here is my logs with debug level and gnb setup,
[gnb.log](https://github.com/srsran/srsRAN_Project/files/15032014/gnb.log)
[gnb_rf_x310_fdd_n1_O.yml.txt](https://github.com/srsran/srsRAN_Project/files/15032021/gnb_rf_x310_fdd_n1_O.yml.txt)
 but i dont think it is useful because the main problem is that the phone cant display carrier like (00101)5G. I tried other Huawei mate40 yesterday and successed. Maybe Redmi note 12 turbo do not support the band I use?

Sticcolet, 2024-04-19:
> Hi, not sure about Redmi note 12 turbo, but Redmi note 12 was reported to work correctly [here](https://github.com/srsran/srsRAN_Project/discussions/36#discussioncomment-8558838) Maybe you need to force 5G?
> 
> Could you share the gnb logs?

now i tried several other bands and other phones. First both Redmi note 12 turbo and note 9 cant see the carrier whether in band n1, n3 or n78. but Huawei successed. I notice that both Redmi note 12 turbo and note 9 use MIUI14, although one of them is Android13 and the other is Android12. The successfully connected Redmi K60 use HyperOS and Huawei use Android 10. Perhaps the problem lies here with MIUI14? Do you know what system the reported Redmi note 12 use?

pgawlowicz, 2024-04-19:
You can also try a different PLMN like 90170

Sticcolet, 2024-04-19:
> You can also try a different PLMN like 90170

I tried 46001 and 46000 and both failed.

pgawlowicz, 2024-04-19:
Could you try the gNB PLMN and IMSI combinations from this table?
https://docs.srsran.com/projects/project/en/latest/knowledge_base/source/cots_ues/source/index.html

pgawlowicz, 2024-04-19:
Please check those two comments:

https://github.com/srsran/srsRAN_Project/discussions/36#discussioncomment-5466323
https://github.com/srsran/srsRAN_Project/discussions/36#discussioncomment-8558838

ninjab3s, 2024-06-12:
As @pgawlowicz mentioned before you maybe need to force NR. Type *#*#4636#*#*, you will be promted a menu, under "preferred network type" select "NR only" and try again. Have you tried that? It can also be that the UE does not support the band you want to use. Unfortunately, we cannot give support for specific UE models. Please make sure that the exact model you have does support the band before proceeding. 

The 46000/46001 are Chinese PLMNs. Are you sure that your UE connects to these PLMNs? Please insert a regular SIM of your location, try to connect to a regular network and reuse this PLMN in your config. Make sure not to interfere with local providers testing this - a shielded environment is advised!

Sticcolet, 2024-06-13:
In the end, I connected to GNB, but I don't know which specific settings worked. I can only list all possible settings here, hoping to provide some help.
1. Set force 5G
2. Set up the use of SA/NSA
3. Turn off 5G intelligent switching
4. Enable the secondary card SA network
5. Turn off Volte calls
I guess the last three points are more important because I should have tried the first two before

ninjab3s, 2024-06-13:
Glad to hear that!

closed 2024-06-13 by ninjab3s