srsran/srsRAN_Project #569: Missing U-Plane messages from RU to DU

shopon16, 2024-04-22:
## Issue Description ##
RU gets Section Type 1 and Type 3 C-Plane messages From DU. But RU does not send U-Plane messages to DU

## Setup Details ##
```
amf:
  addr: 192.168.88.10                                               # The address or hostname of the AMF.
  bind_addr: 192.168.88.233                                            # A local IP that the gNB binds to for traffic from the AMF.

ru_ofh:
  ru_bandwidth_MHz: 100                                           # RU instantaneous bandwidth.
  t1a_max_cp_dl: 514                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 124                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 442                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 86                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 514                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 130                                                  # Minimum T1a on User-Plane in microseconds.
  ta4_max: 170                                                    # Maximum Ta4 on User-Plane in microseconds.
  ta4_min: 86                                                     # Minimum Ta4 on User-Plane in microseconds.                                          # Minimum Ta4 on User-Plane in microseconds.
  is_prach_cp_enabled: true                                      # Configures if Control-Plane messages should be used to receive PRACH messages.
  ignore_ecpri_payload_size: true
  compr_method_ul: bfp                                            # Uplink compression method.
  compr_bitwidth_ul: 8                                            # Uplink IQ samples bitwidth after compression.
  compr_method_dl: bfp                                            # Downlink compression method.
  compr_bitwidth_dl: 8                                            # Downlink IQ samples bitwidth after compression.
  compr_method_prach: bfp                                         # PRACH compression method.
  compr_bitwidth_prach: 8                                         # PRACH IQ samples bitwidth after compression.
  enable_ul_static_compr_hdr: true                                # Configures if the compression header is present for uplink User-Plane messages (false) or not present (true).
  enable_dl_static_compr_hdr: true                                # Configures if the compression header is present for downlink User-Plane messages (false) or not present (true).
  compr_bitwidth_prach: 16
  iq_scaling: 0.8                                               # IQ samples scaling factor applied before compression, should be a positive value smaller than 10.
  cells:
    - network_interface: enp1s0f1                                 # Ethernet interface name used to communicate with the RU.
      ru_mac_addr: 0C:83:CC:0F:E9:04                              # RU MAC address.
      du_mac_addr: 98:b7:85:1e:e7:f0                              # DU MAC address.
      vlan_tag: 564                                                 # VLAN tag value.
      prach_port_id: [0, 1, 2, 3]                                          # PRACH eAxC port value.
      dl_port_id: [0]                                          # Downlink eAxC port values.
      ul_port_id: [0]                                          # Uplink eAxC port values.

cell_cfg:
  dl_arfcn: 641676                                                # ARFCN of the downlink carrier (center frequency).
  band: 78                                                        # The NR band.
  channel_bandwidth_MHz: 100                                      # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                                                  # Subcarrier spacing in kHz used for data.
  plmn: "31501"                                                   # PLMN broadcasted by the gNB.
  tac: 1                                                          # Tracking area code (needs to match the core configuration).
  pci: 1                                                          # Physical cell ID.
  nof_antennas_dl: 1                                              # Number of transmission antennas.
  nof_antennas_ul: 1                                              # Number of reception antennas.
  prach:
    prach_config_index: 159                                       # PRACH configuration index.
    prach_root_sequence_index: 1                                  # PRACH root sequence index.
    zero_correlation_zone: 0                                      # Zero correlation zone.
    prach_frequency_start: 12                                     # Offset in PRBs of lowest PRACH transmission occasion in frequency domain respective to PRB 0.

log:
  filename: /var/log/srsran/gnb.log
  all_level: info                                              # Logging level applied to all layers.

pcap:
  ngap_filename: /tmp/gnb_ngap.pcap     # Optional TEXT (/tmp/gnb_ngap.pcap). Path for NGAP PCAPs. 
  ngap_enable: true                    # Optional BOOL (false). Enable/disable NGAP packet capture.
  e1ap_filename: /tmp/gnb_e1ap.pcap     # Optional TEXT (/tmp/gnb_e1ap.pcap). Path for E1AP PCAPs. 
  e1ap_enable: true                    # Optional BOOL (false). Enable/disable E1AP packet capture.
  f1ap_filename: /tmp/gnb_f1ap.pcap     # Optional TEXT (/tmp/gnb_f1ap.pcap). Path for F1AP PCAPs. 
  f1ap_enable: true                    # Optional BOOL (false). Enable/disable F1AP packet capture.
  rlc_filename: /tmp/gnb_rlc.pcap       # Optional TEXT (tmp/gnb_rlc.pcap). Path for RLC PCAPs. 
  rlc_rb_type: all                      # Optional TEXT. Sets the RLC PCAP RB type. Supported: [all, srb, drb].
  rlc_enable: true                     # Optional BOOL (false). Enable/disable RLC packet capture.  
  mac_filename: /tmp/gnb_mac.pcap       # Optional TEXT (/tmp/gnb_mac.pcap). Path for MAC PCAPs. 
  mac_type: udp                         # Optional TEXT (udp). Sets the MAC PCAP pcap type. Supported: [DLT or UDP].
  mac_enable: false                     # Optional BOOL (false). Enable/disable MAC packet capture.
  e2ap_filename: /tmp/gnb_e2ap.pcap     # Optional TEXT (/tmp/gnb_e2ap.pcap). Path for E2AP PCAPs. 
  e2ap_enable: true                    # Optional BOOL (false). Enable/disable E2AP packet capture.
  gtpu_filename: /tmp/gnb_gtpu.pcap     # Optional TEXT (/tmp/gnb_gtpu.pcap). Path for GTPU PCAPs. 
  gtpu_enable: true                    # Optional BOOL (false). Enable/disable GTPU packet capture.
```
## Expected Behavior ##
DU should get U-Plane messages from RU

## Actual Behaviour ##
DU is continuously sending Type 1 Type 3 C-Plane messages to RU

## Additional Information ##
Here is the related files.
gnb log:
 [gnb.log](https://github.com/srsran/srsRAN_Project/files/15061478/gnb.log)
pcap:
[gnb_.zip](https://github.com/srsran/srsRAN_Project/files/15061522/gnb_.zip)

hudouseko, 2024-04-22:
Hi, I would expect uplink U-plane to be present only when UE wants to send something and is scheduled that uplink transmission is scheduled by gNB. The only exception being PRACH, though in your gnb.log I don't see any successful random access procedure.
Apart from that it is not clear to me, why you have 4 PRACH eAxCIds configured in 1T1R system.

shopon16, 2024-04-22:
> Hi, I would expect uplink U-plane to be present only when UE wants to send something and is scheduled that uplink transmission is scheduled by gNB. The only exception being PRACH, though in your [...]

We are using 1T1R for testing purposes. UE cant see the network with different antenna configuration. Everytime logs appear same

andrepuschmann, 2024-04-22:
What RU is that? Please share a PCAP of the fronthaul traffic. You can capture that with `tcpdump`.

shopon16, 2024-04-23:
> What RU is that? Please share a PCAP of the fronthaul traffic. You can capture that with `tcpdump`.

[testdu.zip](https://github.com/srsran/srsRAN_Project/files/15072783/testdu.zip)

andrepuschmann, 2024-04-23:
You have a MAC address misconfig - the only traffic from the RU in the PCAP is the epcri event indication.

shopon16, 2024-04-23:
> You have a MAC address misconfig - the only traffic from the RU in the PCAP is the epcri event indication.
here is my actual config in RU
```
DBG_RUN=0                                                                               
DBG_DU_MAC_HIGH=0x98b7851e                                                              
DBG_DU_MAC_LOW=0x0000e7f0                                                               
DBG_RU_MAC_HIGH=0x0c83cc0f                                                              
DBG_RU_MAC_LOW=0x0000e904                                                               
OP_DU_MAC=98b7851ee7f0                                                                  
OP_RU_MAC=0c83cc0fe904                                                                  
TCP_IP_ADDRESS=192.168.13.22                                                            
TCP_MASK=255.255.255.0                                                                  
VLAN_IP_ADDR_MASK=10.101.131.128/24                                                     
VLAN_ID=564                                                                             
centerFreq=3625140000                                                                   
AutoBringup=1                                                                           
TX_POWER=23                                                                             
RX_STRENGTH=20                                                                          
CLOCK_ID_MSB=0x00A12EE3                                                                 
CLOCK_ID_LSB=0xEE221A38                                                                 
AGC_ENABLE=0                                                                            
FREQ_KHz=3625140                                                                        
BW_Hz=100000000                                                                         
BITWIDTH=8                                                                              
SCS=1                                                                                   
u8hiPwrRU_Flag=0                                                                        
u8PhaseComp_Flag=0                                                                      
s8Sync_Flag=1   
```

ven457, 2024-04-30:
Hello Guys, 

Is the above issue resolved? If yes, please suggest how did we resolve it? 
I am facing the same issue in my setup with a third party O-RU.

ismagom, 2024-05-15:
I will close this issue due to inactivity. Please reopen again if the issue persists

closed 2024-05-15 by ismagom