srsran/srsRAN_Project #585: Error while installing libzmq

panitsasi, 2024-04-28:
while i was trying to download the libzmq following these commands:
 git clone https://github.com/zeromq/libzmq.git
cd libzmq
./autogen.sh
./configure
make
this issue came up (while executing the command make): 
[image]

Do you have an idea, what is the issue here ?

pgawlowicz, 2024-04-29:
did you try to install with `sudo apt-get install libzmq3-dev` ?

panitsasi, 2024-04-29:
Yes, i followed these instructions: 
[image]
and the issue persists.

pgawlowicz, 2024-04-29:
which OS do you use?

panitsasi, 2024-04-29:
Ubuntu 22.04.1 LTS

panitsasi, 2024-04-30:
no, i tried to re install the package, but again i had the same issue.

pgawlowicz, 2024-04-30:
could you check if you can install in a docker container?

1. Start docker container with ubuntu 22.04:
```
docker run -it ubuntu:22.04 /bin/bash
```
2. Then inside the container install zmq:
```
apt update
apt-get install libzmq3-dev
```

panitsasi, 2024-04-30:
Yes,  i can install in a docker container, but do you know why is this happening ? I had installed in the past, but as i remember i deleted it. Is this is a reason to consider ?

panitsasi, 2024-05-01:
Okay, I reinstall all the packages from the beginning and now everything works fine.

closed 2024-05-02 by pgawlowicz

abcd11abcd, 2024-05-06:
im also facing the same issue. How did you resolve? i tried removing libzmq3-dev and installing again. Still im facing the same problem. You resolved it by using docker container?

panitsasi, 2024-05-06:
Hello, i removed all the installed libraries (and the dependencies) and then it worked.

burnbrigther, 2025-01-28:
> Hello, i removed all the installed libraries (and the dependencies) and then it worked.
Which libraries did you have to remove?  I continue to struggle with this.  Thanks