srsran/srsRAN_Project #587: Could not finish setup request. Deallocating dedicatedInfoNAS PDU

Sticcolet, 2024-04-29:
## Issue Description ##

I am trying to use srsUE and srsRAN, but due to hardware limitations, I have to use 10M Bandwidth. However, at a bandwidth of 10M, srsUE cannot select cells. However, using the same configuration with a bandwidth of 20M, srsUE can successfully select a cell, but will report an error due to overflow.
In fact, a few weeks ago, I did try to successfully connect srsUE and srsRAN at 10M bandwidth by changing the configuration, but I lost the configuration file I used at the time. Can someone help me?

## Setup Details ##

The following are my conf files and logs for 10M and 20M bandwidth. And I used an external GPS clock source.
[gnb.zip](https://github.com/srsran/srsRAN_Project/files/15146784/gnb.zip)
[ue.zip](https://github.com/srsran/srsRAN_Project/files/15146786/ue.zip)

pgawlowicz, 2024-04-29:
Please try `coreset0_index: 6` for 10MHz BW

pgawlowicz, 2024-04-29:
It has to be 6 in case of BW=10MHz and srsUE.

also, try:
```
  clock: default
  sync: external
```

pgawlowicz, 2024-04-29:
could you share the UE logs?

Sticcolet, 2024-04-29:
> could you share the UE logs?

I have included the logs of UE at 20M and 10M bandwidth in ue.zip, where the number in the file name represents the different sampling rates used.

pgawlowicz, 2024-04-29:
Please share logs for `coreset0_index: 6` and 10mhz.

Sticcolet, 2024-04-29:
In the end, I solved this problem by reducing the rx_gain on the UE side to below 10. I don't understand why, but srsUE can indeed establish an RCC connection now. But now I am facing a new problem, the RCC connection will immediately disconnect, and the GNB side will only display communication once with a 100% error rate. I will try to deal with this issue tomorrow.

pgawlowicz, 2024-04-29:
please add the following block to your gnb config, otherwise the UE will be quickly released when there is no traffic:
```
cu_cp:
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].
```

closed 2024-04-29 by Sticcolet