srsran/srsRAN_Project #589: RRC released for ZMQ setup

hududed, 2024-04-29:
## Issue Description ##
RRC is released, suspecting PDU session took too long to establish

## Setup Details ##
ZMQ-based setup, 1 gNB 1 UE single host-machine Ubuntu 22.04
open5GS core locally compiled following srsRAN_Project config (TAC, PLMN, IP etc.)

## Expected Behavior ##
PDU established and RRC connected without release, e.g. 
```
...
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=2094
Random Access Complete.     c-rnti=0x4602, ta=0
RRC Connected
PDU Session Establishment successful. IP: 10.45.1.2
RRC NR reconfiguration successful.
```

## Actual Behaviour ##
```
❯ sudo srsue ue_zmq.conf
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue_zmq.conf...

Built in Release mode using commit ec29b0c1f on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://127.0.0.1:2000
CH0 tx_port=tcp://127.0.0.1:2001
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=654
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
Received RRC Release
```

## Steps to reproduce the problem ##
Follow srsRAN_Project tutorial for ZMQ-based setup, except compiling open5gs locally (the 5gc docker didnt work for me see https://github.com/srsran/srsRAN_Project/issues/500)
I included the config files and logs of gnb connecting to core, ue connecting to gnb and ue release
[ue_zmq.txt](https://github.com/srsran/srsRAN_Project/files/15155412/ue_zmq.txt)
[gnb_zmq.txt](https://github.com/srsran/srsRAN_Project/files/15155414/gnb_zmq.txt)
[ue.log](https://github.com/srsran/srsRAN_Project/files/15155447/ue.log)
[gnb.log](https://github.com/srsran/srsRAN_Project/files/15155448/gnb.log)

pgawlowicz, 2024-04-29:
Please add the following block to your gnb config.

```
cu_cp:
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds.
```

hududed, 2024-04-29:
hm so I did add it but it still fails, but not sure if it actually registered that timer:
```
2024-04-29T22:40:37.540227 [PHY     ] [I] [   384.0] PUCCH: rnti=0x4601 format=1 prb1=105 prb2=na symb=[0, 14) cs=0 occ=1 sr=no t=44.5us
2024-04-29T22:40:37.546712 [PHY     ] [I] [   384.6] PUCCH: rnti=0x4601 format=2 prb=[104, 105) prb2=na symb=[10, 12) csi1=1111 t=42.5us
2024-04-29T22:40:37.561303 [PHY     ] [I] [   386.0] PUCCH: rnti=0x4601 format=1 prb1=105 prb2=na symb=[0, 14) cs=0 occ=1 sr=no t=47.2us
2024-04-29T22:40:37.567757 [PHY     ] [I] [   386.6] PUCCH: rnti=0x4601 format=2 prb=[104, 105) prb2=na symb=[10, 12) csi1=1111 t=40.4us
2024-04-29T22:40:37.570837 [NGAP    ] [W] ue=0 ran_ue_id=0 amf_ue_id=1: PDU session setup timer expired after 3000ms. Requesting UE release
2024-04-29T22:40:37.570845 [NGAP    ] [I] ue=0 ran_ue_id=0 amf_ue_id=1: Scheduling transmission of UeContextReleaseRequest
2024-04-29T22:40:37.570847 [NGAP    ] [I] ue=0 ran_ue_id=0 amf_ue_id=1: Sending UeContextReleaseRequest
2024-04-29T22:40:37.571049 [NGAP    ] [I] ue=0 ran_ue_id=0 amf_ue_id=1: Received UeContextReleaseCommand

```

The `cu_cp` is simply appended to gnb_zmq.yaml right?

pgawlowicz, 2024-04-30:
could you share open5gs logs?

hududed, 2024-04-30:
[ue.log](https://github.com/srsran/srsRAN_Project/files/15167393/ue.log)
[gnb.log](https://github.com/srsran/srsRAN_Project/files/15167394/gnb.log)
core [all.log](https://github.com/srsran/srsRAN_Project/files/15167395/all.log)

```
04/30 09:45:11.054: [gmm] WARNING: [imsi-001010123456780] DNN Not Supported OR Not Subscribed in the Slice (../src/amf/gmm-handler.c:1272)
04/30 09:45:11.077: [gmm] INFO: [imsi-001010123456780] No GUTI allocated (../src/amf/gmm-sm.c:1591)
```

That might be a culprit? I did add the subscriber correctly I hope:
[image]

hududed, 2024-05-01:
The issue is in the subscriber config, in case anyone struggled with this:
[image]

closed 2024-05-01 by hududed