srsran/srsRAN_Project #595: Error trying to set UEs with 25 PRBs

mas-oliveira, 2024-05-02:
## Issue Description ##
Hello,
I'm trying to collect some data using different number of PRBs. Currently I'm only launching 3 UEs and collecting some KPMs using SC-RIC. 
I was able to do everything with the default configs (PRBs = 52; SRate = 11.52MHz; gNB BW = 10MHz). Now I'm trying to do the same thing but with 25 PRBs (in all UEs at the moment), sample rate = 5.76MHz and gNB BW = 5MHz.

## Setup Details ##
[ue_confs.zip](https://github.com/srsran/srsRAN_Project/files/15186965/ue_confs.zip)
The gNB executable is according with the SCRIC tutorial
[gnb_zmq.zip](https://github.com/srsran/srsRAN_Project/files/15186983/gnb_zmq.zip)

## Actual Behaviour ##
When I try to connect 3 UEs they just disconnect with the message RRC Release. I'm using the most basic grc simulation just to make sure that this isn't related to any error of some modifications that I've been doing.
[default.zip](https://github.com/srsran/srsRAN_Project/files/15187008/default.zip)

> Current sample rate is 5.76 MHz with a base rate of 5.76 MHz (x1 decimation)
Current sample rate is 5.76 MHz with a base rate of 5.76 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=42, ra-rnti=0x39, tti=174
Random Access Complete.     c-rnti=0x4602, ta=0
RRC Connected
Received RRC Release

## Additional Information ##
I'm attaching in logs.zip the relevant logs and packet captures generated during the process. 
[logs.zip](https://github.com/srsran/srsRAN_Project/files/15187051/logs.zip)

Thanks

pgawlowicz, 2024-05-02:
Hi, are you sure the Open5gs's subscriber database is correct?
UEs have the following issues in their logs:
```
2024-05-02T10:39:51.832275 [NAS5G  ] [I] Handling Authentication Reject
...
2024-05-02T10:39:51.833681 [RRC-NR ] [I] Received RRC Release
```

mas-oliveira, 2024-05-02:
I'm getting these logs on open5gs:

> open5gs_5gc       | 05/02 11:11:33.279: [gmm] WARNING: Authentication failure(MAC failure) (../src/amf/gmm-sm.c:1347)
open5gs_5gc       | 05/02 11:11:33.279: [amf] WARNING: [suci-0-001-01-0000-0-0-0123456791] Authentication reject (../src/amf/nas-path.c:538)
open5gs_5gc       | 05/02 11:11:33.307: [gmm] WARNING: Authentication failure(MAC failure) (../src/amf/gmm-sm.c:1347)
open5gs_5gc       | 05/02 11:11:33.307: [amf] WARNING: [suci-0-001-01-0000-0-0-0123456780] Authentication reject (../src/amf/nas-path.c:538)
open5gs_5gc       | 05/02 11:11:33.313: [gmm] WARNING: Authentication failure(MAC failure) (../src/amf/gmm-sm.c:1347)
open5gs_5gc       | 05/02 11:11:33.314: [amf] WARNING: [suci-0-001-01-0000-0-0-0123456790] Authentication reject (../src/amf/nas-path.c:538)

But it's weird because I haven't changed anything in the usim configurations of the UEs and I tried to run again the add_users.py inside the docker image of open5gs and it didn't update anything.
I have this subscriber_db and I guess that it matches the UE configs.
[subscriber_db.csv](https://github.com/srsran/srsRAN_Project/files/15187502/subscriber_db.csv)

pgawlowicz, 2024-05-02:
Could you check Open5gs GUI interface to see what is the database content of the currently running instance?

mas-oliveira, 2024-05-02:
[image]
[image]

It looks that is all according to the subscriber_db.csv

pgawlowicz, 2024-05-02:
in your subscriber_db.csv you have:
`opc=63bfa50ee6523365ff14c1f45f88737d`, while the UEs have 
`opc=63bfa50ee6253365ff14c1f45f88737d`

mas-oliveira, 2024-05-02:
Thanks, everything is working again now!

closed 2024-05-02 by mas-oliveira