srsran/srsRAN_Project #599: Failed to connect SCTP socket to 127.0.0.100:38412. error="Connection refused" timeout=0ms srsGNB ERROR: Failed to create SCTP gateway.

sambak22, 2024-05-03:
## Issue Description ##
Failed to connect SCTP socket using USRP B205 Mini

## Setup Details ##
I have Ubuntu 22.04 installed as OS in separate partition. I am using USRP B205 Mini. AMF daemon service is running, however, open5gs core is unable to connect to gnB and giving error of SCTP failed; connection refused. I have followed srsRAN Tutorial step-by-step but the issue persists. I am unable to find way ahead. 

## Exp--== srsRAN gNB (commit 374200dee) ==--

Connecting to AMF on 10.53.1.2:38412
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200'
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz...
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 23.040000 MHz...
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Cell pci=1, bw=20 MHz, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHzected Behavior ##

## Actual Behaviour ##
 sudo ./gnb -c gnb_rf_b200_tdd_n78_20mhz.yml 

The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in dual executor mode.

--== srsRAN gNB (commit 2f90c8b60) ==--

Connecting to AMF on 127.0.0.100:38412
Failed to connect SCTP socket to 127.0.0.100:38412. error="Connection refused" timeout=0ms
srsGNB ERROR: Failed to create SCTP gateway.

## Steps to reproduce the problem ##
Config .yaml files are attached for your reference.

## Additional Information ##
[gnb.yaml.txt](https://github.com/srsran/srsRAN_Project/files/15204819/gnb.yaml.txt)
[amf.yaml.txt](https://github.com/srsran/srsRAN_Project/files/15204820/amf.yaml.txt)
[upf.yaml.txt](https://github.com/srsran/srsRAN_Project/files/15204821/upf.yaml.txt)

dhiaboujebha, 2024-05-07:
@sambak22 did you set up the ip adresses that you are using as amf addr and bind addr as static ip adresses in the gnb and core computers?

pgawlowicz, 2024-05-13:
you use the same IP for `addr` and `bind_addr`.
```
amf:
  addr: 127.0.0.100                                             # The address or hostname of the AMF.
  bind_addr: 127.0.0.100                                          # A local IP that the gNB binds to for traffic from the AMF.
```

pgawlowicz, 2024-06-05:
@sambak22 any update on this issue?

sambak22, 2024-06-05:
Hello!
I could resolve the issue, however, I am unable to connect UE to open5gs
5GC via USRP B205 Mini. Following are the setup details:
UE - Samsung Mobile (5G)
SIM - USIM with following parameters programmed on the SIM:

USIM parameters:
00101000000047

"K-start": ["0x34", "0xb7", "0xd5", "0xee", "0xe1", "0x1a", "0x21", "0x62",
"0xbb", "0x20", "0xf3", "0xc4", "0xe5", "0x77", "0x36", "0x74"],

"OP": ["0x63", "0xbf", "0xa5", "0x0e", "0xe6", "0x52", "0x33", "0x65",
"0xff", "0x14", "0xc1", "0xf4", "0x5f", "0x88", "0x73", "0x7d"],

OPC: CC BA 74 55 FA 47 30 62 0A 28 26 45 BA 66 92 F0

ISIM parameters:

MSISDN: 5432100001 to 5432100022

IMPI-> ***@***.*** to
***@***.***

IMPU-> ***@***.*** to
***@***.***

Domain-> ims.mnc001.mcc001.3gppnetwork.org

K: ["0x34", "0xb7", "0xd5", "0xee", "0xe1", "0x1a", "0x21", "0x62", "0xbb",
"0x20", "0xf3", "0xc4", "0xe5", "0x77", "0x36", "0x74"],

OP: ["0x63", "0xbf", "0xa5", "0x0e", "0xe6", "0x52", "0x33", "0x65",
"0xff", "0x14", "0xc1", "0xf4", "0x5f", "0x88", "0x73", "0x7d"]

OPC: CC BA 74 55 FA 47 30 62 0A 28 26 45 BA 66 92 F0

gNB - USRP B205 Mini

O-RAN - srsRAN 5G

5GC - open5gs

Kindly please help me to connect UE to open5gs via USRP B205 Mini and guide
me through steps to be followed to configure each module.

Thank you & Regards 🙏

pgawlowicz, 2024-06-05:
please follow this tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html

sambak22, 2024-06-09:
Sir,
I could get half way success while attaching UE to open5gs through USRP. UE
is detecting PLMN (00101) but is unable to register UE to the core network.
Following is the error message being displayed in the console:

 Registration request (../src/amf/gmm-sm.c:1061)
06/08 13:47:56.463: [gmm] INFO: [suci-0-001-01-0-0-0-0000000037] SUCI
(../src/amf/gmm-handler.c:157)
06/08 13:47:57.004: [gmm] ERROR: Cannot find Requested NSSAI [1]
(../src/amf/gmm-handler.c:420)
06/08 13:47:57.004: [gmm] ERROR: PLMN_ID[MCC:1 MNC:1]
(../src/amf/gmm-handler.c:423)
06/08 13:47:57.005: [gmm] ERROR: S_NSSAI[SST:1 SD:0x0]
(../src/amf/gmm-handler.c:426)
06/08 13:47:57.005: [gmm] ERROR: [suci-0-001-01-0-0-0-0000000037]
gmm_handle_security_mode_complete() failed [62] in type [65]
(../src/amf/gmm-sm.c:1686)

Also, please find attached screenshot for your reference.

Waiting for your early reply.

Thank you & best regards
Sameer Bakoray

pgawlowicz, 2024-06-10:
It seems there is some problem with slice configuration.

sambak22, 2024-06-10:
Sir, 
Please guide me to resolve the issue. Kindly please give step by step procedure to attach UE to open5gs through USRP B205 Mini. I am able to see the network in UE but unable to register to the core network. I have attached the logs and video for your reference.

Thank you, in anticipation,
Regards 
[image]
[image]
[image]
[image]
[image]

pgawlowicz, 2024-06-11:
seems that the IMSI number in you subscriber database is missing one leading 0?

amitpalkar, 2024-06-11:
I am also facing similar issue . 
--== srsRAN gNB (commit f3ed07a5a) ==--

Connecting to AMF on 10.53.1.2:38412
Failed to connect SCTP socket to 10.53.1.2:38412. error="Connection timed out" timeout=1371ms
srsGNB ERROR: Failed to create SCTP gateway.

I am attaching my files. 
[amf.zip](https://github.com/user-attachments/files/15792367/amf.zip)
@sambak22  How did you resolve the issue
@pgawlowicz can you please let me know if any config issue is present.  I am trying to have the srs gNB to communicate with the 5GS core.  the gNB is unable to connect to AMF.

pgawlowicz, 2024-06-11:
@amitpalkar i have answered your issue here: https://github.com/srsran/srsRAN_Project/discussions/666

sambak22, 2024-06-11:
Sir,

How do I add IMSI data to subscriber database and get UE attached to the
core? Please help me. I am just a step away to register UE to core.
Probably, I am committing some minor error while following the procedure.
Ultimately, I wish to attach 02 x UEs to the core and establish voice
communication between the two UEs over USRP and open5gs.
Please guide me. I am sure you can give step by step guideline to
achieve the above scenario.

Thank you, in anticipation,
Regards

sambak22, 2024-06-12:
> @sambak22 <https://github.com/sambak22> please follow this tutorial:

Sir, I have followed the tutorial multiple times but have failed to attach
the UE to the core. That is the only part left of entire setup of open5gs,
srsRAN using USRP.
Please help me to troubleshoot this issue.

Regards

pgawlowicz, 2024-06-19:
to add UEs to the Open5gs database (dockerized version) please check the readme here: https://github.com/srsran/srsRAN_Project/tree/main/docker/open5gs

closed 2024-06-28 by pgawlowicz