srsran/srsRAN_Project #604: No U-plane packets from RU to DU

ven457, 2024-05-06:
## Issue Description ##
RU gets Section Type 1 and Type 3 C-Plane messages From DU. But RU does not send U-Plane messages to DU. 

## Setup Details ##
gnb config:
# This example configuration outlines how to configure the srsRAN Project CU/DU to use an O-RU and split 7.2. This is specifically for use
# with the Benetel R550 RU. This config will create a single TDD MIMO cell transmitting in band 78, with 100 MHz bandwidth and 30 kHz sub-carrier-spacing.
# The parameters used to configure the RU are found in the `ru_ofh` sub-section. This configuration makes used of the OFH Lib from SRS to enable split 7.2.

amf:
  addr: 192.168.70.220                                            # The address or hostname of the AMF.
  bind_addr: 192.168.70.12                                        # A local IP that the gNB binds to for traffic from the AMF.

ru_ofh:
  ru_bandwidth_MHz: 10                                            # RU instantaneous bandwidth.
  t1a_max_cp_dl: 577                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 505                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 410                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 245                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 452                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 380                                                 # Minimum T1a on User-Plane in microseconds.
  ta4_max: 480                                                    # Maximum Ta4 on User-Plane in microseconds.
  ta4_min: 70                                                     # Minimum Ta4 on User-Plane in microseconds.
  is_prach_cp_enabled: true                                       # Configures if Control-Plane messages should be used to receive PRACH messages.
  compr_method_ul: bfp                                            # Uplink compression method.
  compr_bitwidth_ul: 9                                            # Uplink IQ samples bitwidth after compression.
  compr_method_dl: bfp                                            # Downlink compression method.
  compr_bitwidth_dl: 9                                            # Downlink IQ samples bitwidth after compression.
  compr_method_prach: bfp                                         # PRACH compression method.
  compr_bitwidth_prach: 9                                         # PRACH IQ samples bitwidth after compression.
  enable_ul_static_compr_hdr: false                               # Configures if the compression header is present for uplink User-Plane messages (false) or not present (true).
  enable_dl_static_compr_hdr: false                               # Configures if the compression header is present for downlink User-Plane messages (false) or not present (true).
  iq_scaling: 0.95                                                # IQ samples scaling factor applied before compression, should be a positive value smaller than 10.
  cells:
    - network_interface: enp175s0f0                               # Ethernet interface name used to communicate with the RU.
      ru_mac_addr: ec:79:49:f8:d9:55                              # RU MAC address.
      du_mac_addr: 3c:fd:fe:ed:21:08                              # DU MAC address.
      vlan_tag: 202                                               # VLAN tag value.
      prach_port_id: [4]                                       # PRACH eAxC port value.
      dl_port_id: [1]                                    # Downlink eAxC port values.
      ul_port_id: [1]                                    # Uplink eAxC port values.

cell_cfg:
  dl_arfcn: 129400                                                # ARFCN of the downlink carrier (center frequency).
  band: 71                                                        # The NR band.
  channel_bandwidth_MHz: 10                                       # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                                                  # Subcarrier spacing in kHz used for data.
  plmn: "00101"                                                   # PLMN broadcasted by the gNB.
  tac: 1                                                          # Tracking area code (needs to match the core configuration).
  pci: 140                                                        # Physical cell ID.
  nof_antennas_dl: 1                                              # Downlink number of antennas.
  nof_antennas_ul: 1                                              # Uplink number of anntennas.
  pucch:
   min_k1: 2
  pusch:
   min_k2: 2
  prach:
    prach_config_index: 16                                         # PRACH configuration index.
    prach_root_sequence_index: 1                                  # PRACH root sequence index.
    zero_correlation_zone: 2                                      # Zero correlation zone.
    prach_frequency_start: 0                                      # Offset in PRBs of lowest PRACH transmission occasion in frequency domain respect

log:
  filename: /tmp/gnb.log                                          # Path of the log file.
  fapi_level: info
  ofh_level: info
  all_level: debug                                                # Logging level applied to all layers.

pcap:
  mac_enable: false                                               # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap                                 # Path where the MAC PCAP is stored.
  ngap_enable: false                                              # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap                               # Path where the NGAP PCAP is stored.

## Expected Behavior ##
There should be U-plane packets from RU in response to DU C/U-plane packets

## Actual Behaviour ##
No U-plane packets from RU, however RU is sending M-plane & S-plane messages

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
PCAP file captured on srsDU:

ven457, 2024-05-08:
Attached the PCAP. All packets received at RU are ONTIME. But RU doesn't respond to DU with U-plane packets
[srs_gen1_tb_cisco.pcap.gz](https://github.com/srsran/srsRAN_Project/files/15252973/srs_gen1_tb_cisco.pcap.gz)

ven457, 2024-05-15:
Hello 
Any support from team on this issue?

andrepuschmann, 2024-05-15:
First of all it would be good if you could format the issue text correctly, explain what RU you use and your setup. It's very difficult to read your ticket.
Looking at the PCAP it seems you're using a Fujitsu RU. There are other users with those RUs so maybe check those tickets. The fundamental issue is that the the RU does not respond to the Cplane messages requesting PRACH. Our DU doesn't request UL Cplane unless there is an UE attached that requires it. So what you see is normal and expected. You need to figure out why your RU doesn't send PRACH uplane as requested by the DU.

wei-lou, 2024-05-15:
Hi Andre, actually Ven457 and us have been working on closely the same integration activities from different regions and looking from the different angles.

ven457, 2024-05-16:
Hi @andrepuschmann , 
We are also debugging why RU is not sending PRACH to DU. 
We observed that "framestructure","cpLength" fields in  in C-plane Section type3 msg sent by DU were incorrect compared to RU supported values. RU supports frameStructure=188, but DU sends 204. Can you please suggest us how can we change DU configuration to send framestructure to 188? 
**PCAP c-plane section type 3 msg:**
[image]
**RU PRACH configuration:**
[image]

andrepuschmann, 2024-05-16:
You probably have a mismatch in the PRACH configuration, hence the mismatched frame structure. You're not providing any insights into your RU configuration.

ven457, 2024-05-16:
Please let us know what information is needed from RU side.

ven457, 2024-05-17:
Hi @andrepuschmann , 
Please find attached gnb config file used and gnb log for your reference. 
We see following warnings in the gnb log.
2024-05-17T10:20:22.089488 [SCHED   ] [D] [   233.1] Processed slot events:
- Cell creation: idx=0 pci=140
2024-05-17T10:20:22.092635 [FAPI    ] [W] [   233.4] Real-time failure in FAPI: Received late DL_TTI.request from slot 233.1
2024-05-17T10:20:22.092651 [FAPI    ] [W] [   233.4] Real-time failure in FAPI: Received late UL_TTI.request from slot 233.1
2024-05-17T10:20:22.092675 [SCHED   ] [D] [   233.2] Processed slot events:
- ErrorIndication: slot=233.1, PDCCH discarded, PDSCH discarded
- ErrorIndication: slot=233.1, PUSCH and PUCCH discarded
2024-05-17T10:20:22.093131 [OFH     ] [W] Dropped late downlink resource grid in slot '233.2' and sector#0. No OFH data will be transmitted for this slot
2024-05-17T10:20:22.093276 [OFH     ] [W] Dropped late downlink resource grid in slot '233.3' and sector#0. No OFH data will be transmitted for this slot
2024-05-17T10:20:22.094026 [SCHED   ] [D] [   233.6] Processed slot events:
- ErrorIndication: slot=233.2, PDCCH discarded, PDSCH discarded
- ErrorIndication: slot=233.3, PDCCH discarded, PDSCH discarded

[gnb_17052024.log](https://github.com/srsran/srsRAN_Project/files/15347781/gnb_17052024.log)
[n71_10mhz_4x4_v7.txt](https://github.com/srsran/srsRAN_Project/files/15347782/n71_10mhz_4x4_v7.txt)

ven457, 2024-05-21:
Hi Team,
Any update on this?

andrepuschmann, 2024-05-21:
Hey @ven457 - what PRACH config do you use in the RU? In the gNB config you've commented the section but I doubt that the default values will match your RU config by chance.

ven457, 2024-05-21:
Hi @andrepuschmann , sorry above gNB config shared was with prach config commented out for testing. Please ignore.
Otherwise, we use below prach config at gNB side.
**gNB config:**
    prach_config_index 16                                         # PRACH configuration index.
    prach_root_sequence_index 1                                  # PRACH root sequence index.
    zero_correlation_zone 2                                      # Zero correlation zone.
    prach_frequency_start 0 

**RU side:**
<low-level-rx-endpoints>
<name>Low-Level-Rx-Endpoint-4</name>
<compression>
<compression-type>STATIC</compression-type>
<exponent>4</exponent>
<iq-bitwidth>9</iq-bitwidth>
<bitwidth>9</bitwidth>
</compression>
<frame-structure>188</frame-structure>
<cp-type>NORMAL</cp-type>
<cp-length>160</cp-length>
<cp-length-other>144</cp-length-other>
<offset-to-absolute-frequency-center>0</offset-to-absolute-frequency-center>
<number-of-prb-per-scs>
<scs>KHZ_1_25</scs>
<number-of-prb>70</number-of-prb>
</number-of-prb-per-scs>
<e-axcid>
<o-du-port-bitmask>65024</o-du-port-bitmask>
<band-sector-bitmask>384</band-sector-bitmask>
<ccid-bitmask>112</ccid-bitmask>
<ru-port-bitmask>15</ru-port-bitmask>
<eaxc-id>4</eaxc-id>
</e-axcid>
<non-time-managed-delay-enabled>false</non-time-managed-delay-enabled>
</low-level-rx-endpoints>

pgawlowicz, 2024-06-20:
@ven457 any update on this issue?

wei-lou, 2024-07-03:
With tweaking the UL timings, now this is pcap similar another working system. However we are still getting PRACH msg2 timeout.
[srs_tb_gen1_fft_4096_run15.pcap.gz](https://github.com/user-attachments/files/16090627/srs_tb_gen1_fft_4096_run15.pcap.gz)
[RU_n71_10mhz_4x4_v12_2.yml.txt](https://github.com/user-attachments/files/16090638/RU_n71_10mhz_4x4_v12_2.yml.txt)

closed 2024-10-16 by pgawlowicz