srsran/srsRAN_Project #612: UE Fales to attach to the 5G Network

Moussa-Guemdani, 2024-05-10:
## Issue Description ##
Hello,
I have built the End-To-End 5G Network following the tutorial ""https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html"", but I'm still unable to attach the UE.

## Setup Details ##
# Hardware: - RF:USRPx310.
                    - Ubuntu 24.04.04.
                    - Computer: Dell i5 OptiPlex 7060.
                    - UE: OnePlus8Pro (supports n78 SA).
                    - SysmoISIM2
# Configs:
                    - amf:
                        - addr: 127.0.0.5                                             
                        - bind_addr: 127.0.0.1                                    

                    - ru_sdr:
                        - device_driver: uhd                                            
                        - device_args: type=x300                                        
                        - clock: internal                                           
                        - sync: default                                                       
                        - srate: 23.04
                        - tx_gain: 23
                        - rx_gain: 20
                    - cell_cfg:
                        - dl_arfcn: 627340
                        - #627340
                        - band: 78
                        - channel_bandwidth_MHz: 20
                        - common_scs: 30
                        - plmn: "20870"
                        - tac: 1
                        - pci: 1 

- 5G core: open5Gs        

## Actual Behaviour ##
[image]
[image]

pgawlowicz, 2024-05-13:
did you run this script on the gnb PC? https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance

Also, could you share the details of the PC? (cpu, ram?)

Moussa-Guemdani, 2024-05-13:
Hello,
I have ran the script, but stil the same issue,

12 x CPU: 
             - cpu family	: 6
             - model		: 158
             - model name	: Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz
             - stepping	: 10
             - microcode	: 0xf4
             - cpu MHz		: 800.039
             - cache size	: 12288 KB
             - physical id	: 0
             - siblings	: 12
             
RAM(MB): 
                            total        used       free             shared      buff/cache     available
      Mem:            15Gi       5,1Gi       6,6Gi           269Mi          3,7Gi            9,7Gi
       Swap:          2,0Gi          0B       2,0Gi

pgawlowicz, 2024-05-13:
could you share the configs you use?

Moussa-Guemdani, 2024-05-13:
Here you are:
[image]

pgawlowicz, 2024-05-13:
could you check whether this works? 
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#usrp-x300-x310

Moussa-Guemdani, 2024-05-13:
These are the results after the changes
[image]

Configs: 
[image]

pgawlowicz, 2024-05-13:
how is the x310 connected to the gnb PC?

Moussa-Guemdani, 2024-05-13:
It is connected using the 10 Gigabit Ethernet  via SFP cable
[image]

Moussa-Guemdani, 2024-05-14:
Hello, When I Try the use another version of uhd, and  run the gnb, I asks me to download another version of uhd that is compatible with srsRAN_Project, Then when I do this and follow the steps, still have the same issue

[image]

Moussa-Guemdani, 2024-05-15:
[image]

pgawlowicz, 2024-05-16:
Could you try uhd-4.3 or higher?

Moussa-Guemdani, 2024-05-28:
Hello, after successfully using the project for several days, now the Ue Connects normally, but there is an exclamation mark on the network, and there is no internet access!

I have checked the nat and forwarding rules, 

[image]

Moussa-Guemdani, 2024-05-28:
> You may try these
> 
> ```
> 
> sudo sysctl -w net.ipv4.ip_forward=1
> sudo iptables -t nat -A POSTROUTING -s 10.45.0.0/16 ! -o ogstun -j MASQUERADE
> sudo iptables -I INPUT -i ogstun -j ACCEPT
> sudo ip ro add 10.45.0.0/16 via 10.53.1.2
> ```

HEllo,

I have tried, but the issue is not solved

pgawlowicz, 2024-05-29:
@Moussa-Guemdani do you run open5gs in docker or manually installed?

Moussa-Guemdani, 2024-05-29:
@pgawlowicz  I am running open5gsmanually installed

pgawlowicz, 2024-05-29:
did you follow this tutorial? https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html

Did you configure APN in your phone? Please check the troubleshooting section.

Moussa-Guemdani, 2024-05-30:
Yes, i did, also with the Access Point.

It worked for some time, after re-build, then Same issue.

pgawlowicz, 2024-06-05:
could you try this one:
```
sudo sysctl -w net.ipv4.ip_forward=1
sudo sysctl -w net.ipv4.conf.all.forwarding=1
sudo sysctl -w net.ipv6.conf.all.forwarding=1
sudo iptables -t nat -A POSTROUTING -o IFNAME_CONNECTED_TO_INTERNET -j MASQUERADE
sudo iptables -t nat -A POSTROUTING -s 10.45.0.0/16 ! -o ogstun -j MASQUERADE
sudo iptables -P FORWARD ACCEPT
sudo ip ro add 10.45.0.0/16 via 10.53.1.2
```

pgawlowicz, 2024-06-20:
@Moussa-Guemdani any update on this issue?

Moussa-Guemdani, 2024-06-20:
Hello,
It works very fine fine
Thank you for your help.

closed 2024-06-20 by pgawlowicz