srsran/srsRAN_Project #618: srsRAN gNB with srsUE over ZMQ, but not send PRACH

wzh8267, 2024-05-15:
## Issue Description ##
[Describe the issue in detail]
 two pc and two USRP, and the open5gs is running.  I installed  all the drivers needed , then  build the projects of  srsUE and srsRAN gNB, update the configs ZMQ model needed, then  run gnb,  and ue
## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
gnb log:
root@oai-ue:/home/long/wzh/513gnb/srsRAN_Project/build/apps/gnb# vi gnb_download_zmq.yaml
root@oai-ue:/home/long/wzh/513gnb/srsRAN_Project/build/apps/gnb# ./gnb -c gnb_download_zmq.yaml -c slicing.yml

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in executor blocking mode.

ue log:
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins:
Reading configuration file ue_download_zmq.yaml...

Built in Release mode using 23.04.0.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6
Supported RF device list: UHD zmq file
CHx base_srate=23.04e6
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://127.0.0.1:2000
CH0 tx_port=tcp://127.0.0.1:2001
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...

## Expected Behavior ##
[What you expect to happen]
 ue didn't send prach.
## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
ue_config:
[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 23.04e6
nof_antennas = 1

device_name = zmq
device_args = tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6

[rat.eutra]
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 3
nof_carriers = 1
max_nof_prb = 106
nof_prb = 106

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
opc  = 63BFA50EE6523365FF14C1F45F88737D
k    = 00112233445566778899aabbccddeeff
imsi = 460208960000006
imei = 353490069873319

[rrc]
release = 15
ue_category = 4

[nas]
apn = srsapn
apn_protocol = ipv4

[gw]
netns = ue1
ip_devname = tun_srsue
ip_netmask = 255.255.255.0

[gui]
enable = false

gnb_config
amf:
  addr: 172.16.0.183                  # The address or hostname of the AMF.
  bind_addr: 172.16.1.186             # A local IP that the gNB binds to for traffic from the AMF.

ru_sdr:
  device_driver: zmq                # The RF driver name.
  device_args: tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  srate: 23.04                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 20         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "46020"                     # PLMN broadcasted by the gNB.
  tac: 512000                            # Tracking area code (needs to match the core configuration).
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 12            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE

cu_cp:
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

wzh8267, 2024-05-15:
gnb control platform print:

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in executor blocking mode.

--== srsRAN gNB (commit 1483bda30) ==--

Connecting to AMF on 172.16.0.183:38412
Available radio types: uhd and zmq.
Cell pci=1, bw=20 MHz, 1T1R, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz

==== gNodeB started ===
Type <t> to view trace

wzh8267, 2024-05-15:
Additional:
      COTS UEs  with srsRAN gNB  and opent5gs  work well.

pgawlowicz, 2024-05-15:
Hi, you do not see any PRACH attempts in the gnb.log?

Could you try if those configs work for you? https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup

wzh8267, 2024-05-15:
thank you!    I check gnb log ,    there is not any prach . and  https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup  . I have read the link  above  many times , I have tried  all  possible  configs. the result are same,  Ue didn't send PRACH.   My  USRP type is b200.

pgawlowicz, 2024-05-15:
so do you want to run over ZMQ or USRP?

wzh8267, 2024-05-15:
And  I have try  the model   "Over-the-air Setup“，  Ue can receive the msg4,  but gnb   received  the msg4's feedback value =2. then Retransmission4.     I didn't have GPSDO.   Is this  the root cause ?

wzh8267, 2024-05-15:
> so do you want to run over ZMQ or USRP?

first  I want to run with  USRP , But failed ,because msg4 retransmission. gnb check msg4's feedback  NACK, UE send feedback ack. 
this  way failed , then I tried   ZMQ , and change the config paras needed ,  but afer Ue Print "Attaching UE... ", Ue didn't send PRACH.

wzh8267, 2024-05-15:
these are the logs of gnb and ue ,  USRP model. 
[ue.zip](https://github.com/srsran/srsRAN_Project/files/15318464/ue.zip)
[0509_gnb_log.zip](https://github.com/srsran/srsRAN_Project/files/15318467/0509_gnb_log.zip)

pgawlowicz, 2024-05-15:
Are you sure those logs are from the same experment?
In the UE logs, I can see the UE sent a PRACH and even got RRC connected. But I cannot see the corresponding PRACH in gnb logs.
```
2024-05-09T08:24:05.472903 [PHY-SA ] [I] [ 1290] PRACH: Transmitted preamble=0, tti_tx=1294, CFO=-1.60 KHz, nof_sf=1, target_power=-1.0 dBm
...
2024-05-09T08:24:05.498137 [MAC-NR ] [I] [ 1314] Received Contention Resolution ID 0x154ed6facf26
2024-05-09T08:24:05.498137 [MAC-NR ] [I] [ 1314] Received CONRES ID matches transmitted UE ID
2024-05-09T08:24:05.498146 [MAC-NR ] [I] [ 1314] Random Access Complete.     c-rnti=0x464c, ta=4
2024-05-09T08:24:05.498146 [RRC-NR ] [I] RA completed
...
2024-05-09T08:24:05.504631 [RRC-NR ] [I] SRB1 - Tx rrcSetupComplete (27 B)
...
2024-05-09T08:24:05.504639 [RRC-NR ] [I] Finished Connection Setup successfully
```

On the other hand, there are out of sync issues in ue logs, so probably you will need to connect an external clock.
```
2024-05-09T08:24:05.873727 [PHY-SA ] [E] [ 1690] SYNC: detected out-of-sync... skipping slot ...
2024-05-09T08:24:05.874878 [PHY-SA ] [E] [ 1690] SYNC: detected out-of-sync... skipping slot ...
```

Could you share your configs?

wzh8267, 2024-05-15:
> Are you sure those logs are from the same experment? In the UE logs, I can see the UE sent a PRACH and even got RRC connected. But I cannot see the corresponding PRACH in gnb logs.
>
> ``` [...]

sorry, may gnb and ue log  catch not  at the same time. I will catch the gnb and ue log at the same time and show  my configs. if the external clock must be nessary ?  is it the gps ?

wzh8267, 2024-05-15:
> Are you sure those logs are from the same experment? In the UE logs, I can see the UE sent a PRACH and even got RRC connected. But I cannot see the corresponding PRACH in gnb logs.
>
> ``` [...]

in additional ,  in the ZMQ model , why  ue didn't send prach.  because I did't have an external clok  in USRP model currnetly.

pgawlowicz, 2024-05-15:
in ZMQ-based setup, you do not need the USRP as signal samples are sent between gnb ans srsUE over a TCP sockets.

wzh8267, 2024-05-15:
> in ZMQ-based setup, you do not need the USRP as signal samples are sent between gnb ans srsUE over a TCP sockets.

Thanks a lot.    may  some confusion.  
1.  in the ZMQ model, I didn't use USRP.  Two PC with ubanun system.   control platform print as below.
     gnb  control platform  print:
The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in executor blocking mode.

--== srsRAN gNB (commit 1483bda30) ==--

Connecting to AMF on 172.16.0.183:38412
Available radio types: uhd and zmq.
Cell pci=1, bw=20 MHz, 1T1R, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz

==== gNodeB started ===
Type <t> to view trace

and ue  platform  print :
Inactive RF plugins:
Reading configuration file ue_download_zmq.yaml...

Built in Release mode using 23.04.0.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6
Supported RF device list: UHD zmq file
CHx base_srate=23.04e6
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://127.0.0.1:2000
CH0 tx_port=tcp://127.0.0.1:2001
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...

pgawlowicz, 2024-05-15:
if you run the gnb and srsUE over ZMQ-RF devices, you must set the correct IPs as tx and rx endpoints. You cannot use `127.0.0.1`.

wzh8267, 2024-05-15:
> if you run the gnb and srsUE over ZMQ-RF devices, you must set the correct IPs as tx and rx endpoints. You cannot use `127.0.0.1`.

ok, thanks a lot.  I will try.

wzh8267, 2024-05-15:
> check here: [#568 (comment)](https://github.com/srsran/srsRAN_Project/issues/568#issuecomment-2067746172)

ok, thank you very much!

pgawlowicz, 2024-05-21:
@wzh8267 any update on this issue?

wzh8267, 2024-05-27:
> @wzh8267 any update on this issue?

I am doing some tests, eg: ping , if all the tests  are over, I will close the issue. thank you .

wzh8267, 2024-05-27:
@pgawlowicz,   thank you for help me so much.    I have another issue.    If I want to  supper  n1 band.  I change the gNB and gUe config to  n1 band,  but failed.   as I know,  gNB support n1 band, but ue not support n1 band.    when I change the ue config to n1,  I  found  the ue search cell as n3 band.   what I will do to support  n1 band . where I will modify  about ue.

pgawlowicz, 2024-05-27:
in srsUE you need to correctly set the following parameters as well:

```
[rat.nr]
bands = 3
dl_nr_arfcn = ...
ssb_nr_arfcn = ...
...
```

wzh8267, 2024-05-27:
> in srsUE you need to correctly set the following parameters as well:
> 
> ```
> [rat.nr]
> bands = 3
> dl_nr_arfcn = ...
> ssb_nr_arfcn = ...
> ...
> ```

I know ,  thank you.      but   If  I want to test n1 ,  because at last.   I want to use n1 band  to do something others.

wzh8267, 2024-05-27:
> in srsUE you need to correctly set the following parameters as well:
> 
> ```
> [rat.nr]
> bands = 3
> dl_nr_arfcn = ...
> ssb_nr_arfcn = ...
> ...
> ```

I have test n3 band.  really ,I want to use n1 band to do something, eg.  I subway,  only use n1 band to communciate.

pgawlowicz, 2024-05-27:
just put band 1 and correct arfcn numbers

wzh8267, 2024-05-28:
> just put band 1 and correct arfcn numbers

Ok, I will try, thanks a lot.

wzh8267, 2024-05-28:
@pgawlowicz.   thanks.    I have a question.  before  I  made  ue and open5gs  in one computer ,   gnb  in another  computer.  I think  it's not suitable. so I want to adjust them.    put he gnb and open5gs in a computer ,  ue  in another  computer.  but how to  config the open5gs ip and gnb  ip   in  gnb  config file?

wzh8267, 2024-05-28:
when I  put  gnb and open5gs in one computer.    when gnb connected to open5gs.   the gnb console print "
Connecting to AMF on 172.16.0.183:38412
Failed to bind UDP socket to 172.16.0.183:2152. Address already in use
srsGNB ERROR: Unable to allocate the required NG-U network resources",  
the gnb config file :

amf:
  addr: 172.16.0.183                                               # The address or hostname of the AMF.
  bind_addr: 172.16.0.183                                            # A local IP that the gNB binds to for traffic from the AMF.

wzh8267, 2024-06-06:
> @wzh8267 any update on this issue?

I  haven't  resolve this issue .  when gnb and open5gs in the same machine.  when gnb connected to open5gs,  it will print"Connecting to AMF on 172.16.0.183:38412
Failed to bind UDP socket to 172.16.0.183:2152. Address already in use
srsGNB ERROR: Unable to allocate the required NG-U network resources".       now my environment is  gue and open5gs in a machine,  gnb in another machine.    in this condition,  when I ping from gue to open5gs.  it doesn't  through  gnb.  
> @wzh8267 any update on this issue?

pgawlowicz, 2024-06-07:
when running both gnb and open5gs, why do you want to put use a real interface (172.16.0.183)? 

you can simply use a loopback interface, gnb config should be:
```
    amf:
      addr: 127.0.1.100                                             # The address or hostname of the AMF.
      bind_addr: 127.0.0.1                                          # A local IP that the gNB binds to for traffic from the AMF.
```

amf in open5gs:
```
amf:
    ...
    ngap:
      - addr: 127.0.1.100
  ```

upf:
```
upf:
    pfcp:
      - addr: 127.0.0.7
    gtpu:
      - addr: 127.0.1.100
    subnet:
      - addr: 10.45.0.1/16
      - addr: 2001:db8:cafe::1/48
    metrics:
      - addr: 127.0.0.7
        port: 9090
```

wzh8267, 2024-06-07:
> when running both gnb and open5gs, why do you want to put use a real interface (172.16.0.183)?
>
> you can simply use a loopback interface, gnb config should be:
>
> ```
>     amf:
>       addr: [...]

ok, thanks very  much ,I will try.

wzh8267, 2024-06-07:
> when running both gnb and open5gs, why do you want to put use a real interface (172.16.0.183)?
>
> you can simply use a loopback interface, gnb config should be:
>
> ```
>     amf:
>       addr: [...]

it's ok now. thanks a lot.  may I will have some other issues,  will  I close this issue  or   later ?

pgawlowicz, 2024-06-07:
please start a new issue if needed.

closed 2024-06-07 by pgawlowicz