srsran/srsRAN_Project #619: Antenna Numbers on srsRAN gNB and Amarisoft UE Need Synchronization with New Version

houshangaziziaccelleran, 2024-05-15:
Hello
With following set of antenna config, and the latest release of srsRAN, UE never gets attached. 
gNB:
```

nof_antennas_dl: 2
nof_antennas_ul: 2
```

Amarisoft UE:

```
n_antenna_dl: 2,
n_antenna_ul: 1,
```

With previous release and current release if I set it to 2/2 (dl/ul) in gNB and 2/2 (dl/ul) in Amarisoft, it always works.
I am using B210 and it is connected one by one to Amarisoft.
Attached, you can find the non-working scenarios gNB and UE logs.

[20240515_1.zip](https://github.com/srsran/srsRAN_Project/files/15323079/20240515_1.zip)

xavierarteaga, 2024-05-15:
Many thanks for opening this issue.

The gNb detected the MSG3 with good SNR (above 20dB) but failed to decode it:
```
2024-05-15T13:52:10.107273 [PHY     ] [D] [  801.18] PUSCH: rnti=0x4601 h_id=0 prb=[4, 7) symb=[0, 14) mod=QPSK rv=0 tbs=11 crc=KO iter=6.0 sinr=33.7dB t=769.2us uci_t=0.0us ret_t=0.0us
  rnti=0x4601
  h_id=0
  bwp=[0, 51)
  prb=[4, 7)
  symb=[0, 14)
  oack=0
  ocsi1=0
  part2=entries=[]
  alpha=0.0
  betas=[0.0, 0.0, 0.0]
  mod=QPSK
  tcr=0.1171875
  rv=0
  bg=2
  new_data=true
  n_id=1
  dmrs_mask=00100001000100
  n_scr_id=1
  n_scid=false
  n_cdm_g_wd=2
  dmrs_type=1
  lbrm=3168bytes
  slot=801.18
  cp=normal
  nof_layers=1
  ports=0 1
  dc_position=306
  crc=KO
  iter=6.0
  max_iter=6
  min_iter=6
``` 

I guess this could caused by the signal unbalance in the gNb receive ports.

houshangaziziaccelleran, 2024-05-15:
Thanks @xavierarteaga I wonder why this "issue" wasn't there with previous release of srsRAN.

xavierarteaga, 2024-05-15:
There are two changes related to channel estimation and equalization in the last release:
1. carrier frequency offset measurement and compensation; and
2. a temporary hotfix in the equalizer noise variance.

You can disable the CFO compensation with the following patch:
```diff
diff --git a/include/srsran/phy/upper/signal_processors/signal_processor_factories.h b/include/srsran/phy/upper/signal_processors/signal_processor_factories.h
index bb48607a5..055afda78 100644
--- a/include/srsran/phy/upper/signal_processors/signal_processor_factories.h
+++ b/include/srsran/phy/upper/signal_processors/signal_processor_factories.h
@@ -95,7 +95,7 @@ public:
   virtual std::unique_ptr<port_channel_estimator>
   create(port_channel_estimator_fd_smoothing_strategy fd_smoothing_strategy =
              port_channel_estimator_fd_smoothing_strategy::filter,
-         bool complensate_cfo = true) = 0;
+         bool complensate_cfo = false) = 0;
 };
 
 std::shared_ptr<port_channel_estimator_factory>
```

For reverting the channel equalizer changes:
```diff
diff --git a/lib/phy/upper/equalization/channel_equalizer_zf_impl.cpp b/lib/phy/upper/equalization/channel_equalizer_zf_impl.cpp
index 69632bc61..83b9d2f3c 100644
--- a/lib/phy/upper/equalization/channel_equalizer_zf_impl.cpp
+++ b/lib/phy/upper/equalization/channel_equalizer_zf_impl.cpp
@@ -131,12 +131,8 @@ void channel_equalizer_zf_impl::equalize(span<cf_t>         eq_symbols,
   unsigned nof_rx_ports  = ch_estimates.get_dimension_size(ch_est_list::dims::rx_port);
   unsigned nof_tx_layers = ch_estimates.get_dimension_size(ch_est_list::dims::tx_layer);
 
+  // Compute the average noise variance.
+  float noise_var = 0.0F;
+  for (unsigned i_rx_port = 0; i_rx_port != nof_rx_ports; ++i_rx_port) {
+    noise_var += noise_var_estimates[i_rx_port];
+  }
+  noise_var /= static_cast<float>(nof_rx_ports);
-  // Select the most pessimistic noise variance.
-  float noise_var = *std::max_element(noise_var_estimates.begin(), noise_var_estimates.end());
 
   // Single transmit layer and any number of receive ports.
   if (nof_tx_layers == 1) {
```

We are working on improving the UL resilience against these types of corner cases. Feel free to try the patches and let us know if any of these patches walk around the problem.

houshangaziziaccelleran, 2024-05-15:
Thanks @xavierarteaga We are OK setting number of dl/ul antennas the same on gNB and Amarisoft UE side as there is no issue of that is the case. So for the fix you mentioned, we can wait. Thanks

ismagom, 2024-05-15:
Hi @houshangaziziaccelleran,

What @xavierarteaga was trying to ask was if you could test the two patches for us please. We are unable to reproduce the issue. It would be very helpful if you can test those patches please. Thanks

houshangaziziaccelleran, 2024-05-16:
@xavierarteaga Applying the patch helped. Meaning we don't necessarily need to have the same number of gNB UL antennas on Amarisoft-UE to be able to Attach a UE.
Requested logs are attached.
Cc @ismagom 
[patch_619.zip](https://github.com/srsran/srsRAN_Project/files/15334937/patch_619.zip)

[20240516_1-issue_619.zip](https://github.com/srsran/srsRAN_Project/files/15334810/20240516_1-issue_619.zip)

xavierarteaga, 2024-05-16:
I found a software defect that mixes up the CFO compensation. We created this patch 
[cfo.patch.tar.gz](https://github.com/srsran/srsRAN_Project/files/15338431/cfo.patch.tar.gz).

@houshangaziziaccelleran can you try it and let us know how it goes? :crossed_fingers:

This fix will be included in the next release.

Many thanks for your help.

xavierarteaga, 2024-05-21:
Hi @houshangaziziaccelleran,
we have already released all the changes about the CFO compensation. This issue should not happen again after pulling the latest main ([f3ed07a](https://github.com/srsran/srsRAN_Project/commit/f3ed07a5a6f04245a57ca667194e1a3bf9684149)) code.

I will close this issue and we reopen it again if the problem persists.

Once again, many thanks for helping us to find the issue.

closed 2024-05-21 by xavierarteaga