srsran/srsRAN_Project #623: Starting the gnb got "Floating point exception" on 24.04

wei-lou, 2024-05-21:
## Issue Description ##
Starting the "gnb" got the "Floating point exception". The program exit after the error.

## Setup Details ##
This is a FDD setup with Fujitsu n71 ORAN radio operates at 10MHz with 4x4 MIMO.

## Expected Behavior ##
Starting "gnb" with no exception

## Actual Behaviour ##
"gnb" execution exit after the error

## Steps to reproduce the problem ##
This is a newly compiled 24.04 on bare metal server. Issue is consistent with no other tweaking

## Additional Information ##
Execution terminal display, srsran configuration file and gnb.log are attached as follow.
```
sulabuser@srsgnb03:~$ sudo srsRAN_Project2404/build/apps/gnb/gnb -c srsRAN_Project2404/configs/FUJI_GEN1TB_RU_n71_10mhz_4x4_v7.yml

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 2, SCS 1.25kHz, Rx ports 1}.

--== srsRAN gNB (commit 78238fd15) ==--

Connecting to AMF on 192.168.70.220:38412
Initializing the Open Fronthaul Interface for sector#0: ul_compr=[BFP,9], dl_compr=[BFP,9], prach_compr=[BFP,9], prach_cp_enabled=true, downlink_broadcast=false.
Operating a 10MHz cell over a RU with instantaneous bandwidth of 100MHz
Warning: Configured PRACH occasion collides with PUCCH RBs ([0..1) intersects [0..6)). Some interference between PUCCH and PRACH is expected.
Warning: Configured PRACH occasion collides with PUCCH RBs ([0..1) intersects [0..6)). Some interference between PUCCH and PRACH is expected.
Cell pci=140, bw=10 MHz, 4T4R, dl_arfcn=124400 (n71), dl_freq=622.0 MHz, dl_ssb_arfcn=123890, ul_freq=668.0 MHz

==== gNodeB started ===
Type <t> to view trace
Floating point exception
sulabuser@srsgnb03:~$
``` 
[gnb.log](https://github.com/srsran/srsRAN_Project/files/15392835/gnb.log)
[FUJI_RU_n71_10mhz_4x4_v7.txt](https://github.com/srsran/srsRAN_Project/files/15392848/FUJI_RU_n71_10mhz_4x4_v7.txt)

ismagom, 2024-05-21:
Hi @wei-lou , 

I tested your configuration and works well with the latest commit. I notice you are using the option `ru_bandwidth_MHz: 100`. Is that intended? Doesn't the RU support 10 MHz?

wei-lou, 2024-05-21:
Hi @ismagom,

My RU does support 10MHz. I changed the ru_bandwidth_MHz: 10, now it is working. Thanks you for point this error out.

Wei,

closed 2024-05-21 by ismagom