srsran/srsRAN_Project #624: Data throughput limited to 600-800 Mbps

vvdntech123, 2024-05-23:
## Issue Description ##
While testing UDP performance with the openSRS gNB application in C1 topology it is  observed that the throughput is capped is at 600 - 800 Mbps. To eliminate the possibility of issue related with open5GS core we tried with another core with which we attained 1.5Gbps with another gNB stack.But the issue still persisted.

## Actual Behavior ##
We have checked whether UDP  data is dropped along the way from the UPF. But using the ifstat command we could see that this was not the case. Attaching relevant screenshots for analysis. 
[image]
Note: We connected the NGU in ens2f2 interface.

iperf client log
[image]
gNB trace
[image]
The configuration file is given below:
[gnb_ru_ran_vvdn_lpru_tdd_n78_100mhz_4x4.zip](https://github.com/srsran/srsRAN_Project/files/15417030/gnb_ru_ran_vvdn_lpru_tdd_n78_100mhz_4x4.zip)

andrepuschmann, 2024-05-23:
Hey - in those logs I do see full buffer state so you have enough data to transmit. However, you UL is pretty bad. You need to adjust the power settings so the UE can better adjust Tx power. The key parameters (and their default configs) are:
```
  ssb:
    ssb_block_power_dbm: -16
  pusch:
    p0_nominal_with_grant: -76
  pucch:
    p0_nominal: -90
```
In your configs you're using those default values. What you want to achieve is that the UL rsrs column is something like -10...-5dB. So you need to increase e.g. `p0_nominal_with_grant` by 10dB. Ideally you would set `ssb_block_power_dbm` to the correct value but you need to measure that value with calibrated measurement equipment with 5G NR decoding capabilities. 

What I don't quite understand is the fluctuating `phr` column in your tests. Are you moving the phone at all?

ismagom, 2024-05-23:
Also tell us more about your setup: what UE you are using, how you have connected it, etc

ismagom, 2024-05-24:
Hi @vvdntech123 , regarding this. The main problem I see is the UL. There seems to be a lot of packet drop in the uplink. This might be causing ACK/NACK feedback to not be received and forcing PDSCH retx. 

Can you set `all_level: info` and send us the gnb log? 

Do you see al C/U-plane OFH messages arriving on time at the RU?

Could we start with a wired setup before going over the air?