srsran/srsRAN_Project #627: Issue when connecting 2 RUs, UE is attaching to only one RU

vvdn5gtest, 2024-05-24:
Hi

We have tried connecting 2 RUs, but the UE is attaching only with one o-RU.
The block diagram is as follows:
[image]

The config is given below:
[gnb_ru_ran_vvdn_lpru_tdd_n78_100mhz_4x4_multiple_ru.zip](https://github.com/srsran/srsRAN_Project/files/15429576/gnb_ru_ran_vvdn_lpru_tdd_n78_100mhz_4x4_multiple_ru.zip)

ismagom, 2024-05-24:
Can try commeting out/removing this section? 
```
cu_cp:
  max_nof_dus: 1                          # Optional UINT. Maximum number of DU connections that the CU-CP may accept.
  max_nof_cu_ups: 1   
```

vvdn5gtest, 2024-05-24:
Hi @ismagom 

We have tried that too by changing 
`cu_cp:
  max_nof_dus: 2                         # Optional UINT. Maximum number of DU connections that the CU-CP may accept.
  max_nof_cu_ups: 2 `

ismagom, 2024-05-24:
Can you send gnb logs? 

Do you see CU-plane messages arriving all on time at both RUs?

closed 2024-06-20 by pgawlowicz