srsran/srsRAN_Project #636: Installing srsRAN with DPDK fails

houshangaziziaccelleran, 2024-05-28:
Hello
Can you please help with this issue?
Installing srsRAN with DPDK fails. Please see attached the error messages.
Thanls
Houshang
[srsran_install_dpdk.txt](https://github.com/srsran/srsRAN_Project/files/15473592/srsran_install_dpdk.txt)

ninjab3s, 2024-05-29:
Hello @houshangaziziaccelleran,
I think the DPDK version you have installed is too old. 

```
-- Checking for module 'libdpdk>=22.11'
--   Requested 'libdpdk >= 22.11' but version of DPDK is 21.11.6
```
Please remove DPDK entirely and install version 23.11 with the following steps:

```
sudo apt install build-essential tar wget python3-pip libnuma-dev 
sudo pip3 install meson ninja pyelftools

wget https://fast.dpdk.org/rel/dpdk-23.11.tar.xz
tar xvf dpdk-23.11.tar.xz dpdk-23.11/
cd dpdk-23.11/
meson setup build
cd build
ninja
sudo meson install
sudo ldconfig
```

After that you should be able to build without issues.

houshangaziziaccelleran, 2024-05-29:
Hello @ninjab3s 
Thanks for the prompt reply! 
Version 23.11 of DPDK helped.
Quick question:
Is the attached `dpdk-devbind.py -s` output looks normal?

[dpdk-devbind.txt](https://github.com/srsran/srsRAN_Project/files/15481971/dpdk-devbind.txt)

ninjab3s, 2024-05-29:
Hello,
the output looks normal to me. I can see that you have installed two Intel X710 cards, 1x 4ports, 1x 2ports. Ports 1-3 of the 4 port card are connected. Is this correct?
The next step would be binding one or more of the interfaces to a DPDK compatible driver, depending of what you plan to do. Please refer to our [docs](https://docs.srsran.com/projects/project/en/latest/tutorials/source/dpdk/source/index.html) for this step. If you encounter issues please let us know. Thanks!

closed 2024-05-29 by ismagom