srsran/srsRAN_Project #638: DPDK ON issue

houshangaziziaccelleran, 2024-05-29:
Hello

Based on
```

ad@bm-super11-intel:~/pure$ sudo dpdk-devbind.py -s

Network devices using kernel driver
===================================
0000:04:00.0 'Ethernet Controller 10G X550T 1563' if=eno1 drv=ixgbe unused=vfio-pci *Active*
0000:04:00.1 'Ethernet Controller 10G X550T 1563' if=eno2 drv=ixgbe unused=vfio-pci 
0000:43:00.0 'Ethernet Controller X710 for 10GbE SFP+ 1572' if=ens1f0 drv=i40e unused=vfio-pci 
0000:43:00.1 'Ethernet Controller X710 for 10GbE SFP+ 1572' if=ens1f1 drv=i40e unused=vfio-pci *Active*
0000:43:00.2 'Ethernet Controller X710 for 10GbE SFP+ 1572' if=ens1f2 drv=i40e unused=vfio-pci *Active*
0000:43:00.3 'Ethernet Controller X710 for 10GbE SFP+ 1572' if=ens1f3 drv=i40e unused=vfio-pci *Active*
0000:70:00.0 'Ethernet Controller X710 for 10GbE SFP+ 1572' if=ens2f0 drv=i40e unused=vfio-pci 
0000:70:00.1 'Ethernet Controller X710 for 10GbE SFP+ 1572' if=ens2f1 drv=i40e unused=vfio-pci 
```

The server has 64 CPUs:

```
ad@bm-super11-intel:~/pure$ lscpu
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         46 bits physical, 57 bits virtual
  Byte Order:            Little Endian
CPU(s):                  64
```

knowing `ens1f1` and `ens1f2` are the server interfaces connected to n310, how do I have to configure the following part?

```

hal:
  eal_args: "--lcores (0-1)@(0-63) -a 0000:43:00.2"
  
```
  
With or without `eal_args` part, I always get:

```
The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in quad executor mode.

--== srsRAN gNB (commit f3ed07a5a) ==--

Connecting to AMF on 10.55.7.40:38412
Failed to load RF plugin libsrsran_radio_uhd.so: libsrsran_radio_uhd.so: cannot open shared object file: No such file or directory
Factory for radio type uhd not found. Make sure to select a valid type.
srsGNB ERROR: Unable to create radio session.
```

The issue started when I installed srsRAN with DPDK set to ON.

[gnb.log](https://github.com/srsran/srsRAN_Project/files/15484357/gnb.log)

pgawlowicz, 2024-06-20:
@houshangaziziaccelleran any update on this issue?

houshangaziziaccelleran, 2024-06-20:
Sorry for the late response @pgawlowicz 
I am busy with another project. We can close this for now and if I still have the issue, I will open a new one. Is that OK?

closed 2024-06-20 by pgawlowicz