srsran/srsRAN_Project #639: Missing UP PRACH messages + late wake-up

melkael, 2024-05-29:
Hello!

## Issue Description ##
I managed to run srsRAN with  RAN550 on a beefy xeon server fine. I now want to run it on a smaller PC but I keep getting timing issues in the flavor of (see the attached logs):
> Real-time timing worker woke up late, sleep time has been '1000us', or equivalently, '28' symbols
> Missed incoming User-Plane PRACH messages

The UE also does not attach in this setup but it did on the xeon server
## Setup Details ##

- CPU: intel i9 7900X
- RAN550 using PTP over a switch
- intel x810
- srsRAN 5G gNB version 24.04.0
- Ubuntu 22.04
- kernel 6.5
- 

## Expected Behavior ##
The UE should attach

## Actual Behaviour ##

The UE doesnt attach but the RU receives all the packets on time. However the DU does not seem to receive anything

> 2024-05-29T20:24:54.251883 [OFH     ] [I] Opened successfully the NIC interface 'enp23s0f1np1' used by the Ethernet receiver
> 2024-05-29T20:24:54.251889 [OFH     ] [I] Opened successfully the NIC interface 'enp23s0f1np1' used by the Ethernet transmitter
> 2024-05-29T20:24:54.341216 [OFH     ] [I] Starting the operation of the Open Fronthaul interface
> 2024-05-29T20:24:54.341217 [OFH     ] [I] Starting the realtime timing worker
> 2024-05-29T20:24:54.341225 [OFH     ] [I] Started the realtime timing worker
> 2024-05-29T20:24:54.341227 [OFH     ] [I] Starting the ethernet frame receiver
> 2024-05-29T20:24:54.341244 [OFH     ] [I] Started the ethernet frame receiver
> 2024-05-29T20:24:54.341244 [OFH     ] [I] Started the operation of the Open Fronthaul interface
> 2024-05-29T20:24:54.357017 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '450.19' and sector#0
> 2024-05-29T20:24:54.367022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '451.19' and sector#0
> 2024-05-29T20:24:54.377026 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '452.19' and sector#0
> 2024-05-29T20:24:54.387013 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '453.19' and sector#0
> 2024-05-29T20:24:54.397022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '454.19' and sector#0
> 2024-05-29T20:24:54.407023 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '455.19' and sector#0
> 2024-05-29T20:24:54.417016 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '456.19' and sector#0
> 2024-05-29T20:24:54.427024 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '457.19' and sector#0
> 2024-05-29T20:24:54.437018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '458.19' and sector#0
> 2024-05-29T20:24:54.447016 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '459.19' and sector#0
> 2024-05-29T20:24:54.457018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '460.19' and sector#0
> 2024-05-29T20:24:54.467023 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '461.19' and sector#0
> 2024-05-29T20:24:54.477021 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '462.19' and sector#0
> 2024-05-29T20:24:54.487025 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '463.19' and sector#0
> 2024-05-29T20:24:54.497025 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '464.19' and sector#0
> 2024-05-29T20:24:54.507021 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '465.19' and sector#0
> 2024-05-29T20:24:54.517019 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '466.19' and sector#0
> 2024-05-29T20:24:54.527022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '467.19' and sector#0
> 2024-05-29T20:24:54.537020 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '468.19' and sector#0
> 2024-05-29T20:24:54.547016 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '469.19' and sector#0
> 2024-05-29T20:24:54.557018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '470.19' and sector#0
> 2024-05-29T20:24:54.567014 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '471.19' and sector#0
> 2024-05-29T20:24:54.577015 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '472.19' and sector#0
> 2024-05-29T20:24:54.587022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '473.19' and sector#0
> 2024-05-29T20:24:54.597016 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '474.19' and sector#0
> 2024-05-29T20:24:54.607021 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '475.19' and sector#0
> 2024-05-29T20:24:54.617014 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '476.19' and sector#0
> 2024-05-29T20:24:54.627020 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '477.19' and sector#0
> 2024-05-29T20:24:54.637018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '478.19' and sector#0
> 2024-05-29T20:24:54.647013 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '479.19' and sector#0
> 2024-05-29T20:24:54.657021 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '480.19' and sector#0
> 2024-05-29T20:24:54.667016 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '481.19' and sector#0
> 2024-05-29T20:24:54.677025 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '482.19' and sector#0
> 2024-05-29T20:24:54.687025 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '483.19' and sector#0
> 2024-05-29T20:24:54.697015 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '484.19' and sector#0
> 2024-05-29T20:24:54.707018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '485.19' and sector#0
> 2024-05-29T20:24:54.717020 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '486.19' and sector#0
[... 160 lines cut ...]
>   prach:
>     prach_config_index: 159
>     prach_root_sequence_index: 1
>     zero_correlation_zone: 0
>     prach_frequency_start: 2
>   tdd_ul_dl_cfg:
>     dl_ul_tx_period: 10                     # Optional INT (10). Sets the TDD pattern periodicity in slots. The combination of this value and the chosen numerology must lead to a TDD periodicity of 0.5, >
>     nof_dl_slots: 7                         # Optional INT (6). Number of consecutive full Downlink slots. Supported: [0-80].
>     nof_dl_symbols: 6                       # Optional INT (0). Number of Downlink symbols at the beginning of the slot following full Downlink slots. Supported: [0-13].
>     nof_ul_slots: 2                         # Optional INT (3). Number of consecutive full Uplink slots. Supported: [0 - 80].
>     nof_ul_symbols: 4
>   ssb:
>     ssb_period: 20
>   pdsch:
>     mcs_table: qam256
> 
> log:
>   filename: log.log            # Path of the log file.
>   all_level: warning
>   ofh_level: info

 

## Additional Information ##
i used the srs_performance script

andrepuschmann, 2024-05-29:
No traffic is coming back from the RU so the DU complains. Double check MAC addresses, VLAN, settings, etc.

andrepuschmann, 2024-05-30:
You have a few lates, but not too many. This can be solved with thread affinities and/or use of DPDK. Please check KPI on the RU as well and check if it's emitting a signal.

melkael, 2024-05-30:
I checked the kpi.sh and i have a few RX_late (between 34 and 37). I tried various configs for the CPU pinning without much success. I verified with Network Signal Guru and I can't see the SIB. Which of the pinning parameters typically help in those cases?

ibrahimshikdaher, 2024-08-25:
Hi @melkael ,
did you manage to solve the Issue?
I am having something similar.

waddem01, 2024-09-11:
I am wondering if  @melkaelor @ibrahimshikdaher have been able to make any progress with this?

I am seeing exactly the same behaviour on a very similar system 
(i9-10900 2.8Ghz / 10 cores. Ubuntu 22.04,  6.8 kernel, Local Open5Gs Core, srsRAN 24.04.0, RAN650-1v1.0.4-dda1bf5, FibroLAN Falcon-RX/812/G, E810 O-DU NIC)

gnb.log fills with these messages:

2024-09-11T15:02:01.807015 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '91.19' and sector#0
2024-09-11T15:02:01.817014 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '92.19' and sector#0
2024-09-11T15:02:01.827017 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '93.19' and sector#0
2024-09-11T15:02:01.837011 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '94.19' and sector#0
2024-09-11T15:02:01.847018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '95.19' and sector#0
2024-09-11T15:02:01.857024 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '96.19' and sector#0
2024-09-11T15:02:01.867019 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '97.19' and sector#0
2024-09-11T15:02:01.877013 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '98.19' and sector#0
2024-09-11T15:02:01.887022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '99.19' and sector#0

I am seeing the odd rumble of RX_LATE on the kpi.sh output, but not too many ( up to 10 or so).

Things improved when I changed the option on phc2sys on the O-DU:
phc2sys -s enp2s0f0np0 -c CLOCK_REALTIME -w -m -R 8 -n 24 -f /etc/linuxptp/default.cfg

The corresponding ptp4l service runs::
ptp4l -2 -f /etc/linuxptp/default.cfg -m -I enp2s0f0np0

default.cfg is the srsRAN configuration file (https://docs.srsran.com/projects/project/en/latest/_downloads/76d8d0c734c4c4652b38a3032dc56d7b/default.cfg)
but I have to comment out quite a few parameters before ptp4l will accept this
 
#clientOnly              0
#delay_response_timeout  0
#serverOnly
#power_profile.2011.networkTimeInaccuracy        0
#power_profile.version                           none
#clock_class_threshold   248
#refclock_sock_address   /var/run/refclock.ptp.sock
#uds_file_mode           0660
#uds_ro_address          /var/run/ptp4lro
#uds_ro_file_mode                0666
#phc_index               -1

I switched from the 100MHz 4x2 modes and I am now running the 20MHz 2x2 mode.
The RAN650 produces RF, but UEs do not attach.

I have this in my gnb.yml:

gnb_id: 1
amf:
  addr: 127.0.0.5                          # The address or hostname of the AMF.
  bind_addr: 127.0.0.1                     # A local IP that the gNB binds to for traffic from the AMF (local Open5GS core).

ru_ofh:
  t1a_max_cp_dl: 500                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 450                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 350                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 290                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 360                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 300                                                 # Minimum T1a on User-Plane in microseconds.
  ta4_max: 200                                                    # Maximum Ta4 on User-Plane in microseconds.
  ta4_min: 0                                                      # Minimum Ta4 on User-Plane in microseconds.
  is_prach_cp_enabled: true                                       # Configures if Control-Plane messages should be used to receive PRACH messages.
  compr_method_ul: bfp                                            # Uplink compression method.
  compr_bitwidth_ul: 9                                            # Uplink IQ samples bitwidth after compression.
  compr_method_dl: bfp                                            # Downlink compression method.
  compr_bitwidth_dl: 9                                            # Downlink IQ samples bitwidth after compression.
  compr_method_prach: bfp                                         # PRACH compression method.
  compr_bitwidth_prach: 9                                         # PRACH IQ samples bitwidth after compression.
  enable_ul_static_compr_hdr: false                               # Configures if the compression header is present for uplink User-Plane messages (false) or not present (true).
  enable_dl_static_compr_hdr: false                               # Configures if the compression header is present for downlink User-Plane messages (false) or not present (true).
  iq_scaling: 4.5                                                 # IQ samples scaling factor applied before compression, should be a positive value smaller than 10.
  cells:
[... 94 lines cut ...]
#                 false
lf_prach_compression_enable=true

# C-Plane per-symbol workaround
# Variable name : cplane_per_symbol_workaround
# Valid options : enabled
#                 disabled
cplane_per_symbol_workaround=disabled

# CUPLANE dl couplling sectionID
# Variable name : cuplane_dl_coupling_sectionI
# Valid options : enabled
#                 disabled
cuplane_dl_coupling_sectionID=disabled

# FlexRAN PRACH workaround
# Variable name : flexran_prach_workaround
# Valid options : enabled
#                 disabled
flexran_prach_workaround=disabled

#DL_special slot tunning
#Variable name : dl_tuning_special_slot
#Valid range   : 0x0:0x13bc
dl_tuning_special_slot=0x13b6

andrepuschmann, 2024-09-13:
@waddem01 The issue you're having is that the DU doesn't get any PRACH uplane back from the RU. So there is clearly a misconfiguration. Double check MACs, eAxC values, etc.

@melkael @ibrahimshikdaher Have you tried with a spectrum analyzer/USRP to see if the RU is emitting any signal at all?

pgawlowicz, 2024-10-16:
@melkael any updates on this issue?

ibrahimshikdaher, 2024-10-16:
Hi all,

There was an issue matching the prach configs. I set it to 159 short. but there was still an issue until I went down to 20 MHz, then the setup worked stably with a speed reaching up to 100 Mbps in the evening. The only running setup for me is 4Tx2R 20 MHz. I need to figure out the issue with 100 Mhz.

closed 2024-10-30 by pgawlowicz

sushila-s, 2025-03-25:
@melkael @ibrahimshikdaher I am facing the same issue. I have tried all the suggestions here but still cannot get it to work for me. Would you mind sharing your working config file for 2x2 MIMO? Thanks in advance!