srsran/srsRAN_Project #640: Compilation ERROR - srsRAN

pgupta408, 2024-05-30:
## Issue Description ##

Compile Error - two places & then exit..
In file included from /home/mobilestack/srsRAN_Project/lib/srsvec/accumulate.cpp:23:
and 
In file included from /home/mobilestack/srsRAN_Project/lib/srsvec/add.cpp:24:

Similar ERROR at both places. 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

UBUNTU 22.04  - Newly installed - with apt update & apt upgrade - looks good

sudo apt-get install cmake make gcc g++ pkg-config libfftw3-dev libmbedtls-dev libsctp-dev libyaml-cpp-dev libgtest-dev

UHD LTS 4.0 installed -- 
sudo add-apt-repository ppa:ettusresearch/uhd
sudo apt-get update
sudo apt-get install libuhd-dev uhd-host

Clone and Build srsRAN --
git clone https://github.com/srsRAN/srsRAN_Project.git
cd srsRAN_Project
mkdir build
cd build
cmake ../
make -j $(nproc)

## Expected Behavior ##
Expect Compile to work

## Actual Behaviour ##

Compile ERROR 

[ 15%] Linking CXX executable complex_normal_random_test
In file included from /home/mobilestack/srsRAN_Project/lib/srsvec/accumulate.cpp:23:
/home/mobilestack/srsRAN_Project/lib/srsvec/simd.h: In function ‘srsran::simd_cf_t srsran::srsran_simd_cf_interleave_low(srsran::simd_cf_t, srsran::simd_cf_t)’:
/home/mobilestack/srsRAN_Project/lib/srsvec/simd.h:1212:40: error: cannot convert ‘srsran::simd_f_t’ {aka ‘__m128’} to ‘__m256’
 1212 |   __m256 temp_a0 = _mm256_permute_ps(a.re, 0b11011000);
      |                                      ~~^~
      |                                        |
      |                                        srsran::simd_f_t {aka __m128}
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:43,
                 from /home/mobilestack/srsRAN_Project/lib/srsvec/simd.h:36,
                 from /home/mobilestack/srsRAN_Project/lib/srsvec/accumulate.cpp:23:
/usr/lib/gcc/x86_64-linux-gnu/11/include/avxintrin.h:668:27: note:   initializing argument 1 of ‘__m256 _mm256_permute_ps(__m256, int)’
  668 | _mm256_permute_ps (__m256 __X, const int __C)
      |                    ~~~~~~~^~~
In file included from /home/mobilestack/srsRAN_Project/lib/srsvec/accumulate.cpp:23:
/home/mobilestack/srsRAN_Project/lib/srsvec/simd.h:1213:40: error: cannot convert ‘srsran::simd_f_t’ {aka ‘__m128’} to ‘__m256’
 1213 |   __m256 temp_a1 = _mm256_permute_ps(a.im, 0b11011000);
      |                                      ~~^~
      |                                        |
      |                                        srsran::simd_f_t {aka __m128}
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:43,
                 from /home/mobilestack/srsRAN_Project/lib/srsvec/simd.h:36,
                 from /home/mobilestack/srsRAN_Project/lib/srsvec/accumulate.cpp:23:
/usr/lib/gcc/x86_64-linux-gnu/11/include/avxintrin.h:668:27: note:   initializing argument 1 of ‘__m256 _mm256_permute_ps(__m256, int)’
  668 | _mm256_permute_ps (__m256 __X, const int __C)
      |                    ~~~~~~~^~~
In file included from /home/mobilestack/srsRAN_Project/lib/srsvec/accumulate.cpp:23:
/home/mobilestack/srsRAN_Project/lib/srsvec/simd.h:1217:40: error: cannot convert ‘srsran::simd_f_t’ {aka ‘__m128’} to ‘__m256’
 1217 |   __m256 temp_b0 = _mm256_permute_ps(b.re, 0b11011000);
      |                                      ~~^~
      |                                        |
      |                                        srsran::simd_f_t {aka __m128}
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/include/immintrin.h:43,
[... 207 lines cut ...]
      |                    ~~~~~~~^~~
In file included from /home/mobilestack/srsRAN_Project/lib/srsvec/add.cpp:24:
/home/mobilestack/srsRAN_Project/lib/srsvec/simd.h:1291:34: error: cannot convert ‘__m256’ to ‘srsran::simd_f_t’ {aka ‘__m128’} in assignment
 1291 |   ret.re     = _mm256_unpacklo_ps(in1, in2);
      |                ~~~~~~~~~~~~~~~~~~^~~~~~~~~~
      |                                  |
      |                                  __m256
/home/mobilestack/srsRAN_Project/lib/srsvec/simd.h:1292:34: error: cannot convert ‘__m256’ to ‘srsran::simd_f_t’ {aka ‘__m128’} in assignment
 1292 |   ret.im     = _mm256_unpackhi_ps(in1, in2);
      |                ~~~~~~~~~~~~~~~~~~^~~~~~~~~~
      |                                  |
      |                                  __m256
make[2]: *** [lib/srsvec/CMakeFiles/srsvec.dir/build.make:90: lib/srsvec/CMakeFiles/srsvec.dir/add.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:10741: lib/srsvec/CMakeFiles/srsvec.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

Refer Setup Details Section 

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

None

dinberuD, 2024-06-05:
Hello, can anybody help me by fixing the following error? Thank you. here is the error: "+ mongosh open5gs ./open5gs-2.7.0/docs/assets/webui/mongo-init.js
MongoNetworkError: connect ECONNREFUSED 127.0.0.1:27017
Error executing command, exiting"

ninjab3s, 2024-06-12:
Can you please provide the cmake output of a fresh build? Thanks!

dinberuD, 2024-06-17:
The cmake has built successfully and i think the problem resides in mongodb; i can't find the mongodb installed on my virtual machine.

ninjab3s, 2024-06-17:
Mongodb is not needed for building srsRAN.

dinberuD, 2024-06-17:
yeah, actually i am trying to open5gs core to demonstrate the srsRAN project.

dinberuD, 2024-06-17:
I am newbie and is there any other way to demonstrate it without using mongodb?

ninjab3s, 2024-06-17:
I am afraid there is no way to do this, at least I am not aware. Follow this guide step by step and you should have a running instance of Open5GS in your machine/VM. https://open5gs.org/open5gs/docs/guide/02-building-open5gs-from-sources/

We also have a docker-compose for Open5GS. For more information on the docker container please refer to: https://github.com/srsran/srsRAN_Project/tree/main/docker

Use `docker compose up 5gc` to build and start the Open5GS container.

closed 2024-06-20 by pgawlowicz