srsran/srsRAN_Project #643: Iphone is not receiving incoming calls

abcd11abcd, 2024-05-30:
Im using open5gs core, srsrRAN and benetel650 for this experiment.
Im using samsung s22 ultra and iphone 14. Both the phones are running on latest software versions. When I try to call samsung from iphone, the call is establishing and voice is also going, But when I try to call iphone from samsung, the iphone is not receiving any calls. Means, there is no incoming call on iphone side.  The sim is fine. When I keep the sim of iphone in other samsung mobile (s22), then call between both samsung mobiles are fine. 
The data connection is fine on iphone. Im getting 2 pdu responses(internet and ims) from core side too. The problem is with iphone. Are there any iphone specific settings we have to change? Focus mode is turned off, no blocked numbers, no  call forwarding. Still no incoming. Any iphone users, kindly help.

peihaoY, 2024-06-03:
Hi, I am using my iPhone to do the test. I felt so confused. Could you tell me how to do “require IMS”? What's the meaning of this setting? I would so appreciate it.

abcd11abcd, 2024-06-05:
goto seetings > mobile networks. enable 2nd apn name as ims. default must be the internet apn

peihaoY, 2024-06-05:
Thanks for your information! However I cannot find a 2nd apn. I attached my screenshot. Only with this setting, I cannot connected my iphone to the gnb. Could you help me to take a look? Any screenshot or explanation would be greatly appreciated! [aad25229771147fe1d4eea8ededefc0](https://github.com/srsran/srsRAN_Project/assets/147773058/64c64f84-322d-44ef-a4d9-32d62a607cf7)

abcd11abcd, 2024-06-06:
in the cellular data section, keep the apn name as internet. This should be the name of your default apn.

peihaoY, 2024-06-06:
Thanks! I think that is my setting problem. But My iphone only has setting-> cellular network->cellular data->apn, without seetings > mobile networks. The default apn is none. How to add a 2nd apn? Perhaps you have some materials or links to help me  understand iphone apn setting better?

mrahman2-vt, 2024-07-30:
@abcd11abcd are you using 2 different gNB cells for your 2 phones? Did you try the same gNB cell with your 2 phones?

abcd11abcd, 2024-07-30:
same gnb cell for the two mobiles. only single cell is there.

mrahman2-vt, 2024-07-30:
@abcd11abcd I see in all gNB config files, PCI 1 is configured, which is the other PCI you see for other UE? Are you using the master branch? which tag?

abcd11abcd, 2024-07-31:
im not using the updated version of master branch(CU/DU split). im using the monolithic gnb code. Both the UEs are in the same cell, so they both use same pci. The issue is with iphone that too incoming voice call. If i use two samsung phones, the call works fine on both ways.

pgawlowicz, 2024-10-16:
@abcd11abcd any updates on this issue?

closed 2024-10-30 by pgawlowicz