srsran/srsRAN_Project #652: DL Task Queue Full Bug

wozzamac, 2024-06-05:
## Issue Description ##
We are doing testing for our customer Antevia.

On more recent "test" repository software loads we have seen the srsRAN stop working and the following error message streaming in the gnb.log file:

**_2024-06-04T08:47:32.873004 [MAC     ] [W] [   367.6] Skipped slot indication=367.11. Cause: DL task queue is full.

2024-06-04T08:47:32.873502 [MAC     ] [W] [   367.6] Skipped slot indication=367.12. Cause: DL task queue is full.

2024-06-04T08:47:32.874001 [MAC     ] [W] [   367.6] Skipped slot indication=367.13. Cause: DL task queue is full._**

When the srsRAN stack get's into this state we have to restart it.

We've seen this issue creep into more recent loads from April 2024 onwards.

I've tried a May21st version of the "test" repository and I still see the issue.

When I use a Feb23rd version of the "test" repository I don't see the problem.

## Setup Details ##
Using a Picocom 802R RU with the srsRAN, with uFalcon PTP grandmaster in a very similar fashion as described on your website set-up

## Expected Behavior ##
This is a bug and needs to be resolved in order for the srsRAN stack to be more resilient

## Actual Behaviour ##
Not see this issue

## Steps to reproduce the problem ##
Use a Picocom 802R Split 7.2 set-up as detailed on your website

## Additional Information ##
I believe the issue has crept into more recent software loads since April 2024

pgawlowicz, 2024-06-05:
Hi, could you share your gnb config and logs?

wozzamac, 2024-06-06:
Hi Piotr,

Sorry for not getting back more quickly.

Please find attached the srsRAN config that we're using.

I will capture a gnb.log soon and also get that too you, unfortunately we've wiped our previous log file capturing the issue.

Regards,
[gnb_ru_picocom_scb_tdd_n77_30mhz_May21.zip](https://github.com/user-attachments/files/15622294/gnb_ru_picocom_scb_tdd_n77_30mhz_May21.zip)

Warren

wozzamac, 2024-06-18:
Hi Piotr,

Sorry for the delay in getting back to you as it took some time for the system to repeat the issue.

Please find attached a config file and gnb log file showing the problem.

The system was running for a number of hours with one UE connected streaming a BBC media story and then the srsRAN get's into a state where the "DL task queue full" error message just constantly streams and the srsRAN stack becomes unresponsive and has to be rebooted.

[srsRAN_DL_Task_Queue_Issue.zip](https://github.com/user-attachments/files/15882095/srsRAN_DL_Task_Queue_Issue.zip)

ismagom, 2024-06-18:
It doesn't seem to be a bug. Your configuration is not correct for the performance you are trying to achieve. You should use DPDK and also pin the threads according to your CPU architecture. 

Use a discussion post to get advice from the community.

closed 2024-06-18 by ismagom