srsran/srsRAN_Project #654: gNB Handover using two USRP?

BGarcs, 2024-06-05:
Hi everyone, I have been reading the gNB Handover tutorial and I wanted to ask if it is possible to implement this functionality using two USRP b210 instead of one x310 (or similar USRP). Has anyone tried this or know if it is possible to do it?

Thanks in advance

pgawlowicz, 2024-06-07:
Hi, unfortunately it is not supported, and not planned to be supported soon.

pgawlowicz, 2024-06-07:
Actually, it will be supported once we support CU/DU split. Specifically, each DU will be running as an independent process and will connect to its own USRP.

closed 2024-06-20 by pgawlowicz

fllay, 2024-07-06:
Is it support now?

pgawlowicz, 2024-07-08:
yes, it should be possible, just ran two DUs, each with own usrp, most probably you will need to both usrps to a common ref clock.

Jonas-Penailillo, 2024-07-24:
Has anyone tried it?

fllay, 2024-07-25:
yes, it works just fine for forcing handover however, for measuring handover it is not complete yet as discussed here [https://github.com/srsran/srsRAN_Project/discussions/715](url)

frankist, 2024-07-26:
>yes, it works just fine for forcing handover however, for measuring handover it is not complete yet as discussed here [https://github.com/srsran/srsRAN_Project/discussions/715](https://github.com/srsran/srsRAN_Project/issues/url)

Wrong link?

The common ref clock is important for the RRC measurements to work well. If the two USRPs are not well synchronized, CFO can make the neighbor cell signal look very weak.

stavrosdim, 2025-03-26:
Can anyone provide a tutorial or maybe the yml files used for this? I am trying to complete this using 2 B210s with srsRAN gNB 24.04, but have not had any luck making the UE report the neighboring cell measurements.

Thanks in advance.