srsran/srsRAN_Project #657: no internet connection [BladeRF xA4] connected to  [srsRAN (docker)] connected to [open5gs (docker)]

MamdouhEgy, 2024-06-06:
## Issue Description ##
I followed [open5gs](https://github.com/srsran/srsRAN_Project/tree/main/docker/open5gs).

## Setup Details ##
Host OS: Ubuntu20.04
bladeRF: micro xA4 
SIM cards: SysmoISIM-SJA2
UE: Xiaomi 11 Lite 5G NE

I followed this sequence of commands:
**docker build --target open5gs -t open5gs-docker .**   #create the image from the dockerfile

**docker network create --subnet=10.53.1.0/24 open5gsnet**    #create the container network switch

**docker run --net open5gsnet --ip 10.53.1.2 --env-file open5gs.env --privileged --publish 3000:3000 open5gs-docker ./build/tests/app/5gc -c open5gs-5gc.yml** # create container based on the above created image and passing some arguments to it like to have specific ip and to mirror port 3000 between host and container

**docker run -it --device=/dev/bus/usb/002/003 --network open5gsnet 3c8dd6ec0655 /bin/bash** #here i run a container from customised [pavonis/srs-gnb-blade](https://hub.docker.com/r/pavonis/srs-gnb-blade) images, in which i adjusted the file gnb_rf_b200_tdd_n78_20mhz.yml [image] 

**cd /opt/srsRAN_Project/configs**

**gnb -c gnb_rf_b200_tdd_n78_20mhz.yml**

then i get that for the srsRAN container: [image]

and get for the Open5gs container:
[image]

then i disable the aeroplane mode in my UE and i got an IP that i predefined in the open5gs.env as follows:
[image]
and also defined it in add_users.py file as follows:

[image]
[image]

In my UE i got that:

[image]

In Open5GS container i got (u see the Open5gs recognized the UE and the IP address is the same as in the previous screenshot):

[image]

when access **localhost:3000**, these are the configurations:

[image]

## Expected Behavior ##
I expect internet connection on my UE

## Actual Behaviour ##
no internet connection as in the UE screenshots

## Additional Information ##
i followed this guide but i think it didnot help. (step1-3 are right as in screenhots, step 4: Restart all Open5GS services and try again, i restarted the open5gs container then again started the srsRAN container but didnot help)

screenshot of **ip address** command on my host:
[image]

screenshot of **ping 10.45.0.4** command (to ping the UE from the open5gs container):
[image]

pgawlowicz, 2024-06-06:
Please follow our tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html

In addition, you will need to enable IP forwaring on the PC running open5gs. Please check open5gs tutorial: https://open5gs.org/open5gs/docs/guide/01-quickstart/

It should work with:
```
sudo sysctl -w net.ipv4.ip_forward=1
sudo iptables -t nat -A POSTROUTING -o <IFNAME> -j MASQUERADE

Where <IFNAME> is the name of the interface connected to the internet.
```

MamdouhEgy, 2024-06-13:
@pgawlowicz Thanks!! this was the solution and now everything worked as expected.

closed 2024-06-13 by MamdouhEgy