srsran/srsRAN_Project #670: iperf3 doesn't work when eUPF connected to srsRAN

s5uishida, 2024-06-13:
Hi Community,

Sorry for linking an issue I wrote elsewhere.

https://github.com/edgecomllc/eupf/issues/555

If you know anything about the above issue on the srsRAN side (gNodeB and eNodeB), I would be very glad if you could let me know.

Best regards,

--Shigeru

herlesupreeth, 2024-06-13:
@s5uishida I would check the MTU set in smf.yaml and also the MTU of the of eUPF interface. Set both to 1450 and give it a try

s5uishida, 2024-06-13:
Hi @herlesupreeth 

Thank you very much for your reply.

But it didn't work. The results were the same.

For C-Plane, only smf.yaml was changed to `mtu: 1450`.
On the other hand, for eUPF, I directly changed the N3/N4/N6 interfaces to `mtu: 1450`.

I would be very glad if you could point out any mistakes in my configuration.

Best regards,

--Shigeru

herlesupreeth, 2024-06-13:
Can you post logs of gNB and UE in info level when the UE attaches ? Also, can you mention what are the IPs of the machine running gNB, UE, open5gs C-plane and eUPF U-plane?

s5uishida, 2024-06-13:
Hi @herlesupreeth 

```
       gNB - 192.168.66.121
        UE - 192.168.66.122
Open5GS CP - 192.168.66.111
      eUPF - 192.168.66.151
     DN/GW - 192.168.66.152 (iperf3 -s running)
```
In the diagram in the link below, this is the network where 192.168.0.0/24 has been changed to 192.168.66.0/24.

https://github.com/s5uishida/simple_measurement_of_upf_performance

The log and configuration files are as follows.

[ue.log.gz](https://github.com/user-attachments/files/15815468/ue.log.gz)
[ue_zmq.conf.gz](https://github.com/user-attachments/files/15815478/ue_zmq.conf.gz)

[gnb.log.gz](https://github.com/user-attachments/files/15815487/gnb.log.gz)
[gnb_zmq.yaml.gz](https://github.com/user-attachments/files/15815495/gnb_zmq.yaml.gz)

[log_open5gs.tar.gz](https://github.com/user-attachments/files/15815500/log_open5gs.tar.gz)
[etc_open5gs.tar.gz](https://github.com/user-attachments/files/15815503/etc_open5gs.tar.gz)

[log_eupf.log.gz](https://github.com/user-attachments/files/15815522/log_eupf.log.gz)
[config.yml.gz](https://github.com/user-attachments/files/15815523/config.yml.gz)

Thank you very much!

herlesupreeth, 2024-06-13:
Thank you @s5uishida for the logs. I didnt see any issue in either gNB nor UE and the entire attach process. In order to know more, can you post a pcap taken on Open5GS CP and eUPF machine for the following scenario:

1. Start the trace on Open5GS CP and eUPF
2. Attach the UE
3. Start the iperf
4. Stop the trace and post it here

s5uishida, 2024-06-13:
Hi @herlesupreeth 

I took the pcaps following your scenario.

[open5gs_cp_01.pcap.gz](https://github.com/user-attachments/files/15816211/open5gs_cp_01.pcap.gz)
[eupf_01.pcap.gz](https://github.com/user-attachments/files/15816214/eupf_01.pcap.gz)

Thank you very much!

herlesupreeth, 2024-06-13:
I think there is IP fragmentation going on

[image]

Can you try running iperf3 with following argument to set packet size and see if that works?

`--length 1450`

s5uishida, 2024-06-14:
Hi @herlesupreeth 

It worked with the following for both 5G and 4G. Note that the `mtu` in `smf.yaml` has not been changed.

1. Set the `mtu` of the eUFP's N6 interface to `1450`.
2. In the UE VM, set the `mtu` of the `tun_srsue` interface to `1400`.
  ```
  ip link set tun_srsue mtu 1400
  ```
However, it did not work with `mtu: 1450` of `tun_srsue`, or `--length 1450` option of iperf3.
Setting `mtu: 1400` or `--length 1400` worked.

This made it work without the `--length 1400` option.

Thank you for the very useful information!

closed 2024-06-14 by s5uishida

HaoxinSEU, 2024-06-20:
Hi @s5uishida,

We are also trying to work with eUPF and srsRAN. In our test, the UE (real phone) can connect and do a speedtest. However, the DL throughput is very low. When using a 20 MHz BW gNB (TDD), the DL throughput is 10 Mbps (normal value should be 55 Mbps), while the UL throughput is reasonable (20 Mbps). Have you ever experienced similar throughput bottlenecks? It seems to be on the session, instead of gNB, because connecting two UEs results in each UE with 10 Mbps DL....

Thank you for your help!

Best regards,
Haoxin

andrepuschmann, 2024-06-20:
Can you post the console trace? Check the dl_bs column. Likely UPF not pushing enough data.

HaoxinSEU, 2024-06-20:
Hi Andre,

You are right, the DL buffer status is mostly empty. And we also checked the gNB RB allocation, most of them are not used. Here are the console logs of gNB when running a speedtest on DL.
```
           -----------------DL-----------------------|------------------UL--------------------
 pci rnti  cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  mcs  brate   ok  nok  (%)    bsr
   1 4601   15   1   27   1.0M  104    0   0%      0 |  32.9   27   355k   73    0   0%      0
   1 4601   15   1   27   1.5M  169    0   0%      0 |  33.8   27   475k   99    0   0%      0
   1 4601   15   1   27   1.3M  188    1   0%      0 |  30.4   27   444k   90   10  10%      0
   1 4601   15   1   26   964k  119    1   0%      0 |  36.1   27   516k  106    0   0%      0
   1 4601   15   1   26   708k   96    2   2%      0 |  35.1   27   501k  103    0   0%      0
   1 4601   15   1   26   865k  140    1   0%    140 |  30.5   27   621k  118    8   6%      0
   1 4601   15   1   26   2.8M  215    0   0%      0 |  23.0   20   748k  139   17  10%      0
   1 4601   15   1   26   9.4M  559    2   0%  1.41k |  22.8   21   451k  118   35  22%    745
   1 4601   15   1   27   9.9M  609    1   0%      0 |  22.6   21   601k  118   39  24%      0
   1 4601   15   1   27    13M  719    0   0%      0 |  26.9   26   648k  117   26  18%      0
   1 4601   15   1   27    15M  801    1   0%      8 |  31.6   27   669k  139    4   2%  1.04k

           -----------------DL-----------------------|------------------UL--------------------
 pci rnti  cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  mcs  brate   ok  nok  (%)    bsr
   1 4601   15   1   27    16M  835    2   0%      0 |  32.4   27   645k  149    0   0%    535
   1 4601   15   1   27    14M  748    2   0%      0 |  27.4   26   1.0M  120   16  11%    745
   1 4601   15   1   27    14M  717    2   0%      0 |  27.1   26   646k  138   10   6%     28
   1 4601   15   1   27    12M  618    3   0%      8 |  29.3   26   504k  110   19  14%      0
   1 4601   15   1   27    15M  750    0   0%      0 |  31.7   27   677k  144    5   3%      0
   1 4601   15   1   27    15M  769    3   0%      0 |  32.8   27   641k  140    2   1%      0
   1 4601   15   1   27    16M  767    0   0%      0 |  33.5   27   689k  153    0   0%  1.45k
   1 4601   15   1   27    15M  717    1   0%    850 |  35.1   27   686k  148    0   0%      0
   1 4601   15   1   27    15M  734    1   0%      0 |  33.9   27   652k  142    1   0%      0
   1 4601   15   1   27    15M  725    1   0%  1.41k |  34.2   27   674k  145    0   0%    102
   1 4601   15   1   26    15M  736    0   0%     11 |  33.3   27   697k  151    2   1%  2.81k
```