srsran/srsRAN_Project #671: Monitor xApp FlexRIC with last version of srsRAN

apperezG, 2024-06-13:
## Issue Description ##
Hi, I am using the latest version of srsRAN release_24_04 and I am having problems with the FlexRIC monitoring xApp. The RIC execution stops as soon as I launch the app with the following error:
_nearRT-RIC: /home/vscode/flexric/src/sm/kpm_sm/kpm_sm_v02.03/dec/kpm_dec_asn.c:184: kpm_dec_ind_msg_asn: Assertion `rval.code == RC_OK && "Are you sending data in ATS_ALIGNED_BASIC_PER syntax?"' failed._

What commit are you using from the br-flexric branch? I don't know if I'm using a different one that is not compatible.

I was previously using srsRAN version 23_05 and had no problem with this.

## Setup Details ##
FlexRIC branch br-flexric last commit
srsRAN Project last release
srsRAN 4G UE
Open5Gs core

## Expected Behavior ##
The monitoring xApp obtains the data.

## Actual Behaviour ##
The RIC execution stops as soon as I launch the app with the following error:
_nearRT-RIC: /home/vscode/flexric/src/sm/kpm_sm/kpm_sm_v02.03/dec/kpm_dec_asn.c:184: kpm_dec_ind_msg_asn: Assertion `rval.code == RC_OK && "Are you sending data in ATS_ALIGNED_BASIC_PER syntax?"' failed._

brendan-mcauliffe, 2024-06-14:
Hi @apperezG, 

Please check the documentation as it has been updated recently. You can find the tutorial relating to the use of nRT RICs [here](https://docs.srsran.com/projects/project/en/latest/tutorials/source/near-rt-ric/source/index.html). 

From the documentation you will see that you need to use `commit: 1a3903a7` from br-flexric. 

Please re-try while following the latest documentation.

apperezG, 2024-06-17:
Hi @brendan-mcauliffe,
I am testing with that specific commit and with srsRAN version 24_04 (following the guide exactly) and still the xapp_oran_moni is not working, giving the same error.
I am using the configuration files you provide.

apperezG, 2024-06-19:
Hi @brendan-mcauliffe,
I also tried with the commit 0b2702cca of srsRAN Project but it gives me the same error:
"kpm_dec_ind_hdr_asn: Assertion `rval.code == RC_OK && "Are you sending data in ATS_ALIGNED_BASIC_PER syntax?"' failed."

pgawlowicz, 2024-06-20:
@apperezG could you check if this solution (a patch that removes `granul_period` from the RIC indication message) works for you?
https://github.com/srsran/srsRAN_Project/discussions/567#discussioncomment-9189455

apperezG, 2024-06-21:
@pgawlowicz With these changes it is now working properly! 

Thank you very much for your help @pgawlowicz @brendan-mcauliffe

closed 2024-06-21 by apperezG