srsran/srsRAN_Project #673: Error while executing sudo apt-get install srsran -y

amitpalkar, 2024-06-14:
## Issue Description ##
Error while executing sudo apt-get install sran -y

## Setup Details ##
Ubuntu, version 22.04.4

## Expected Behavior ##
Expected to run successfully.

## Actual Behaviour ##
Seeing the error message "Unable to locate package srsran"

## Steps to reproduce the problem ##

## Additional Information ##

Can someone tell me what the issue is?

brendan-mcauliffe, 2024-06-14:
Hi @amitpalkar please refer to the documentation, specifically the [Installation Guide](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/installation.html#packages). 

The correct commands to install from packages are as follows: 
```
sudo add-apt-repository ppa:softwareradiosystems/srsran-project
sudo apt-get update
sudo apt-get install srsran-project -y
```

closed 2024-06-14 by brendan-mcauliffe

amitpalkar, 2024-06-15:
Hi,
   I tried the above command sequence.  It worked for the first 2 commands.  For the third command sudo apt-get install srsran-project -y it gave a failure. "Unable to locate package srsran".   Are there some machine setting issues or permissions which i need to take care of?  I have priviledge for all sudo commands..
Regards,
Amit Palkar

brendan-mcauliffe, 2024-06-17:
Hi @amitpalkar, are you sure you are copying the command correctly? it should be 
```
sudo apt-get install srsran-project -y
```
If you are seeing an error that reads ``Unable to locate package srsran``, then you most likely are trying to install the wrong package. 
Try again making sure that you are looking for ``srsran-project`` **not** ``srsran``.