srsran/srsRAN_Project #674: OFH - optimum IQ Amplitude

RafalArciszewski, 2024-06-14:
I am integrating srsRAN DU with RU from Metanoia via OFH interface. So far S-PLANE, C-PLANE and U-PLANE is working more or less stable. Now I struggle with tuning the IQ samples amplitudes, FullScale and FullScale Offset configuration.

**Downlink**
With iq_scaling = 1.0 the BPSK amplitudes are -2,23 to 2,23, QPSK aplitudes are -1,57, 1,57

[image]

[image]

When I set iq_scaling to maximum 10.0 the BPSK amplitudes are -22,3 to 22,3, QPSK aplitudes are -15,7, 15,7 as expected.

[image]

[image]

The problem is that RU requires even bigger amplitudes, is there any way to increase it in the configuration?

**Uplink**
I have a possibility to bump up the IQ samples amplitudes in RU but I don't know what are the optimum values for srsRAN DU? Could you please advice?

I attach pcaps with one slot with SSB in symbols 2,3,4,5.

Regards,
Rafał Arciszewski
[ssb_iq_scaling.zip](https://github.com/user-attachments/files/15837153/ssb_iq_scaling.zip)

andrepuschmann, 2024-06-14:
Hey @RafalArciszewski - the `iq_scaling` value is limited to `20` currently because we never had to configure a value above. But it very much depends on the RU firmware as you said. Some RUs have an internal amplification that can be configured over the RU config. If you want to increase "software gain" further just patch the limit here https://github.com/srsran/srsRAN_Project/blob/c33cacba7d940e734ac7bad08935cbc35578fad9/apps/units/flexible_du/split_7_2/ru_ofh_config_cli11_schema.cpp#L169 and recompile the code.

The optimum value is hard to say, probably the best is to consult the RU vendor and/or hook up a NR waveform analyzer and try to fine-tune values to get the best signal quality, e.g. by minimizing EVM.

RafalArciszewski, 2024-06-14:
Thanks, I don't know why I assumed max iq_scaling is 10 (maybe it was the limit in previous releases). 

What about the Uplink? What is srsRAN Fullscale? For example what typical IQ values are received for a background noise?  Is it <0,1> range or <0,256> or <0, 32768>?

RafalArciszewski, 2024-06-25:
Hi @andrepuschmann 
I have tuned the RU and now I get average IQ module ~ 4 for the noise signal. However I have a problem with PRACH procedure. The UE is doing the PRACH. I can see increasing power (according to ramping step from SIB) on oran interface  but on srsRAN gnb.log there is no information about the recognized preambles.  
I gathered gnb.log on debug level + fronthaul pcap.  Uplink PCAP post-analysis shows that there is increse power in each step. Each row is one slot.

[image]

In gnb.log for 200.19 (oran slot numbering 200-9-1) I have following entries. No indication there was a PRACH received but I am pretty sure that UE have sent it bacause on oran the IQsamples power are quite high.
 
[image]

Could you take a look at this?
* are the uplink IQSamples values ok for srsRAN?
* do I have a PRACH eAXC mismatch? In the logs PRACH is sent to eAxC '4' and '5' but Handling is done on ports '0' and '1'
* Is there any log that shows if decoding the preambles was a success?

Regards,
Rafał Arciszewski

[gnb_log.zip](https://github.com/user-attachments/files/15965957/gnb_log.zip)
[oran_uplink_prach.zip](https://github.com/user-attachments/files/15965958/oran_uplink_prach.zip)
[vw_metanoia_2x2_10Mhz.zip](https://github.com/user-attachments/files/15965961/vw_metanoia_2x2_10Mhz.zip)

closed 2024-07-01 by RafalArciszewski