srsran/srsRAN_Project #686: Number of symbols is 0 in O-RAN Fronthaul C-Plane for PRACH

terry-hasegawa, 2024-06-22:
## Issue Description ##
When using Long Preamble, the number of symbols included in the O-RAN Fronthaul C-Plane for PRACH is 0.
The number of symbols is defined by O-RAN WG4 CUS as follows:
[image]

However, when using Long Preamble Format 0, the number of symbols is 0.
[image]

## Setup Details ##
Set up srs gNB with Long Preamble Format.

## Expected Behavior ##
An appropriate number of symbols is set in the Fronthaul packet.

## Actual Behaviour ##
The Fronthaul C-Plane for PRACH has number of symbols set to 0.

## Steps to reproduce the problem ##
Set up srs gNB with Long Preamble Format.

## Additional Information ##
For Short Preamble, the number of symbols is set in lib\ran\prach\prach_preamble_information.cpp::get_prach_preamble_short_info(). 
However, for Long Preamble, the number of symbols is not set in lib\ran\prach\prach_preamble_information.cpp::get_prach_preamble_long_info().

get_prach_preamble_long_info()
[image]

get_prach_preamble_short_info()
[image]

andrepuschmann, 2024-06-25:
The issue should solved in current `main`. Could you retest and check please?

terry-hasegawa, 2024-06-26:
Hi @andrepuschmann, This issue was also reproduced on the main branch. Please check.

andrepuschmann, 2024-06-26:
I just double checked with current main and the config @wei-lou posted and I confirm that it works. See here:
[image]

I guess you are using an older version.

wei-lou, 2024-06-26:
Hi @andrepuschmann ,

This is the link I pulled the latest 24.04 load and then compiled. Is this the right link for the latest load or there are some other links? Thanks,

https://github.com/srsran/srsRAN_Project/archive/refs/tags/release_24_04.tar.gz

[image]

andrepuschmann, 2024-06-26:
The release is from April 22nd. You need to git clone the repo and use the main branch to get the latest code. We only do tagged releases every 6 months.

wei-lou, 2024-06-26:
> The release is from April 22nd. You need to git clone the repo and use the main branch to get the latest code. We only do tagged releases every 6 months.

> The release is from April 22nd. You need to git clone the repo and use the main branch to get the latest code. We only do tagged releases every 6 months.

Got it, thanks for the clarification

terry-hasegawa, 2024-06-26:
@andrepuschmann Did git pull, now the numsymbol is 1. Thank you.
[image]

closed 2024-06-26 by terry-hasegawa