srsran/srsRAN_Project #688: KPM xApp causes the ric stop

Amjed468, 2024-06-24:
Dear all,
I have set up the srsRAN project, and I am currently trying to execute the monitoring xApp:

./flexric/build/examples/xApp/c/monitor/xapp_oran_moni -c ./xapp_mon_e2sm_kpm.conf

But, the Flexric stop suddenly during the execution:

./flexric/build/examples/xApp/c/monitor/xapp_oran_moni -c ./xapp_mon_e2sm_kpm.conf :
nearRT-RIC: /home/labadmin/flexric/src/sm/kpm_sm/kpm_sm_v03.00/dec/kpm_dec_asn.c:186: kpm_dec_ind_msg_asn: Assertion `rval.code == RC_OK && "Are you sending data in ATS_ALIGNED_BASIC_PER syntax?"' failed.
Aborted (core dumped)

Any suggestion please ?

yagoda, 2024-06-24:
Hi, 

First of all this error comes from FlexRic so you are best off asking them about it.

Having said that if you can share the configs, logs and a pcap we can take a look.

Regards,
Justin

closed 2024-06-24 by pgawlowicz

Amjed468, 2024-06-25:
> Dear Justin,
> Thanks for your swift response. Here are the logs and configuration files.
>
> Best Regards

Dear Justin,
Thanks for your help, everything works fine now with UE simulation. How can
I run the kpm_rc, as I tried to execute it directly which causes the
execution to be aborted

.

Regards

Amjed468, 2024-06-27:
Dear community,

Recently I have been able to run the "xapp_oran_moni" by following the
instruction in (
https://docs.srsran.com/projects/project/en/latest/tutorials/source/near-rt-ric/source/index.html).
Now I am trying to run a control xApp from Flexric like "xapp_kpm_rc" but
the Flexric stop suddenly as below:

Pending event timeout happened. Communication with E2 Node lost?
nearRT-RIC: /home/labadmin/flexric/src/ric/near_ric.c:270: consume_fd:
Assertion `fd > 0' failed.
Aborted (core dumped)

-- 
*Best Regards...*

*Amjed H. J. MohammedPhD Candidate / University of Glasgow / UK*

pgawlowicz, 2024-06-28:
Could you provide the pcap file with the e2 messages exchanged between srsgnb and flexric?

Amjed468, 2024-06-30:
Dear Sir,
The pcap file got empty when I ran the xApp of kpm_rc. Is this xApp
supposed to work normally with the srsRAN project and Flexric ?

Regards

pgawlowicz, 2024-07-01:
did you enable e2 pcap collection in the gnb config? it should contain at least setup request and response.

Amjed468, 2024-07-01:
Dear Sir,
I have been able to get pcap file with reading when I added the config file
of kpm xApp like this :
./flexric/build/examples/xApp/c/kpm_rc/xapp_kpm_rc -c
./xapp_mon_e2sm_kpm.conf

Best Regards