srsran/srsRAN_Project #693: srsRAN 4G RAN and UE: Cannot connect iperf3 server for bandwidth testing

Eric-Wang738, 2024-06-27:
## Issue Description ##
I installed srsRAN 4G RAN, UE with ZeroMQ by following instruction.
https://github.com/s5uishida/build_srsran_4g_zmq_disable_rf_plugins?tab=readme-ov-file#install_libs
https://docs.srsran.com/projects/4g/en/latest/app_notes/source/zeromq/source/index.html#zeromq-appnote

## Setup Details ##
**With configuraiton file of RAN and UE, I can attach UE to core and tried to ping DN server(192.168.2.2 as iperf3 server) behind the core.**
[enb.conf.txt](https://github.com/user-attachments/files/16010497/enb.conf.txt)
[ue.conf.txt](https://github.com/user-attachments/files/16010498/ue.conf.txt)

_root@srsran02:~# ip netns exec ue1 ip a
1: lo: <LOOPBACK> mtu 65536 qdisc noop state DOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
7: tun_srsue: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 500
    link/none
    inet 10.4.0.3/24 scope global tun_srsue
       valid_lft forever preferred_lft forever

root@srsran02:~# ip netns exec ue1 ping 192.168.2.2
ping: connect: Network is unreachable_

**When I specify interface, PING command is stopping there. But I can see ICMP is replied from tcpdump command.**
root@srsran02:~# ip netns exec ue1 ping -I tun_srsue 192.168.2.2
PING 192.168.2.2 (192.168.2.2) from 10.4.0.3 tun_srsue: 56(84) bytes of data.

root@srsran02:~# ip netns exec ue1 tcpdump -i any
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked v1), capture size 262144 bytes
08:30:49.210320 IP 10.4.0.3 > 192.168.2.2: ICMP echo request, id 40445, seq 16, length 64
08:30:49.239230 IP 192.168.2.2 > 10.4.0.3: ICMP echo reply, id 40445, seq 16, length 64
08:30:50.234322 IP 10.4.0.3 > 192.168.2.2: ICMP echo request, id 40445, seq 17, length 64
08:30:50.258347 IP 192.168.2.2 > 10.4.0.3: ICMP echo reply, id 40445, seq 17, length 64
08:30:51.258322 IP 10.4.0.3 > 192.168.2.2: ICMP echo request, id 40445, seq 18, length 64
08:30:51.277286 IP 192.168.2.2 > 10.4.0.3: ICMP echo reply, id 40445, seq 18, length 64
08:30:52.282389 IP 10.4.0.3 > 192.168.2.2: ICMP echo request, id 40445, seq 19, length 64
08:30:52.317507 IP 192.168.2.2 > 10.4.0.3: ICMP echo reply, id 40445, seq 19, length 64

**And there is also date counted in UE console trace**

---------Signal-----------|-----------------DL-----------------|-----------UL-----------
 cc  pci  rsrp   pl   cfo | mcs  snr  iter  brate  bler  ta_us | mcs   buff  brate  bler
  0    1    62  -62 -206n |  15  139   1.0    890    0%    0.0 |  22    0.0   8.8k    0%
  0    1    62  -62 -308n |  15  139   1.0    889    0%    0.0 |  22    0.0   8.8k    0%
  0    1    62  -62 -281n |  15  139   1.0    890    0%    0.0 |  22    0.0   8.8k    0%
  0    1    62  -62 -210n |  15  139   1.0    889    0%    0.0 |  22    0.0   8.8k    0%
  0    1    62  -62 -260n |  15  139   1.0    889    0%    0.0 |  22    0.0   8.8k    0%
  0    1    62  -62 -279n |  15  139   1.0    889    0%    0.0 |  22    0.0   8.8k    0%
  0    1    62  -62 -276n |  15  139   1.0    890    0%    0.0 |  22    0.0   8.8k    0%
  0    1    62  -62 -223n |  15  139   1.0    890    0%    0.0 |  22    0.0   8.8k    0%
  0    1    62  -62 -348n |  15  139   1.0    890    0%    0.0 |  22    0.0   8.8k    0%
  0    1    62  -62 -232n |  15  139   1.0    889    0%    0.0 |  22    0.0   8.8k    0%
  0    1    62  -62 -268n |  15  139   1.0    889    0%    0.0 |  22    0.0   8.8k    0%

**But when I execute iperf3 command to connect iperf server, it's showing Network is unreachable whatever I specified source IP or not.**

[image]

## Expected Behavior ##
It there anything missing in my configuration file to connect iperf server?

pgawlowicz, 2024-06-27:
Hi, this repository is about the 5G srsRAN-Project, and you are asking about the srsRAN-4G project. Please ask there in the Discussion section, as this is not an issue, but rather a routing misconfiguration.

closed 2024-06-27 by pgawlowicz

Eric-Wang738, 2024-06-27:
Sorry for sending wrong working group. My problem was resolved by your comment "a routing misconfiguration". Thanks for you hint.