srsran/srsRAN_Project #700: Unable to do iperf3 in downlink (from core to UE) in 5G SA mode (Zero MQ mode)

amitpalkar, 2024-07-03:
## Issue Description ##
Unable to do iperf3 in Downlink for 5G SA mode with ZMQ mode.

## Setup Details ##
PRETTY_NAME="Ubuntu 22.04.4 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.4 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

## Expected Behavior ##
I am using SRS RAN project for 5G testing in zeroMQ mode.  I am running a dockerized version of the 5G core in a container. 
I am able to attach UE.  I am able to ping in both directions DL and UL.  I am able to do an iperf3 in uplink and able to do data transfer.  In downlink, i am unable to do data transfer.
I enter the container using the below command:
   sudo docker exec -it <CONTAINER ID> /bin/bash

Inside the container (running the 5G Core) i give the below command
iperf -c 10.45.1.2 -p 5201 -u -b 50M -i 1 -t 60

On another terminal i give the below command:
sudo ip netns exec ue1 iperf -s -i 1
I do not see any data only see the below
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------

Note:
a) My output of route -n is given below
   Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.1.115.1      0.0.0.0         UG    100    0        0 eno1
10.1.115.0      0.0.0.0         255.255.255.0   U     100    0        0 eno1
10.45.0.0       10.53.1.2       255.255.0.0     UG    0      0        0 br-d32e126b4140
10.45.0.0       0.0.0.0         255.255.0.0     U     0      0        0 ogstun
10.53.1.0       10.53.1.2       255.255.255.0   UG    0      0        0 br-d32e126b4140
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-f7bb8ae0ec84
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 br-6aac9306bce6
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-6aac9306bce6
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-069322a98f54

b) i am unable to use the command "iperf3 -c 10.45.1.2 -u -b 65M -i 1 -t 30" inside the container since it gives an error saying bad file descriptor

c) However when i run the command "iperf3 -s -i 1" inside the container for uplink traffic, then i get the data sent from UE to the UPF.

d) I am able to do a ping 10.45.1.2 from inside the container.  (UE Ip address is 10.45.1.2)

e) Console output from the UE given below
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ../configs/ue_zmq.conf...

Built in Release mode using commit ec29b0c on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:3001,rx_port=tcp://localhost:3000,id=ue,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://localhost:3000
CH0 tx_port=tcp://*:3001
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=174
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
PDU Session Establishment successful. IP: 10.45.1.2
RRC NR reconfiguration successful.

pgawlowicz, 2024-07-03:
you have multiple routing entries with the same destination but over different interfaces:
```
10.45.0.0       10.53.1.2       255.255.0.0     UG    0      0        0 br-d32e126b4140
10.45.0.0       0.0.0.0         255.255.0.0     U     0      0        0 ogstun
10.53.1.0       10.53.1.2       255.255.255.0   UG    0      0        0 br-d32e126b4140
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-f7bb8ae0ec84
```

Check here: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#nd-open5gs-instance-installed-manually

amitpalkar, 2024-07-03:
Hi,
    Thanks a lot.  I have few questions.  
a) How is ping working in both directions?  
b) how come iperf3 is working in uplink directions?    
c) Which entries (network interfaces) do i remove from the table?  Should i use ip route delete command

amitpalkar, 2024-07-03:
default via 10.1.115.1 dev eno1 proto dhcp metric 100 
10.1.115.0/24 dev eno1 proto kernel scope link src 10.1.115.50 metric 100 
10.45.0.0/16 via 10.53.1.2 dev br-d32e126b4140 
10.45.0.0/16 dev ogstun proto static 
10.53.1.0/24 via 10.53.1.2 dev br-d32e126b4140 
10.53.1.0/24 dev br-f7bb8ae0ec84 proto kernel scope link src 10.53.1.1 linkdown 
169.254.0.0/16 dev br-6aac9306bce6 scope link metric 1000 
172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 linkdown 
172.19.1.0/24 dev br-6aac9306bce6 proto kernel scope link src 172.19.1.1 
172.19.1.0/24 dev br-069322a98f54 proto kernel scope link src 172.19.1.1 linkdown 

The above is the output of command ip route.

pgawlowicz, 2024-07-03:
a) How is ping working in both directions?
Because you are pinging from UE to open5gs docker container. You cannot ping any other address than 10.45.0.1 from the UE. what is missing is the correct routing entry in on the PC hosting open5gs docker container.

b) how come iperf3 is working in uplink directions?
same as above, iperf3 server running in the container. 

c) Which entries (network interfaces) do i remove from the table? Should i use ip route delete command
you have to keep the entries that go over the bridge interface connected to the open5gs container, probably those:
```
10.45.0.0       10.53.1.2       255.255.0.0     UG    0      0        0 br-d32e126b4140
10.53.1.0       10.53.1.2       255.255.255.0   UG    0      0        0 br-d32e126b4140
```
and remove:
``
10.45.0.0       0.0.0.0         255.255.0.0     U     0      0        0 ogstun
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-f7bb8ae0ec84
``

amitpalkar, 2024-07-03:
Hi,
    I deleted the routes as shown by you.  Below is the output from route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.1.115.1      0.0.0.0         UG    100    0        0 eno1
10.1.115.0      0.0.0.0         255.255.255.0   U     100    0        0 eno1
10.45.0.0       10.53.1.2       255.255.0.0     UG    0      0        0 br-d32e126b4140
10.53.1.0       10.53.1.2       255.255.255.0   UG    0      0        0 br-d32e126b4140
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 br-6aac9306bce6
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-6aac9306bce6
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-069322a98f54

b) After doing this still iperf in downlink is not happening

c) I am using iperf and not iperf3 since using iperf3 is giving me an error message bad descriptor.

d) sudo ip netns exec ue1 iperf -s -i 1
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------

I do not use any packets received

amitpalkar, 2024-07-03:
root@5313df0a2f1f:/open5gs# iperf -c 10.45.1.2 -u -b 65M -i 1 -t 30
------------------------------------------------------------
Client connecting to 10.45.1.2, UDP port 5001
Sending 1470 byte datagrams, IPG target: 172.54 us (kalman adjust)
UDP buffer size:  208 KByte (default)
------------------------------------------------------------
[  1] local 10.45.1.1 port 36719 connected with 10.45.1.2 port 5001
[ ID] Interval       Transfer     Bandwidth
[  1] 0.0000-1.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 1.0000-2.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 2.0000-3.0000 sec  8.12 MBytes  68.1 Mbits/sec
[  1] 3.0000-4.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 4.0000-5.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 5.0000-6.0000 sec  8.12 MBytes  68.1 Mbits/sec
[  1] 6.0000-7.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 7.0000-8.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 8.0000-9.0000 sec  8.12 MBytes  68.1 Mbits/sec
[  1] 9.0000-10.0000 sec  8.13 MBytes  68.2 Mbits/sec

i give the above command on a bash shell after i enter the container

pgawlowicz, 2024-07-03:
could you print routing table at UE?
```
sudo ip netns exec ue1 route -n
```

amitpalkar, 2024-07-03:
sudo ip netns exec ue1 route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
10.45.1.0       0.0.0.0         255.255.255.0   U     0      0        0 tun_srsue

pgawlowicz, 2024-07-03:
so the ue has no idea how to reach other networks.
Please add the default route:
```
sudo ip netns exec ue1 ip route add default via 10.45.1.1 dev tun_srsue
```

amitpalkar, 2024-07-03:
Should the below route also be present in UE side??
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.45.1.1       0.0.0.0         UG    0      0        0 tun_srsue

pgawlowicz, 2024-07-03:
yes, please check in the tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#id9

amitpalkar, 2024-07-03:
I have modified the route also on the UE side.  Below is the output. 
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.45.1.1       0.0.0.0         UG    0      0        0 tun_srsue
10.45.1.0       0.0.0.0         255.255.255.0   U     0      0        0 tun_srsue

a) Even after doing this, iperf is not working in DL.  Note: i am unable to use iperf3 command inside the container because of error message as shown below.
   iperf3 -c 10.45.1.2 -u -b 65M -i 1 -t 30
   iperf3: error - unable to send control message: Bad file descriptor

pgawlowicz, 2024-07-03:
could you print routeing table on the pc running open5gs container?

amitpalkar, 2024-07-03:
iperf -c 10.45.1.2 -u -b 65M -i 1 -t 30  ===> This command is used from bash shell inside the container.
sudo ip netns exec ue1 iperf -s -i 1     ===>  This command is used on another terminal

amitpalkar, 2024-07-03:
It is the same pc running the open5gs container.  Below is the output.. 
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.53.1.1       0.0.0.0         UG    0      0        0 eth0
10.45.0.0       0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.1.0       0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.2.0       0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.3.0       0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.4.0       0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.5.0       0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.6.0       0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.7.0       0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.8.0       0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.9.0       0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.10.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.11.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.12.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.13.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.14.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.15.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.16.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.17.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.18.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.19.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.20.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.21.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.22.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.23.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.24.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.25.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.26.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.27.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.28.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.29.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.30.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.31.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.32.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.33.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.34.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.35.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.36.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.37.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.38.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.39.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.40.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.41.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.42.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.43.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.44.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.45.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.46.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.47.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.48.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.49.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.50.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.51.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.52.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.53.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.54.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.55.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.56.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.57.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.58.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.59.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.60.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.61.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.62.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.63.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
10.45.64.0      0.0.0.0         255.255.255.0   U     0      0        0 ogstun
............
10.45.255.0     0.0.0.0         255.255.255.0   U     0      0        0 ogstun
[... cut ...]
127.0.0.0       0.0.0.0         255.255.255.0   U     0      0        0 lo19
127.0.0.0       0.0.0.0         255.255.255.0   U     0      0        0 lo20
127.0.0.0       0.0.0.0         255.255.255.0   U     0      0        0 lo21
127.0.0.0       0.0.0.0         255.255.255.0   U     0      0        0 lo22

Here is the output of route -n when i type it in a bash shell (after entering the container using the command sudo docker exec -it <CONTAINER ID> /bin/bash

Note: Ping is working from within the container to the UE (IP 10.45.1.2)

pgawlowicz, 2024-07-03:
Please share the routing table of the host PC

amitpalkar, 2024-07-03:
ee210469@pcz-mmalab-2:~/work/opensource_ran/srsran/5G/srsRAN_Project/build$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.1.115.1      0.0.0.0         UG    100    0        0 eno1
10.1.115.0      0.0.0.0         255.255.255.0   U     100    0        0 eno1
10.45.0.0       10.53.1.2       255.255.0.0     UG    0      0        0 br-d32e126b4140
10.53.1.0       10.53.1.2       255.255.255.0   UG    0      0        0 br-d32e126b4140
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 br-6aac9306bce6
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-6aac9306bce6
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-069322a98f54

This is the command i give outside the container on a command prompt

pgawlowicz, 2024-07-03:
ok, thanks, can you ping 10.53.1.2 from the host pc?

amitpalkar, 2024-07-03:
Hi,
    Yes i can ping 10.53.1.2 from the host Pc.
***@***.***:~/work/opensource_ran/srsran/5G/srsRAN_Project/build$
ping 10.53.1.2
PING 10.53.1.2 (10.53.1.2) 56(84) bytes of data.
64 bytes from 10.53.1.2: icmp_seq=1 ttl=64 time=0.036 ms
64 bytes from 10.53.1.2: icmp_seq=2 ttl=64 time=0.027 ms
64 bytes from 10.53.1.2: icmp_seq=3 ttl=64 time=0.030 ms
64 bytes from 10.53.1.2: icmp_seq=4 ttl=64 time=0.048 ms
64 bytes from 10.53.1.2: icmp_seq=5 ttl=64 time=0.034 ms
64 bytes from 10.53.1.2: icmp_seq=6 ttl=64 time=0.045 ms
64 bytes from 10.53.1.2: icmp_seq=7 ttl=64 time=0.075 ms
64 bytes from 10.53.1.2: icmp_seq=8 ttl=64 time=0.046 ms
64 bytes from 10.53.1.2: icmp_seq=9 ttl=64 time=0.040 ms
64 bytes from 10.53.1.2: icmp_seq=10 ttl=64 time=0.033 ms

Regards,
Amit Palkar

pgawlowicz, 2024-07-03:
please try to enable IP forwarding on the host PC:
```
sudo sysctl -w net.ipv4.ip_forward=1
```

pgawlowicz, 2024-07-03:
could you also switch off the ogstun interface:
```
sudo ifconfig ogstun 0.0.0.0 down
```

amitpalkar, 2024-07-03:
> Ok shall i try both the options or one option at a time?

Ok tried both the above options one by one.  iperf is not working..

The below is the command line display from within the container.

***@***.***:/open5gs# iperf -c 10.45.1.2 -u -b 65M -i 1 -t 30
------------------------------------------------------------
Client connecting to 10.45.1.2, UDP port 5001
Sending 1470 byte datagrams, IPG target: 172.54 us (kalman adjust)
UDP buffer size:  208 KByte (default)
------------------------------------------------------------
[  1] local 10.45.1.1 port 36667 connected with 10.45.1.2 port 5001
[ ID] Interval       Transfer     Bandwidth
[  1] 0.0000-1.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 1.0000-2.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 2.0000-3.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 3.0000-4.0000 sec  8.12 MBytes  68.1 Mbits/sec
[  1] 4.0000-5.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 5.0000-6.0000 sec  8.12 MBytes  68.1 Mbits/sec
[  1] 6.0000-7.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 7.0000-8.0000 sec  8.13 MBytes  68.2 Mbits/sec
[  1] 8.0000-9.0000 sec  8.12 MBytes  68.1 Mbits/sec

The below is the display from another console terminal.

sudo ip netns exec ue1 iperf -s -i 1
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------

Regards,
Amit Palkar

amitpalkar, 2024-07-03:
Hi,
    The downlink iperf3 started working.  The thing is we cannot give the
iperf command from inside the container.  I opened another terminal.
There i gave the command docker exec -it open5gs_5gc iperf3 -c 10.45.1.2 -u
-b 50M -i 1 -t 100
On another terminal i gave the command sudo ip netns exec ue1 iperf3 -s -i
Witt this i am able to do downlink data transfer.
Thanks a lot for your support and help.

One questions. i am seeing a maximum of 50 Mbps, is it possible to increase
throughput via higher modulation config (since i am using an SRS UE
simulation via ZMQ)..

Regards,
Amit Palkar

pgawlowicz, 2024-07-03:
I think that 50Mbps is max what the srsUE can achieve with 20Mhz channel.

closed 2024-07-03 by pgawlowicz

amitpalkar, 2024-07-03:
Hi,
     Is there a way in which srs ue can support higher modulation schemes
like 256 QAM ?
Regards,
Amit Palkar

pgawlowicz, 2024-07-04:
this is what i get from the gnb console trace when with srsUE and 20MHz channel:
```
          |--------------------DL---------------------|-------------------------UL------------------------------
 pci rnti | cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  rsrp  mcs  brate   ok  nok  (%)    bsr     ta  phr
   1 4601 |  15 1.0   28    59M 1000    0   0%  6.15M |  65.4   ovl   28    74M 1000    0   0%   300k      0  n/a
   1 4601 |  15 1.0   28    59M 1000    0   0%  6.15M |  65.4   ovl   28    74M 1000    0   0%   300k      0  n/a
   1 4601 |  15 1.0   28    59M 1000    0   0%  6.15M |  65.4   ovl   28    74M 1000    0   0%   300k      0  n/a
   1 4601 |  15 1.0   28    59M 1000    0   0%  6.15M |  65.4   ovl   28    74M 1000    0   0%   300k      0  n/a
   1 4601 |  15 1.0   28    59M 1000    0   0%  6.15M |  65.5   ovl   28    74M 1000    0   0%   300k      0  n/
```

I am using iperf with UDP to generate traffic from core in DL, and from UE in UL.