srsran/srsRAN_Project #701: NG Setup Procedure Failed - AMF NGAP Cause: Unknown-PLMN-or-SNPN

dominikheinz, 2024-07-03:
## Issue Description ##

I am trying to setup a private standalone 5G network with a USRP B210 using srsRAN and open5gs.
For this, I followed the official documentation and the quickstart guide:
- Open5gs quickstart guide: https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html
- srsRAN documentation: https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html

Currently, I am having trouble starting gnb with my configuration. I keep getting the following error:

```
--== srsRAN gNB (commit a804c5b9c) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in quad executor mode.
N2: Connection to AMF on 127.0.0.5:38412 completed
"NG Setup Procedure" failed. AMF NGAP cause: "unknown-PLMN-or-SNPN"
srsLog error - Unable to create log file "/tmp/gnb.log": Permission denied
``` 

I am not sure what causes the error `"NG Setup Procedure" failed. AMF NGAP cause: "unknown-PLMN-or-SNPN"`.

## Setup Details ##
- *SDF:* USRP B210
- *OS:* Ubuntu 22.04

## Expected Behavior ##

srsRAN gNB starts normally.

## Actual Behaviour ##

srsRAN fails to start with the error `"NG Setup Procedure" failed. AMF NGAP cause: "unknown-PLMN-or-SNPN"`.

## Steps to reproduce the problem ##

- Install open5gs from package manager
- Install srsRAN from source
- Connect the B210 and install the firmware for it
- Try to start gnb loading the config referenced below (`sudo ./gnb -c gnb_b210_20MHz_oneplus_8t.yaml`) and observe the error. 

## Additional Information ##

*/etc/open5gs/mme.yaml*
```
logger:
  default:
    timestamp: false
  file:
    path: /var/log/open5gs/mme.log
    timestamp: true

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

mme:
  freeDiameter: /etc/freeDiameter/mme.conf
  s1ap:
    server:
      - address: 127.0.0.2
  gtpc:
    server:
      - address: 127.0.0.2
    client:
      sgwc:
        - address: 127.0.0.3
      smf:
        - address: 127.0.0.4
  metrics:
    server:
      - address: 127.0.0.2
        port: 9090
  gummei:
    - plmn_id:
        mcc: 001
        mnc: 01
      mme_gid: 2
      mme_code: 1
  tai:
    - plmn_id:
        mcc: 001
        mnc: 01
      tac: 1
  security:
    integrity_order : [ EIA2, EIA1, EIA0 ]
    ciphering_order : [ EEA0, EEA1, EEA2 ]
  network_name:
    full: Open5GS
    short: Next
  mme_name: open5gs-mme0
  time:
```

**/etc/open5gs/amf.yaml*

```
logger:
  file:
    path: /var/log/open5gs/amf.log
    level: trace
global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

amf:
  sbi:
    server:
      - address: 127.0.0.5
        port: 7777
    client:
#      nrf:
#        - uri: http://127.0.0.10:7777
      scp:
        - uri: http://127.0.0.200:7777
  ngap:
    server:
      - address: 127.0.0.5
  metrics:
    server:
      - address: 127.0.0.5
        port: 9090
  guami:
    - plmn_id:
        mcc: 001
        mnc: 01
      amf_id:
        region: 2
        set: 1
  tai:
    - plmn_id:
        mcc: 001
        mnc: 01
      tac: 1
  plmn_support:
    - plmn_id:
        mcc: 001
        mnc: 01
      s_nssai:
        - sst: 1
  security:
    integrity_order : [ NIA2, NIA1, NIA0 ]
    ciphering_order : [ NEA0, NEA1, NEA2 ]
  network_name:
    full: Open5GS
    short: Next
  amf_name: open5gs-amf0
  time:
    t3512:
      value: 540    # 9 minutes * 60 = 540 seconds
```

**/etc/open5gs/upf.yaml**

```
logger:
  file:
    path: /var/log/open5gs/upf.log

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

upf:
  pfcp:
    server:
      - address: 127.0.0.7
    client:
#      smf:     #  UPF PFCP Client try to associate SMF PFCP Server
#        - address: 127.0.0.4
  gtpu:
    server:
      - address: 127.0.0.7
  session:
    - subnet: 10.45.0.0/16
      gateway: 10.45.0.1
    - subnet: 2001:db8:cafe::/48
      gateway: 2001:db8:cafe::1
  metrics:
    server:
      - address: 127.0.0.7
        port: 9090
```

**/etc/open5gs/nrf.yaml**

```
logger:
  file:
    path: /var/log/open5gs/nrf.log

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

nrf:
  serving:  # 5G roaming requires PLMN in NRF
    - plmn_id:
        mcc: 001
        mnc: 01
  sbi:
    server:
      - address: 127.0.0.10
        port: 7777
```

And finally the configuration that I am trying to use with gnb: `sudo ./gnb -c config.yaml`

**gnb_b210_20MHz_oneplus_8t.yml**

```
# This configuration file example shows how to configure the srsRAN Project gNB to connect to a COTS UE. As with the 
# associated tutorial, this config has been tested with a OnePlus 8T. A B210 USRP is used as the RF-frontend.   
# This config creates a TDD cell with 20 MHz bandwidth in band 78. 
# To run the srsRAN Project gNB with this config, use the following command: 
#   sudo ./gnb -c gnb_b210_20MHz_oneplus_8t.yaml

amf:
  addr: 127.0.0.5 #127.0.1.100                                             # The address or hostname of the AMF.
  bind_addr: 127.0.0.1                                          # A local IP that the gNB binds to for traffic from the AMF.

ru_sdr:
  device_driver: uhd                                            # The RF driver name.
  device_args: type=b200,num_recv_frames=64,num_send_frames=64  # Optionally pass arguments to the selected RF driver.
  sync: external                                                # Specify the sync source used by the RF. NOTE: Set to internal if NOT using an external 10 MHz reference clock. 
  srate: 23.04                                                  # RF sample rate might need to be adjusted according to selected bandwidth.
  otw_format: sc12
  tx_gain: 80                                                   # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 40                                                   # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 627340                                              # ARFCN of the downlink carrier (center frequency).
  band: 78                                                      # The NR band.
  channel_bandwidth_MHz: 20                                     # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                                                # Subcarrier spacing in kHz used for data.
  plmn: "90170"                                                 # PLMN broadcasted by the gNB.
  tac: 7                                                        # Tracking area code (needs to match the core configuration).
  pci: 1                                                        # Physical cell ID.

pcap:
  mac_enable: false                                             # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap                               # Path where the MAC PCAP is stored.
  ngap_enable: false                                            # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap                             # Path where the NGAP PCAP is stored.

```

pgawlowicz, 2024-07-03:
Please unify the following parameters:

- `cell_cfg.plmn` in gnb config
- and all  `plmn_id` in open5gs config files.

This is described in our tutorial here: https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html#gnb

dominikheinz, 2024-07-03:
> Please unify the following parameters:
> 
> * `cell_cfg.plmn` in gnb config
> * and all  `plmn_id` in open5gs config files.
> 
> This is described in our tutorial here: https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html#gnb

Thanks for the fast response, very much appreciated!
In order to follow the tutorial, should I set all instances of `plmnd_id` in all open5gs config files to 901 for mcc and 70 to mnc, or do you mean the other way around, just change the "cell_cfg.plmn" value to 00101 in the gnb config? (I have already tried the latter with no success)

dominikheinz, 2024-07-03:
I tried both ways, and I still get the same error.

pgawlowicz, 2024-07-03:
also `tac` has to be the same

pgawlowicz, 2024-07-03:
did you restart open5gs services?

dominikheinz, 2024-07-03:
> did you restart open5gs services?

Aaaaah that seemed to been the issue, I never restarted the services so the changes in the config files were not loaded it seems. Thank You, now gnb starts.

Another issue however seems to be the warning I get at the start:
```
The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in quad executor mode.
```

Do you know by any chance what I could do to solve this? It suggests lowering PHY in quad executor mode, but I dont know what that is, where to change it and what to change it to.

Thanks again for being so responsive by the way :)

pgawlowicz, 2024-07-03:
check this answer: https://github.com/srsran/srsRAN_Project/discussions/414#discussioncomment-8233889

pgawlowicz, 2024-07-03:
```
Lower PHY in quad executor mode.
```
it is not a suggestion, but display the config of the `Lower PHY execution mode`

closed 2024-07-03 by pgawlowicz