srsran/srsRAN_Project #706: gNB refuses connection for UE

dominikheinz, 2024-07-05:
## Issue Description ##
I am trying to setup a simple SA 5G network with srsRAN and open5gs.
The UE tries to connect, shows up briefly in the logs with connection refused and then is removed.

## Setup Details ##
My setup uses the following components:
For the basestation I have a USRP B210, connected to a system running Ubuntu 22.04.
The UE consists of a [RM520NGLAA-M20-SGASA 5G Module](https://shop.tekmodul.de/RM520N-GL-p541645396) connected via USB to a [M.2 Carrier Board](https://shop.tekmodul.de/M-2-USB-modem-carrier-board-standard-edition-p541645344).
The UE uses a [sysmoISIM](https://shop.sysmocom.de/sysmoISIM-SJA5-SIM-USIM-ISIM-Card-10-pack-with-ADM-keys/sysmoISIM-SJA5-9FV-10p-adm).
The UE is connected to a separate Laptop, also running Ubuntu 22.04.

I start the gnb like so with my config (config is linked below):
```
sudo ./gnb -c gnb_b210_20MHz_oneplus_8t.yml
```
The gNB does start, and then, when powering the UE, after a short moment, it does show up in the output, but only briefly:

```
$ sudo ./gnb -c gnb_b210_20MHz_oneplus_8t.yml 

--== srsRAN gNB (commit a804c5b9c) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in quad executor mode.
N2: Connection to AMF on 127.0.0.5:38412 completed
Cell pci=1, bw=20 MHz, 1T1R, dl_arfcn=627340 (n78), dl_freq=3410.1 MHz, dl_ssb_arfcn=626976, ul_freq=3410.1 MHz

Available radio types: uhd.
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.7.0.0-0ubuntu1~jammy1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200,num_recv_frames=64,num_send_frames=64'
[INFO] [B200] Detected Device: B210
[INFO] [B200] Loading FPGA image: /usr/share/uhd/images/usrp_b210_fpga.bin...
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] No GPSDO found
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
==== gNB started ===
Type <h> to view help
t

          |--------------------DL---------------------|-------------------------UL------------------------------
 pci rnti | cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  rsrp  mcs  brate   ok  nok  (%)    bsr     ta  phr
   1 4601 |  15 1.0   23   8.6k   11    0   0%      0 |  27.1 -25.6   26    43k   10    0   0%      0  -317n   12
   1 4601 |  15 1.0    0      0    0    0   0%      0 |  27.2 -26.2   28   8.7k    2    0   0%      0  -244n   11
   1 4601 |  15 1.0    0      0    0    0   0%      0 |   n/a   n/a    0      0    0    0   0%      0  -195n   11
```
It only shows these 3 lines of output, nothing more.
The SIM card used in the UE is correctly registered in open5gs with IMSI, Subscriber key and Operator key.

Looking into the logs `/var/logs/open5gs/amf.log` reveals that the conncetion is refused?

```
07/05 09:53:52.367: [amf] INFO: gNB-N2 accepted[127.0.0.1]:34318 in ng-path module (../src/amf/ngap-sctp.c:113)
07/05 09:53:52.367: [amf] INFO: gNB-N2 accepted[127.0.0.1] in master_sm module (../src/amf/amf-sm.c:757)
07/05 09:53:52.370: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1236)
07/05 09:53:52.370: [amf] INFO: gNB-N2[127.0.0.1] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:796)
07/05 09:54:52.631: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
07/05 09:54:52.631: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2662)
[... 216 lines cut ...]
        port: 9090
```

**/etc/open5gs/nrf/yaml:**
```
logger:
  file:
    path: /var/log/open5gs/nrf.log
#  level: info   # fatal|error|warn|info(default)|debug|trace

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

nrf:
  serving:  # 5G roaming requires PLMN in NRF
    - plmn_id:
        mcc: 999
        mnc: 70
  sbi:
    server:
      - address: 127.0.0.10
        port: 7777
```

pgawlowicz, 2024-07-05:
please set TZ env variable in the PC running Open5gs, eg.
```
export TZ=Europe/Madrid
```

dominikheinz, 2024-07-05:
> please set TZ env variable in the PC running Open5gs, eg.
> 
> ```
> export TZ=Europe/Madrid
> ```

I did set it, and made sure to restart all services.
Now I don't see the connection refused error in the logs, but the UE still gets disconnected. These are the logs:

```
07/05 11:44:07.167: [app] INFO: Configuration: '/etc/open5gs/amf.yaml' (../lib/app/ogs-init.c:133)
07/05 11:44:07.167: [app] INFO: File Logging: '/var/log/open5gs/amf.log' (../lib/app/ogs-init.c:136)
07/05 11:44:07.168: [metrics] INFO: metrics_server() [http://127.0.0.5]:9090 (../lib/metrics/prometheus/context.c:299)
07/05 11:44:07.168: [sbi] INFO: NF Service [namf-comm] (../lib/sbi/context.c:1841)
07/05 11:44:07.169: [sbi] INFO: nghttp2_server() [http://127.0.0.5]:7777 (../lib/sbi/nghttp2-server.c:414)
07/05 11:44:07.169: [amf] INFO: ngap_server() [127.0.0.5]:38412 (../src/amf/ngap-sctp.c:61)
07/05 11:44:07.169: [sctp] INFO: AMF initialize...done (../src/amf/app.c:33)
07/05 11:44:07.169: [sbi] WARNING: [7] Failed to connect to 127.0.0.200 port 7777 after 0 ms: Connection refused (../lib/sbi/client.c:698)
07/05 11:44:07.169: [sbi] WARNING: ogs_sbi_client_handler() failed [-1] (../lib/sbi/path.c:62)
07/05 11:44:18.180: [sbi] WARNING: [1f5f8108-3ab3-41ef-b98d-3b012d91cb8a] Retry registration with NRF (../lib/sbi/nf-sm.c:182)
07/05 11:44:18.181: [sbi] INFO: [1f5f8108-3ab3-41ef-b98d-3b012d91cb8a] NF registered [Heartbeat:10s] (../lib/sbi/nf-sm.c:221)
07/05 11:44:18.182: [sbi] INFO: [25f004c0-3ab3-41ef-b8c2-c7a1c0cafa01] Subscription created until 2024-07-06T11:44:18.181862+02:00 [duration:86400,validity:86399.999517,patch:43199.999758] (../lib/sbi/nnrf-handler.c:813)
07/05 11:44:18.182: [sbi] INFO: [25f0065a-3ab3-41ef-b8c2-c7a1c0cafa01] Subscription created until 2024-07-06T11:44:18.181899+02:00 [duration:86400,validity:86399.999375,patch:43199.999687] (../lib/sbi/nnrf-handler.c:813)
07/05 11:44:18.182: [sbi] INFO: [25f0070e-3ab3-41ef-b8c2-c7a1c0cafa01] Subscription created until 2024-07-06T11:44:18.181916+02:00 [duration:86400,validity:86399.999373,patch:43199.999686] (../lib/sbi/nnrf-handler.c:813)
07/05 11:44:18.182: [sbi] INFO: [25f0092a-3ab3-41ef-b8c2-c7a1c0cafa01] Subscription created until 2024-07-06T11:44:18.181971+02:00 [duration:86400,validity:86399.999402,patch:43199.999701] (../lib/sbi/nnrf-handler.c:813)
07/05 11:44:18.182: [sbi] INFO: [25f00ac4-3ab3-41ef-b8c2-c7a1c0cafa01] Subscription created until 2024-07-06T11:44:18.182011+02:00 [duration:86400,validity:86399.999435,patch:43199.999717] (../lib/sbi/nnrf-handler.c:813)
07/05 11:44:18.182: [sbi] INFO: [25f00b96-3ab3-41ef-b8c2-c7a1c0cafa01] Subscription created until 2024-07-06T11:44:18.182032+02:00 [duration:86400,validity:86399.999442,patch:43199.999721] (../lib/sbi/nnrf-handler.c:813)
07/05 11:44:18.182: [sbi] INFO: [25f00c72-3ab3-41ef-b8c2-c7a1c0cafa01] Subscription created until 2024-07-06T11:44:18.182055+02:00 [duration:86400,validity:86399.999450,patch:43199.999725] (../lib/sbi/nnrf-handler.c:813)
07/05 11:44:18.202: [sbi] INFO: (NRF-notify) NF registered [1f610f50-3ab3-41ef-84fa-e9978b7a93a3:1] (../lib/sbi/nnrf-handler.c:1058)
07/05 11:44:18.202: [sbi] INFO: [SMF] (NRF-notify) NF Profile updated [1f610f50-3ab3-41ef-84fa-e9978b7a93a3:1] (../lib/sbi/nnrf-handler.c:1072)
07/05 11:44:29.124: [amf] INFO: gNB-N2 accepted[127.0.0.1]:58539 in ng-path module (../src/amf/ngap-sctp.c:113)
07/05 11:44:29.124: [amf] INFO: gNB-N2 accepted[127.0.0.1] in master_sm module (../src/amf/amf-sm.c:757)
07/05 11:44:29.127: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1236)
07/05 11:44:29.127: [amf] INFO: gNB-N2[127.0.0.1] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:796)
07/05 11:45:36.075: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
07/05 11:45:36.075: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2662)
07/05 11:45:36.075: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] TAC[7] CellID[0x66c000] (../src/amf/ngap-handler.c:562)
07/05 11:45:36.075: [amf] INFO: Unknown UE by 5G-S_TMSI[AMF_ID:0x20040,M_TMSI:0xc000046b] (../src/amf/context.c:1862)
07/05 11:45:36.075: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1621)
07/05 11:45:36.075: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1214)
07/05 11:45:36.075: [gmm] INFO: [Unknown ID]    5G-S_GUTI[AMF_ID:0x20040,M_TMSI:0xc000046b] (../src/amf/gmm-handler.c:185)
07/05 11:45:36.094: [gmm] INFO: Identity response (../src/amf/gmm-sm.c:1482)
07/05 11:45:36.094: [gmm] INFO: [suci-0-999-70-0-0-0-0000066491]    SUCI (../src/amf/gmm-handler.c:944)
07/05 11:45:36.098: [sbi] INFO: [AUSF] (SCP-discover) NF registered [1f665df2-3ab3-41ef-bf9b-c542a487b81e:1] (../lib/sbi/path.c:211)
07/05 11:45:36.196: [sbi] INFO: [UDM] (SCP-discover) NF registered [1f6715a8-3ab3-41ef-9ef2-67f3736f3184:1] (../lib/sbi/path.c:211)
07/05 11:45:36.197: [sbi] WARNING: [UDM] (SCP-discover) NF has already been added [1f6715a8-3ab3-41ef-9ef2-67f3736f3184:2] (../lib/sbi/path.c:216)
07/05 11:45:36.202: [sbi] INFO: [PCF] (SCP-discover) NF registered [1f6852c4-3ab3-41ef-9937-2f9e05705934:1] (../lib/sbi/path.c:211)
07/05 11:45:36.202: [amf] WARNING: NF EndPoint(addr) updated [127.0.0.12:7777] (../src/amf/npcf-handler.c:143)
07/05 11:45:36.435: [gmm] INFO: [imsi-<censored>] Registration complete (../src/amf/gmm-sm.c:2313)
07/05 11:45:36.435: [amf] INFO: [imsi-<censored>] Configuration update command (../src/amf/nas-path.c:591)
[... cut ...]
 pci rnti | cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  rsrp  mcs  brate   ok  nok  (%)    bsr     ta  phr
   1 4601 |  15 1.0    5   2.6k    2    0   0%      0 |  23.7 -25.2    9   4.2k    1    0   0%      0   -65n    7
   1 4601 |  15 1.0   27   6.0k    9    0   0%      0 |  28.1 -24.8   28    30k    7    0   0%      0   -56n   12
   1 4601 |  15 1.0    0      0    0    0   0%      0 |  28.1 -25.1   28   8.7k    2    0   0%      0    65n   12
   1 4601 |  14 1.0   27      0    0    0   0%      0 |   n/a   n/a    0      0    0    0   0%      0    40n   12
Late: 0; Underflow: 1; Overflow: 0;
```

pgawlowicz, 2024-07-05:
could you share ngap.pcap?

dominikheinz, 2024-07-05:
> could you share ngap.pcap?

Sure, but I can't seem to find it on my system? How do I capture it?

dominikheinz, 2024-07-05:
> could you share ngap.pcap?

Ah, well I guess it is defined in my config that I pass to gnb.
However, I can't seem to find the pcap files in the tmp dir, where they should appear according to my config, no?
There is however a gnb.log file, and among a lot of nonsense output, it also says:

```
2024-07-05T10:39:58.156080 [NGAP    ] [W] ue=0 ran_ue=6 amf_ue=7: PDU session setup timer expired after 3000ms. Requesting UE release
```
Is that helpful?

pgawlowicz, 2024-07-05:
could you run with debug log mode and share logs?

dominikheinz, 2024-07-05:
> could you run with debug log mode and share logs?

I attached the debug logs and also the *.pcap files (forgot to set pcap logging to true in the config earlier).
[debug-logs.zip](https://github.com/user-attachments/files/16109050/debug-logs.zip)

herlesupreeth, 2024-07-05:
@dominikheinz Quectel 5G modems by default does a initial RRC Setup procedure only and no PDN session is requested thats why we release the UE from gNB if there is no PDN session (also the reason you see only 3 lines printed in trace output of gNB)

In order to establish a PDN connection you need do the following on the machine to which your Quectel modem is connected

```
# Lists all the modem on the system
sudo mmcli -L
# Take note of the modem number corresponding to Quectel modem from the above list of modems
# Enable modem
sudo mmcli -m <modem_number> -e
# Establish a PDN connection to DNN/APN 'srsapn'
sudo mmcli -m <modem_number> --simple-connect="apn=srsapn"
```

Then, machine to which your Quectel modem is connected find out the modem interface e.g. wwan0 or wwan1

Once, the IP is assigned in open5gs output take note of that
```
sudo ip link set wwan0 up
sudo ip a add <UE_IP>/<SUBNET_MASK> dev wwan0
```

dominikheinz, 2024-07-05:
> @dominikheinz Quectel 5G modems by default does only initial RRC Setup procedure and no PDN session is requested thats why we release the UE from gNB if there is no PDN session (also the reason you see only 3 lines printed in trace output of gNB)
> 
> In order to establish a PDN connection you need do the following on the machine to which your Quectel modem is connected
> 
> ```
> # Lists all the modem on the system
> sudo mmcli -L
> # Take note of the modem number corresponding to Quectel modem from the above list of modems
> # Enable modem
> sudo mmcli -m <modem_number> -e
> # Establish a PDN connection to DNN/APN 'srsapn'
> sudo mmcli -m <modem_number> --simple-connect="apn=srsapn"
> ```
> 
> Then, machine to which your Quectel modem is connected find out the modem interface e.g. wwan0 or wwan1
> 
> Once, the IP is assigned in open5gs output take note of that
> 
> ```
> sudo ip link set wwan0 up
> sudo ip a add <UE_IP>/<SUBNET_MASK> dev wwan0
> ```

Thanks for the reply.
Trying to follow your steps, currently. However, the system where the UE is connected to fails at the step where you try to do the simple connect:

```
lab@lab-Latitude-3:~$ sudo mmcli -L
[sudo] password for lab: 
    /org/freedesktop/ModemManager1/Modem/4 [Quectel] RM520N-GL
lab@lab-Latitude-3:~$ sudo mmcli -m 4 -e
successfully enabled the modem
lab@lab-Latitude-3:~$ sudo mmcli -m 4 --simple-connect="apn=srsapn"
error: couldn't connect the modem: 'GDBus.Error:org.freedesktop.ModemManager1.Error.MobileEquipment.NetworkTimeout: Network timeout'
lab@lab-Latitude-3:~$ 
```

herlesupreeth, 2024-07-05:
> Trying to follow your steps, currently. However, the system where the UE is connected to fails at the step where you try to do the simple connect:

You have to perform those steps while the UE connects to gNB

dominikheinz, 2024-07-05:
> > Trying to follow your steps, currently. However, the system where the UE is connected to fails at the step where you try to do the simple connect:
> 
> You have to perform those steps while the UE connects to gNB. Also, make sure the APN name you configured in 5GC is same as what you are using in that command

Okay, I think I got it working now? Kind of? At least I got more continous output in gnb. But after a while, it also stopped again and the UE got disconnected again. Console output:

```
fiveglab@fiveglab-Precision-3680:~/Documents/srsRAN_Project/build/apps/gnb$ sudo ./gnb -c gnb_b210_20MHz_oneplus_8t.yml 

--== srsRAN gNB (commit a804c5b9c) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in quad executor mode.
N2: Connection to AMF on 127.0.0.5:38412 completed
Cell pci=1, bw=20 MHz, 1T1R, dl_arfcn=627340 (n78), dl_freq=3410.1 MHz, dl_ssb_arfcn=626976, ul_freq=3410.1 MHz

Available radio types: uhd.
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.7.0.0-0ubuntu1~jammy1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200,num_recv_frames=64,num_send_frames=64'
[DEBUG] [B200] the firmware image: /usr/share/uhd/images/usrp_b200_fw.hex
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[DEBUG] [AD936X] Performing CODEC loopback test... 
[DEBUG] [AD936X] CODEC loopback test passed.
[DEBUG] [AD936X] Performing CODEC loopback test... 
[DEBUG] [AD936X] CODEC loopback test passed.
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[DEBUG] [CORES] Performing timer loopback test... 
[DEBUG] [CORES] Timer loopback test passed.
[DEBUG] [CORES] Performing timer loopback test... 
[DEBUG] [CORES] Timer loopback test passed.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
[DEBUG] [CORES] Performing timer loopback test... 
[DEBUG] [CORES] Timer loopback test passed.
[DEBUG] [CORES] Performing timer loopback test... 
[DEBUG] [CORES] Timer loopback test passed.
[DEBUG] [CONVERT] get_converter: For converter ID: conversion ID
  Input format:  fc32
  Num inputs:    1
  Output format: sc12_item32_le
  Num outputs:   1
 Using best available prio: 0
[DEBUG] [CONVERT] get_converter: For converter ID: conversion ID
  Input format:  sc12_item32_le
  Num inputs:    1
  Output format: fc32
  Num outputs:   1
 Using best available prio: 0
==== gNB started ===
Type <h> to view help
Late: 0; Underflow: 1; Overflow: 0;
[... 114 lines cut ...]
   1 4602 |  15 1.0    0      0    0    0   0%      0 |  30.6 -21.1   28   4.4k    1    0   0%      0    -8n   14

          |--------------------DL---------------------|-------------------------UL------------------------------
 pci rnti | cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  rsrp  mcs  brate   ok  nok  (%)    bsr     ta  phr
   1 4602 |  15 1.0    0      0    0    0   0%      0 |  30.9 -21.1   28   8.7k    2    0   0%      0  -113n   14
   1 4602 |  15 1.0    0      0    0    0   0%      0 |   n/a   n/a    0      0    0    0   0%      0   -40n   14
   1 4602 |  15 1.0    0      0    0    0   0%      0 |   n/a   n/a    0      0    0    0   0%      0    48n   14
Late: 0; Underflow: 1; Overflow: 0;
   1 4602 |  15 1.0    0      0    0    0   0%      0 |  30.6 -21.1   28   4.4k    1    0   0%      0    40n   14
   1 4602 |  15 1.0    0      0    0    0   0%      0 |   n/a   n/a    0      0    0    0   0%      0      0   14
   1 4602 |  15 1.0    0      0    0    0   0%      0 |  30.2 -21.3   28   4.4k    1    0   0%      0      0   14
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
^CStopping ..
NGAP PCAP stored in /tmp/gnb_ngap.pcap
MAC PCAP stored in /tmp/gnb_mac.pcap
```

I am not sure what the APN of my base station is? Is that specified in the config that I pass to gnb? Because I don't see it there anywhere? I just used the value "srsapn" from your example, maybe the mismatch is the cause why the connection drops?

What exactly does the simple command do, that I run via mmcli? How is it different from what happened earlier? (You mentioned the RRC Setup procedure and PDN, I am not fully sure what exactly this means). 

I attached the logs and the pcap files again at the bottom, - after a while, the connection seems to drop again, and I terminate gnb.

[logs2.zip](https://github.com/user-attachments/files/16109439/logs2.zip)

dominikheinz, 2024-07-05:
> An APN/DNN is not a parameter of base station rather a 5G Core network and UE. Its specified in open5gs SMF configuration file.

Okay. I currently don't see it present in my `smf.yaml`.  What would be a minimal example to set it?

> From the NGAP pcap I see that APN/DNN configured in SMF is not 'srsapn' rather its 'internet'. So you should use internet in that --simple-connect command

Okay, I tried that, but the same result. It stays connected for a while, until it gets removed (because its idle for to long, as you explained, I assume)

> It trigger a PDN session establishment request towards the 5G Core Network via gNB using the APN/DNN provided in the command. mmcli --> ModemManager Command Line

Okay, I see. I assume no PDN session get established? Otherwise, the UE shouldnt get disconnected after some time?

> The sequence of operation when a Quectel modem attaches the following in 5G
> 
> 1. Modem sees the gNB
> 2. Modem initiates RRC Setup procedure and completes the authentication with the Core Network
> 3. Then, stops and doesnt do anything further
> 4. Only, when you ask via mmcli to create PDN session, modem requests the Core Network to create a data path for traffic such as internet etc. to flow
> 
> Without any PDN session, at gNB we release that UE after certain time of inactivity. This is done mainly for two reasons
> 
> 1. Save UE power
> 2. Not to waste precious allocation resources in gNB for a UE who doesn't have a path setup for traffic to flow

Okay, how can I tell the UE via mmcli to create a PDN session? Or is that part of the simple-connect command? Ideally, all these steps should happen automatically when the UE is plugged in.
The goal is that the laptop where the UE is connected is able to have an internet connection via 5G. I also noticed, when running `ip a`, the wwan0 interface does not get assigned an ip address.

What would be the next steps to ensure the UE stays connected, has a PDN session and can access the internet via the modem? 

Also: Thank you for being so helpful and responsive, much appreciated!

herlesupreeth, 2024-07-05:
> Okay. I currently don't see it present in my smf.yaml. What would be a minimal example to set it?

Then, you would have configured it in the open5gs WebUI where you entered subscriber (SIM) details

> Okay, I tried that, but the same result. It stays connected for a while, until it gets removed (because its idle for to long, as you explained, I assume)

There should be some traffic atleast a ping

> Okay, how can I tell the UE via mmcli to create a PDN session? Or is that part of the simple-connect command?

simple-connect is the command which tells the UE to create PDN session

> Ideally, all these steps should happen automatically when the UE is plugged in.

You can configure nmcli (not mmcli) to do all this steps automatically (i havent done so but I know its possible)

> The goal is that the laptop where the UE is connected is able to have an internet connection via 5G. I also noticed, when running ip a, the wwan0 interface does not get assigned an ip address.

As mentioned before, you need to perform the steps mentioned at the bottom here (https://github.com/srsran/srsRAN_Project/issues/706#issuecomment-2210713080) to assign IP address to wwan0 interface. Even this step of manually assigning IP address be skipped if you configure via nmcli.

dominikheinz, 2024-07-08:
@herlesupreeth 

> Then, you would have configured it in the open5gs WebUI where you entered subscriber (SIM) details

Got it. The APN name should be correct, then.

> There should be some traffic atleast a ping

Could you clarify what you mean by ping exactly? Since both systems are also connected via LAN, when I ping, for example, the Laptop with the UE from the gNB, it works, but it’s not pinging over 5G, just over Ethernet... (Both test systems are in the same subnet).

> simple-connect is the command which tells the UE to create PDN session

Understood.

> You can configure nmcli (not mmcli) to do all this steps automatically (i havent done so but I know its possible)

Thanks for the tip. I will look into this once I get the setup working manually with mmcli. First, I need to get it to work, then make it more efficient :wink:

> As mentioned before, you need to perform the steps mentioned at the bottom here (https://github.com/srsran/srsRAN_Project/issues/706#issuecomment-2210713080) to assign IP address to wwan0 interface. Even this step of manually assigning IP address be skipped if you configure via nmcli.

I did this. Note that the Laptop the UE is connected to is also connected via Ethernet for remote development. Regardless, when I assign an IP address to the wwan0 interface (different from the IP that the laptop has on the Ethernet interface), I still can't ping the UE from the gNB. Pinging locally, aka pinging from the Laptop the UE is connected to, to the IP I assigned to wwan0 obviously works, but I assume that is just loopback. I might be missing something. I can see that a connection is established because I can see the trace output on the gNB side. However, since there is no traffic, srsRAN will timeout the UE after a few minutes.

The main issue here is that I am unsure how to proceed. What is a good way to troubleshoot this? It is essential to figure out, why there is no data being sent between the gNB and the UE. I assume it is either a configuration issue with open5gs or srsRAN, or something with mmcli and setting up the modem is not right. Do you have any suggestions on what to try next?

herlesupreeth, 2024-07-08:
@dominikheinz After you assign the IP to wwan0 interface then you can perform a ping to 5G Core as follows (I am assuming that the TUN interface in open5gs is created with IP 10.45.0.1)

`ping 10.45.0.1 -I wwan0`

dominikheinz, 2024-07-08:
> ping 10.45.0.1 -I wwan0

@herlesupreeth Welp, no pings are going through, so I definitely did something wrong :smile: 

Just to make sure, I didn't mess up with the network configuration, this is how I assigned the IP: `sudo ip a add 192.168.42.80/255.255.255.0 dev wwan0`
I wasn't sure what to pick for the IP, so I picked something thats free in the network... But I feel like this is already wrong..?

herlesupreeth, 2024-07-08:
> Just to make sure, I didn't mess up with the network configuration, this is how I assigned the IP: sudo ip a add 192.168.42.80/255.255.255.0 dev wwan0
> I wasn't sure what to pick for the IP, so I picked something thats free in the network... But I feel like this is already wrong..?

yes, this is wrong. The IP address to assign to Quectel should be printed in open5gs SMF/UPF logs. It would be in the subnet 10.45.0.0/16. So the command would be as follows (set correct IP based on what you see in open5gs logs when you run mmcli --simple-connect command)

`sudo ip a add 10.45.0.3/16 dev wwan0`

dominikheinz, 2024-07-08:
@herlesupreeth Awesome! Now I can also ping to5G core via wwan0 as you explained previously. :smile: Great progress! Thanks for being so patient with assisting me, very much appreciated!

Now the next questions are:
- How can I test this connection properly? E.g. surf the internet via the UE or download a file or something. This is currently not possible? What is the best way to ensure data is exchanged over the 5G connection?
- Currently, the UE still gets released/dropped after like 2 minutes or so - I assume because no data is being exchanged? am I missing another step? What is the next step to ensure the connection remains up. After all, once surfing the internet via the 5G network is possible, it can still happen that the system wont connect to the internet for a while, so it would be weird if it just gets disconnected suddenly?

open5gs logs:

```
07/08 15:55:52.352: [amf] INFO: [suci-0-999-70-0-0-0-0000066491] Known UE by 5G-S_TMSI[AMF_ID:0x20040,M_TMSI:0xc0000414] (../src/amf/context.c:1905)
07/08 15:55:52.352: [gmm] INFO: Service request (../src/amf/gmm-sm.c:1400)
07/08 15:55:52.352: [gmm] INFO: [suci-0-999-70-0-0-0-0000066491]    5G-S_GUTI[AMF_ID:0x20040,M_TMSI:0xc0000414] (../src/amf/gmm-handler.c:651)
07/08 15:55:52.433: [amf] INFO: [imsi-999700000066491:1:11][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:915)
07/08 15:58:00.303: [amf] INFO: [imsi-999700000066491:1:13][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:915)
07/08 15:58:00.440: [amf] INFO: UE Context Release [Action:2] (../src/amf/ngap-handler.c:1696)
07/08 15:58:00.440: [amf] INFO:     RAN_UE_NGAP_ID[2] AMF_UE_NGAP_ID[31] (../src/amf/ngap-handler.c:1697)
07/08 15:58:00.440: [amf] INFO:     SUCI[suci-0-999-70-0-0-0-0000066491] (../src/amf/ngap-handler.c:1701)
07/08 15:58:00.440: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2669)
```

herlesupreeth, 2024-07-08:
> How can I test this connection properly? E.g. surf the internet via the UE or download a file or something. This is currently not possible? What is the best way to ensure data is exchanged over the 5G connection?

For this you need to add default route via wwan0 interface after you assign IP to it. Then, all the traffic would go through wwan0 interface. Caution: you may get logged out of SSH session to add route appropriately

> Currently, the UE still gets released/dropped after like 2 minutes or so - I assume because no data is being exchanged? am I missing another step? What is the next step to ensure the connection [...]

You dont have to worry even if UE gets release because as soon as you start pushing any kind of traffic UE will be active again and traffic should flow.

dominikheinz, 2024-07-09:
@herlesupreeth 
Okay, I added the route like so: `sudo ip route add default via 10.45.0.7 dev wwan0`. It is now possible to ping the 5G core over wwan0 with `ping 10.45.0.1 -I wwan0`, but also external services e.g. the google dns server via `ping 8.8.8.8 -I wwan0`.
However, I still can't surf, probably because DNS does not work as it should? Is there anything I need to do, here?

herlesupreeth, 2024-07-09:
@dominikheinz If you are able to ping 8.8.8.8 from wwan0 interface and cant surf then its an issue in your local network setup and not an issue in srsRAN_Project gNB. Please check your local networking setup

closed 2024-08-08 by pgawlowicz